raspberrypi/linux #3172: rpi-4.19-rt does not boot on RPi3 or RPi2

gtrainavicius, 2019-08-22:
Testing out 4.19-rt kernel builds, built using [this script](https://github.com/BlokasLabs/rpi-kernel-rt/blob/master/build.sh), I found it to boot successfully on RPi4 and RPi2, but not on RPi3 B+.

The issue on RPi2 and RPi3 seems to be different, the RPi2 seems to be able to get further in the boot process, sometimes it even shows log in prompt, but hangs moments after.

**Systems and Software used**

* 2019-07-10 Raspbian Buster with desktop
* Kernel based on rpi-4.19.y-rt branch built for all RPi variations
* Resulting boot/ and lib/ folders copied over to /
* UART serial console interface enabled
* Boot logs collected using UART interface.
* Raspberry Pi 4 (4GB)
* Raspberry Pi 3B+
* Raspberry Pi 2B
* Nothing attached to GPIO, apart from UART interface

**Logs**
<details>
<summary>RPi3 boot log</summary>
<p>

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.59-rt23-v7+ (giedrius@giedrius-vb) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03)) #1 SMP PREEMPT RT Tue Jul 30 10:56:20 EEST 2019
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] random: get_random_bytes called from start_kernel+0x7c/0x4ac with crng_init=0
[    0.000000] percpu: Embedded 18 pages/cpu s41536 r8192 d24000 u73728
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=6d99072a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 938424K/970752K available (8192K kernel code, 629K rwdata, 2188K rodata, 1024K init, 957K bss, 24136K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 630 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 958 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26350 entries in 78 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU priority boosting: priority 1 delay 500 ms.
[    0.000000]  No expedited grace period (rcu_normal_after_boot).
[    0.000000]  Tasks RCU enabled.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000017] Switching to timer-based delay loop, resolution 52ns
[    0.000274] Console: colour dummy device 80x30
[    0.001399] console [tty1] enabled
[    0.001436] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001449] pid_max: default: 32768 minimum: 301
[    0.001829] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001843] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002935] CPU: Testing write buffer coherency: ok
[    0.003490] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.060149] Setting up static identity map for 0x100000 - 0x10003c
[    0.100132] rcu: Hierarchical SRCU implementation.
[    0.180324] smp: Bringing up secondary CPUs ...
[    0.321042] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[... 720 lines cut ...]
[  109.778650] 3fa0:                                     00000000 00000000 00000000 00000000
[  109.778661] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  109.778671] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  152.825875] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[  152.825898] rcu:     0-...!: (0 ticks this GP) idle=c22/1/0x40000002 softirq=0/0 fqs=0
[  152.825911] rcu:     1-...!: (0 ticks this GP) idle=00a/1/0x40000000 softirq=0/0 fqs=0
[  152.825914] rcu:     (detected by 2, t=14712 jiffies, g=-923, q=752)
[  152.825932] Sending NMI from CPU 2 to CPUs 0:
[  162.827218] Sending NMI from CPU 2 to CPUs 1:
[  172.828511] rcu: rcu_preempt kthread starved for 16712 jiffies! g-923 f0x2 RCU_GP_WAIT_FQS(5) ->state=0x0 ->cpu=1
[  172.828514] rcu: RCU grace-period kthread stack dump:
[  172.828518] rcu_preempt     R  running task        0    11      2 0x00000000
[  172.828552] [<808039f8>] (__schedule) from [<80803e00>] (schedule+0x64/0x108)
[  172.828572] [<80803e00>] (schedule) from [<80806898>] (schedule_timeout+0x1f4/0x494)
[  172.828592] [<80806898>] (schedule_timeout) from [<8018fa1c>] (rcu_gp_kthread+0x48c/0xb74)
[  172.828610] [<8018fa1c>] (rcu_gp_kthread) from [<80140c68>] (kthread+0x12c/0x168)
[  172.828626] [<80140c68>] (kthread) from [<801010b4>] (ret_from_fork+0x14/0x20)
[  172.828632] Exception stack(0xb9d53fb0 to 0xb9d53ff8)
[  172.828642] 3fa0:                                     00000000 00000000 00000000 00000000
[  172.828654] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  172.828664] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
```

</p>
</details>

gtrainavicius, 2019-09-05:
Tagging @TiejunChina, you seem to be doing the changes in the rpi-4.19.y-rt branch :). Do you have any ideas what's missing for RPi3 and RPi2?

akshayhiregoudar, 2019-09-13:
I installed the upgraded -rt branch on RPi3 and was able to boot it. But, the USB ports (connected a mouse and keyboard) stopped working after a couple of minutes of getting to the desktop.

gtrainavicius, 2019-09-16:
Still does not boot on 3B+, here are the logs:

<details><summary>3B+ log 1</summary>
<pre>
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.71-rt24-v7+ (giedrius@giedrius-vb) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03)) #1 SMP PREEMPT RT Sat Sep 14 17:54:07 EEST 2019
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] random: get_random_bytes called from start_kernel+0x7c/0x4ac with crng_init=0
[    0.000000] percpu: Embedded 18 pages/cpu s41536 r8192 d24000 u73728
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=8b59ba8e-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 938424K/970752K available (8192K kernel code, 629K rwdata, 2188K rodata, 1024K init, 957K bss, 24136K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 630 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 958 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26399 entries in 78 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU priority boosting: priority 1 delay 500 ms.
[    0.000000]  No expedited grace period (rcu_normal_after_boot).
[    0.000000]  Tasks RCU enabled.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000017] Switching to timer-based delay loop, resolution 52ns
[    0.000272] Console: colour dummy device 80x30
[    0.001393] console [tty1] enabled
[    0.001430] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001443] pid_max: default: 32768 minimum: 301
[    0.001825] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001839] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002929] CPU: Testing write buffer coherency: ok
[    0.003485] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.060146] Setting up static identity map for 0x100000 - 0x10003c
[    0.100132] rcu: Hierarchical SRCU implementation.
[    0.180320] smp: Bringing up secondary CPUs ...
[    0.321035] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.461267] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.601524] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.601738] smp: Brought up 1 node, 4 CPUs
[    0.601856] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.601893] CPU: All CPU(s) started in HYP mode.
[    0.601924] CPU: Virtualization extensions available.
[    0.603313] devtmpfs: initialized
[    0.636093] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.636793] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.636865] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.637628] pinctrl core: initialized pinctrl subsystem
[    0.639494] NET: Registered protocol family 16
[    0.645214] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.656337] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.656388] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.656638] Serial: AMBA PL011 UART driver
[    0.660339] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.729032] bcm2835-dma 3f007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
[... 1036 lines cut ...]
[   47.529466] [c145b22f] REQ> b8c8e0d0 0
[   47.534325] [c30825d3] TIM< 0 0
[   47.538609] mmc0: sbc op 23 arg 0x8 flags 0x15 - resp 00000900 00000000 00000000 00000000, err 0
[   47.550098] mmc0: cmd op 18 arg 0x12acd8 flags 0xb5 - resp 00000900 00000000 00000000 00000000, err 0
[   47.562531] mmc0: data blocks 8 blksz 200 - err 0
[... cut ...]
[   47.605300] mmc0: SDCDIV 0x00000003
[   47.610764] mmc0: SDRSP0 0x00000900
[   47.616197] mmc0: SDRSP1 0x00001269
[   47.621605] mmc0: SDRSP2 0x7fffffff
[   47.626985] mmc0: SDRSP3 0x00024007
[   47.632355] mmc0: SDHSTS 0x00000081
[   47.637736] mmc0: SDVDD  0x00000001
[   47.643099] mmc0: SDEDM  0x00010831
[   47.648469] mmc0: SDHCFG 0x0000040e
[   47.653789] mmc0: SDHBCT 0x00000200
[   47.659012] mmc0: SDHBLC 0x00000001
[   47.664073] mmc0: ===========================================

</pre>
</details>

gtrainavicius, 2019-09-16:
Does not boot on 2B:

<details><summary>2B log 1</summary>
<pre>
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 4.19.71-rt24-v7+ (giedrius@giedrius-vb) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03)) #1 SMP PREEMPT RT Sat Sep 14 17:54:07 EEST 2019
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] random: get_random_bytes called from start_kernel+0x7c/0x4ac with crng_init=0
[    0.000000] percpu: Embedded 18 pages/cpu s41536 r8192 d24000 u73728
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=8b59ba8e-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 938428K/970752K available (8192K kernel code, 629K rwdata, 2188K rodata, 1024K init, 957K bss, 24132K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 630 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 958 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26399 entries in 78 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU priority boosting: priority 1 delay 500 ms.
[    0.000000]  No expedited grace period (rcu_normal_after_boot).
[    0.000000]  Tasks RCU enabled.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000009] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000358] Console: colour dummy device 80x30
[    0.001621] console [tty1] enabled
[    0.001666] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001682] pid_max: default: 32768 minimum: 301
[    0.002168] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002191] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.003643] CPU: Testing write buffer coherency: ok
[    0.004423] CPU0: update cpu_capacity 1024
[    0.004658] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.060250] Setting up static identity map for 0x100000 - 0x10003c
[    0.100230] rcu: Hierarchical SRCU implementation.
[    0.180529] smp: Bringing up secondary CPUs ...
[    0.321570] CPU1: update cpu_capacity 1024
[    0.321580] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.461933] CPU2: update cpu_capacity 1024
[    0.461942] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.602306] CPU3: update cpu_capacity 1024
[    0.602316] CPU3: thread -1, cpu 3, socket 15, mpidr 80000f03
[    0.602587] smp: Brought up 1 node, 4 CPUs
[    0.602797] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.602841] CPU: All CPU(s) started in HYP mode.
[    0.602877] CPU: Virtualization extensions available.
[    0.604693] devtmpfs: initialized
[    0.644537] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.645515] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.645607] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.646685] pinctrl core: initialized pinctrl subsystem
[    0.649204] NET: Registered protocol family 16
[    0.655867] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.670086] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[... 411 lines cut ...]
[    3.648428] CPU: 2 PID: 81 Comm: irq/56-dwc_otg_ Tainted: G        W         4.19.71-rt24-v7+ #1
[    3.648433] Hardware name: BCM2835
[    3.648456] [<801110c8>] (unwind_backtrace) from [<8010c9ec>] (show_stack+0x20/0x24)
[    3.648474] [<8010c9ec>] (show_stack) from [<807efa58>] (dump_stack+0xbc/0x108)
[    3.648491] [<807efa58>] (dump_stack) from [<8011f8d0>] (__warn+0xf4/0x120)
[    3.648508] [<8011f8d0>] (__warn) from [<8011f9e8>] (warn_slowpath_null+0x4c/0x58)
[... cut ...]
[    3.648648] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
[    3.648655] ---[ end trace 0000000000000003 ]---
[    3.861748] Indeed it is in host mode hprt0 = 00001101
[    4.107257] usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 2.00
[    4.124696] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    4.146456] hub 1-1:1.0: USB hub found
[    4.158389] hub 1-1:1.0: 5 ports detected
[    4.294319] random: crng init done
</pre>
</details>

gtrainavicius, 2019-09-16:
Boots fine on 4B (4GB)

TiejunChina, 2019-09-16:
Could you apply this to try that again?

diff --git a/drivers/spi/spi-bcm2835aux.c b/drivers/spi/spi-bcm2835aux.c
index 3094d818cf06..4c18df321c82 100644
--- a/drivers/spi/spi-bcm2835aux.c
+++ b/drivers/spi/spi-bcm2835aux.c
@@ -489,7 +489,11 @@ static int bcm2835aux_spi_probe(struct platform_device *pdev)
 
        err = devm_request_irq(&pdev->dev, bs->irq,
                               bcm2835aux_spi_interrupt,
+#ifdef CONFIG_PREEMPT_RT_FULL
+                              IRQF_SHARED | IRQF_NO_THREAD,
+#else
                               IRQF_SHARED,
+#endif
                               dev_name(&pdev->dev), master);
        if (err) {
                dev_err(&pdev->dev, "could not request IRQ: %d\n", err);
diff --git a/drivers/tty/serial/8250/8250_bcm2835aux.c b/drivers/tty/serial/8250/8250_bcm2835aux.c
index aa7e7eb3019b..3b41e079a9bc 100644
--- a/drivers/tty/serial/8250/8250_bcm2835aux.c
+++ b/drivers/tty/serial/8250/8250_bcm2835aux.c
@@ -63,6 +63,10 @@ static int bcm2835aux_serial_probe(struct platform_device *pdev)
        }
        data->uart.port.irq = ret;
 
+#ifdef CONFIG_PREEMPT_RT_FULL
+       data->uart.port.irqflags = IRQF_NO_THREAD;
+#endif
+
        /* map the main registers */
        res = platform_get_resource(pdev, IORESOURCE_MEM, 0);
        if (!res) {
diff --git a/drivers/usb/host/dwc_otg/dwc_otg_hcd.c b/drivers/usb/host/dwc_otg/dwc_otg_hcd.c
index 2378a841ef17..9929f49fcbaa 100644
--- a/drivers/usb/host/dwc_otg/dwc_otg_hcd.c
+++ b/drivers/usb/host/dwc_otg/dwc_otg_hcd.c
@@ -1429,8 +1429,6 @@ static void assign_and_init_hc(dwc_otg_hcd_t * hcd, dwc_otg_qh_t * qh)
 
        dwc_otg_hc_init(hcd->core_if, hc);
 
-       local_irq_save(flags);
-
        if (fiq_enable)
                fiq_fsm_spin_lock_irqsave(&hcd->fiq_state->lock, flags);
        else

gtrainavicius, 2019-09-17:
I have applied the patch, you can check the [commit](https://github.com/gtrainavicius/linux/commit/e395f95b0806e1f9fff3584ee72d5df7914f4266) to see if it was done correctly.

Pi2 seems to get slightly further, one time it showed login prompt via serial, but it still hangs. Also in log 2, I see similar mmc errors, for the first time, to the ones that usually occur on 3B+.

<details><summary>2B log 1</summary>
<pre>
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 4.19.71-rt24-v7+ (giedrius@giedrius-vb) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03)) #1 SMP PREEMPT RT Tue Sep 17 12:10:23 EEST 2019
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] random: get_random_bytes called from start_kernel+0x7c/0x4ac with crng_init=0
[    0.000000] percpu: Embedded 18 pages/cpu s41536 r8192 d24000 u73728
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=8b59ba8e-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 938428K/970752K available (8192K kernel code, 629K rwdata, 2188K rodata, 1024K init, 957K bss, 24132K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 630 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 958 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26399 entries in 78 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU priority boosting: priority 1 delay 500 ms.
[    0.000000]  No expedited grace period (rcu_normal_after_boot).
[    0.000000]  Tasks RCU enabled.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000009] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000358] Console: colour dummy device 80x30
[    0.001622] console [tty1] enabled
[    0.001667] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001683] pid_max: default: 32768 minimum: 301
[    0.002166] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002185] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.003637] CPU: Testing write buffer coherency: ok
[    0.004422] CPU0: update cpu_capacity 1024
[    0.004658] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.060247] Setting up static identity map for 0x100000 - 0x10003c
[    0.100233] rcu: Hierarchical SRCU implementation.
[    0.180528] smp: Bringing up secondary CPUs ...
[    0.321567] CPU1: update cpu_capacity 1024
[    0.321576] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.461928] CPU2: update cpu_capacity 1024
[    0.461939] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.602286] CPU3: update cpu_capacity 1024
[    0.602295] CPU3: thread -1, cpu 3, socket 15, mpidr 80000f03
[    0.602561] smp: Brought up 1 node, 4 CPUs
[    0.602772] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.602815] CPU: All CPU(s) started in HYP mode.
[    0.602852] CPU: Virtualization extensions available.
[    0.604653] devtmpfs: initialized
[    0.644526] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.645503] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.645597] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.646663] pinctrl core: initialized pinctrl subsystem
[    0.649180] NET: Registered protocol family 16
[... 2175 lines cut ...]
[   84.444368] [853f1cb5] FDA< b8c6d8d0 0
[   84.449411] [853f1cb7] TCM< b8c6d9b0 ffffff92
[   84.455084] [853f1cb8] WTC< 10902 0
[   84.459904] mmc0: cmd op 17 arg 0x884e68 flags 0xb5 - resp 00000900 00000000 00000000 00000000, err 0
[   84.472085] mmc0: data blocks 1 blksz 200 - err -110
[... cut ...]
[   84.511646] mmc0: SDRSP1 0x00001133
[   84.516808] mmc0: SDRSP2 0x7fffffff
[   84.521967] mmc0: SDRSP3 0x00024037
[   84.527091] mmc0: SDHSTS 0x00000009
[   84.532179] mmc0: SDVDD  0x00000001
[   84.537242] mmc0: SDEDM  0x00010902
[   84.542287] mmc0: SDHCFG 0x0000040a
[   84.547315] mmc0: SDHBCT 0x000001fc
[   84.552311] mmc0: SDHBLC 0x00000001
[   84.557267] mmc0: ===========================================
[   84.564939] print_req_error: I/O error, dev mmcblk0, sector 8932968
[   84.573602] print_req
</pre>
</details>

gtrainavicius, 2019-09-17:
3B+ still hangs.

<details><summary>3B+ log 1</summary>
<pre>
▒[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.71-rt24-v7+ (giedrius@giedrius-vb) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03)) #1 SMP PREEMPT RT Tue Sep 17 12:10:23 EEST 2019
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] random: get_random_bytes called from start_kernel+0x7c/0x4ac with crng_init=0
[    0.000000] percpu: Embedded 18 pages/cpu s41536 r8192 d24000 u73728
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=8b59ba8e-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 938424K/970752K available (8192K kernel code, 629K rwdata, 2188K rodata, 1024K init, 957K bss, 24136K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 630 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 958 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26399 entries in 78 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU priority boosting: priority 1 delay 500 ms.
[    0.000000]  No expedited grace period (rcu_normal_after_boot).
[    0.000000]  Tasks RCU enabled.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000017] Switching to timer-based delay loop, resolution 52ns
[    0.000276] Console: colour dummy device 80x30
[    0.001398] console [tty1] enabled
[    0.001435] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001448] pid_max: default: 32768 minimum: 301
[    0.001829] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001843] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002936] CPU: Testing write buffer coherency: ok
[    0.003489] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.060148] Setting up static identity map for 0x100000 - 0x10003c
[    0.100131] rcu: Hierarchical SRCU implementation.
[    0.180321] smp: Bringing up secondary CPUs ...
[    0.321036] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.461270] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.601524] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.601737] smp: Brought up 1 node, 4 CPUs
[    0.601856] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.601893] CPU: All CPU(s) started in HYP mode.
[    0.601924] CPU: Virtualization extensions available.
[    0.603314] devtmpfs: initialized
[    0.636115] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.636814] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.636886] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.637652] pinctrl core: initialized pinctrl subsystem
[    0.639513] NET: Registered protocol family 16
[    0.645229] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.656384] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.656435] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.656685] Serial: AMBA PL011 UART driver
[    0.660382] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.729172] bcm2835-dma 3f007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
[... 1183 lines cut ...]
[   49.073211] [8288d573] FCM> b975dd24 0
[   49.078119] [8288d573] CMD  6 40
[   49.082485] [8288d574] REQ> b975dd24 0
[   49.087395] [832270e4] TIM< 0 0
[   49.091689] mmc0: cmd op 6 arg 0x80fffff1 flags 0xb5 - resp 00000900 00000000 00000000 00000000, err 0
[   49.103789] mmc0: data blocks 1 blksz 40 - err 0
[   49.109953] mmc0: =========== REGISTER DUMP ===========
[   49.116761] mmc0: SDCMD  0x00000046
[... cut ...]
[   49.136770] mmc0: SDRSP0 0x00000900
[   49.141738] mmc0: SDRSP1 0x0000066e
[   49.146703] mmc0: SDRSP2 0x7fffffff
[   49.151677] mmc0: SDRSP3 0x0001c01d
[   49.156643] mmc0: SDHSTS 0x00000001
[   49.161586] mmc0: SDVDD  0x00000001
[   49.166510] mmc0: SDEDM  0x00010901
[   49.171398] mmc0: SDHCFG 0x0000041a
[   49.176255] mmc0: SDHBCT 0x00000040
[   49.181092] mmc0: SDHBLC 0x00000001
[   49.185910] mmc0: ===========================================
[   49.193235] mmc0: tried to HW reset card, got error -110
</pre>
</details>

gtrainavicius, 2019-09-17:
And boots fine on 4B.

jrversteegh, 2019-09-21:
Same here. Doesn't boot on 3B+, patch doesn't help. With `dtoverlay=dwc2` it does boot.

pietervandermeer, 2019-09-23:
Previously the 4.19-rt branch worked just fine. Was the dwc_otg driver patch applied correctly?

Indeed dtoverlay=dwc2 fixes the problem for me as well. The issue with that is that you need a pulldown on USB_OTGID, which I don't have on my carrier board.

jrversteegh, 2019-09-23:
What dwc_otg driver patch?

pietervandermeer, 2019-09-23:
> What dwc_otg driver patch?

There used to be one to get the highly optimized dwc_otg driver working properly with the RT_PREEMPT patch. I don't know if it's around anymore. I was assuming this got automatically patch in the -rt branches, but not sure. 

But in any case, I just built the 4.14.y-rt branch and this works perfectly on the CM3, including everything USB!

JamesH65, 2019-12-20:
@gtrainavicius Is this still an issue?

pietervandermeer, 2019-12-22:
No, in November somebody fixed the 4.19-rt branch for pi3. I can confirm at least for CM3 that all USB issues are gone.

gtrainavicius, 2019-12-27:
I will check out latest builds for RPi 2, 3 and 4 in the coming week and report back.

gtrainavicius, 2019-12-27:
Ok, found some time to do this today :)

The results are pretty good - I built today version '4.19.71-rt24' for all 3 RPi configurations as suggested by https://www.raspberrypi.org/documentation/linux/kernel/building.md - it booted on 2B, 3B+ and 4B (4GB)

Only the 3B+ reports some kernel warning about an irq handler enabling interrupts:

```
[   14.416916] irq 166 handler irq_default_primary_handler+0x0/0x1c enabled interrupts
```

Just in case I will attach the boot logs from every boot I did.

[rpi2.log](https://github.com/raspberrypi/linux/files/4005335/rpi2.log)
[rpi3b+.log](https://github.com/raspberrypi/linux/files/4005336/rpi3b%2B.log)
[rpi3b+_2.log](https://github.com/raspberrypi/linux/files/4005337/rpi3b%2B_2.log)
[rpi4.log](https://github.com/raspberrypi/linux/files/4005338/rpi4.log)

Is the 3B+ warning something to be concerned about?

pietervandermeer, 2019-12-28:
I've been running 4.19.71-rt24-v7+ for a while now on CM3.. I thought everything was fine again, but all-in-all the scheduling isn't as spot-on as with 4.19.59-rt23-v7+ .. Occassionally there are weird scheduling mishaps, it seems. Deadlines are exceeded by a lot.. 20 ms or so. It's less bad than in the plain vanilla kernel, however. Perhaps I need to re-enable the dwc_otg fiq modes.. even if USB isn't used a lot in my application..

gtrainavicius, 2019-12-28:
Do you get the same warning in `dmesg` on CM3 as RPi3B+?

gtrainavicius, 2020-01-02:
@pietervandermeer ping :)

pietervandermeer, 2020-01-03:
@gtrainavicius will check the next time i try this kernel version.. i still have it prepped on my dev system somewhere.