raspberrypi/linux #3174: CVE-2018-19860 in Bluetooth firmware

lategoodbye, 2019-08-22:
**Is this the right place for my bug report?**
I hope it's okay to open this Bluetooth firmware issue here

**Describe the bug**
Bluetooth firmware does not properly restrict LMP commands and executes certain memory contents upon receiving an LMP command, as demonstrated by executing an HCI command. 

**To reproduce**
Please follow this project:
https://github.com/seemoo-lab/internalblue

**System**
* Which model of Raspberry Pi? Pi 3B, 3B+ and 4B

**Additional context**
https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2018-19860

According to this [bulletin](https://source.android.com/security/bulletin/2019-05-01) this issue has been fixed for Android, so there must be a fixed firmware.

satmandu, 2019-08-25:
Also more bluetooth issues which might result in updated firmware:
[https://knobattack.com/](https://knobattack.com/)
[https://nvd.nist.gov/vuln/detail/CVE-2019-9506](https://nvd.nist.gov/vuln/detail/CVE-2019-9506)

JamesH65, 2019-12-20:
@pelwell FYI