raspberrypi/linux #3176: libegl warning dri2 failed to authenticate raspberry pi 3

ping996, 2019-08-23:
libegl warning dri2 failed to authenticate raspberry pi 3, what's problem of this.

6by9, 2019-08-23:
When you create a new issue it gives you a template of the basic information that we require.
You have provided none of that information, therefore we can't really help at present.

What application are you running? On what kernel and firmware? Under X, or not under X?

I'd hazard a guess that you were trying to run a DRM/KMS application whilst X is running. DRM/KMS supports a single master and if X is running then it is the master. All other clients trying to render stuff will get an authentication failure.

ping996, 2019-08-26:
@6by9 ,
Hi, here are my detail information about the issue:
I write an applications about EGL +BCM to run the flutter on raspberry pi 3, the libegl warning was fixed by change some settings, But the new issue is `eglCreateContext()` failed with error code `EGL_BAD_ALLOC`, I have checked the code of this function, it was caused by `malloc()`:
`EGLContext _eglCreateContext(EGLDisplay dpy, EGLConfig config, EGLContext share_context, const EGLint *attrib_list)
{
	if (!attrib_list)
	{
		g_localStorage.error = EGL_BAD_PARAMETER;

		return EGL_FALSE;
	}

	if (g_localStorage.api == EGL_NONE)
	{
		g_localStorage.error = EGL_BAD_MATCH;

		return EGL_NO_CONTEXT;
	}

	EGLDisplayImpl* walkerDpy = g_localStorage.rootDpy;

	while (walkerDpy)
	{
		if ((EGLDisplay)walkerDpy == dpy)
		{
			if (!walkerDpy->initialized || walkerDpy->destroy)
			{
				g_localStorage.error = EGL_NOT_INITIALIZED;

				return EGL_FALSE;
			}

			EGLConfigImpl* walkerConfig = walkerDpy->rootConfig;

			while (walkerConfig)
			{
				if ((EGLConfig)walkerConfig == config)
				{
					EGLint target_attrib_list[CONTEXT_ATTRIB_LIST_SIZE];

					if (!__processAttribList(target_attrib_list, attrib_list, &g_localStorage.error))
					{
						return EGL_FALSE;
					}

					EGLContextImpl* sharedCtx = 0;

					if (share_context != EGL_NO_CONTEXT)
					{
						EGLDisplayImpl* sharedWalkerDpy = g_localStorage.rootDpy;

						while (sharedWalkerDpy)
						{
							EGLContextImpl* sharedWalkerCtx = sharedWalkerDpy->rootCtx;

							while (sharedWalkerCtx)
							{
								if ((EGLContext)sharedWalkerCtx == share_context)
								{
									if (!sharedWalkerCtx->initialized || sharedWalkerCtx->destroy)
									{
										g_localStorage.error = EGL_BAD_CONTEXT;

										return EGL_FALSE;
									}

									sharedCtx = sharedWalkerCtx;

									break;
								}

								sharedWalkerCtx = sharedWalkerCtx->next;
							}

							sharedWalkerDpy = sharedWalkerDpy->next;
						}

						if (!sharedCtx)
						{
							g_localStorage.error = EGL_BAD_CONTEXT;

							return EGL_FALSE;
						}
					}

					EGLContextImpl* newCtx = (EGLContextImpl*)malloc(sizeof(EGLContextImpl));

					if (!newCtx)
					{
						g_localStorage.error = EGL_BAD_ALLOC;     //malloc failed

						return EGL_FALSE;
					}

					// Move the atttibutes for later creation.
					memcpy(newCtx->attribList, target_attrib_list, CONTEXT_ATTRIB_LIST_SIZE * sizeof(EGLint));

					newCtx->initialized = EGL_TRUE;
					newCtx->destroy = EGL_FALSE;
					newCtx->configId = walkerConfig->configId;
					newCtx->sharedCtx = sharedCtx;
					newCtx->rootCtxList = 0;

					newCtx->next = walkerDpy->rootCtx;
					walkerDpy->rootCtx = newCtx;

					return (EGLContext)newCtx;
				}

				walkerConfig = walkerConfig->next;
			}

			g_localStorage.error = EGL_BAD_CONFIG;

			return EGL_NO_CONTEXT;
		}

		walkerDpy = walkerDpy->next;
	}

	g_localStorage.error = EGL_BAD_DISPLAY;

	return EGL_NO_CONTEXT;
}`

I checked the memory is available for new a memory.
The device info:
Linux raspberrypi 4.19.57-v7+ armv7l GNU/Linux.

ping996, 2019-08-28:
I reflash the image to my device, this error never happened again, I guss the EGL_BAD_ALLOC error caused by the solve wanring of `libEGL warning: DRI2: failed to authenticate`.

closed 2019-09-24 by ping996