raspberrypi/linux #3177: vmcs_sm.c:1208:3: error: implicit declaration of function 'outer_inv_range'

NguyenTrongThinh, 2019-08-23:
**Rebuild kernel for raspberry pi 3 B+ (64 bit)**
When re build the kernel for Raspberry PI 3 B+ 64 bit. I got these error 

/home/thinhnguyen/DATA/raspberry/build-rpi/tmp/work-shared/kumino/kernel-source/drivers/char/broadcom/vc_sm/vmcs_sm.c: In function 'vcsm_vma_fault':
/home/thinhnguyen/DATA/raspberry/build-rpi/tmp/work-shared/kumino/kernel-source/drivers/char/broadcom/vc_sm/**vmcs_sm.c:1208:3: error: implicit declaration of function 'outer_inv_range'; did you mean 'time_in_range'?** [-Werror=implicit-function-declaration]
   outer_inv_range(__pfn_to_phys(pfn),
   ^~~~~~~~~~~~~~~
   time_in_range
/home/thinhnguyen/DATA/raspberry/build-rpi/tmp/work-shared/kumino/kernel-source/drivers/char/broadcom/vc_sm/vmcs_sm.c: In function 'cache_op_to_func':
/home/thinhnguyen/DATA/raspberry/build-rpi/tmp/work-shared/kumino/kernel-source/drivers/char/broadcom/vc_sm/vmcs_sm.c:1268:10: **error: 'dmac_inv_range' undeclared (first use in this function); did you mean 'time_in_range'?**
   return dmac_inv_range;
          ^~~~~~~~~~~~~~
          time_in_range
/home/thinhnguyen/DATA/raspberry/build-rpi/tmp/work-shared/kumino/kernel-source/drivers/char/broadcom/vc_sm/vmcs_sm.c:1268:10: note: each undeclared identifier is reported only once for each function it appears in
/home/thinhnguyen/DATA/raspberry/build-rpi/tmp/work-shared/kumino/kernel-source/drivers/char/broadcom/vc_sm/vmcs_sm.c:1271:10: error: 'dmac_clean_range' undeclared (first use in this function); did you mean 'remap_pfn_range'?
   return dmac_clean_range;
          ^~~~~~~~~~~~~~~~
          remap_pfn_range
/home/thinhnguyen/DATA/raspberry/build-rpi/tmp/work-shared/kumino/kernel-source/drivers/char/broadcom/vc_sm/vmcs_sm.c:1274:10: error: 'dmac_flush_range' undeclared (first use in this function); did you mean 'proc_flush_task'?
   return dmac_flush_range;
          ^~~~~~~~~~~~~~~~
          proc_flush_task

**The Header file is not Available**
After searching in kernel source; I found that the header which describe these functions is not available in /arch/arm64/include/asm 

**To reproduce**
build 64 bit kernel with this config
CONFIG_BCM_VC_SM=y
 
**Expected behaviour**
Kernel buid without compile error

**Actual behaviour**
kernel report "undefine reference to some functions"

**System**
Raspberry Pin 3 b+
Host Machine: Ununtu 18.04

6by9, 2019-08-23:
BCM_VC_SM is not supported on 64 bit kernels. It has far too many assumptions over pointers being 32 bit.
This will not be fixed (I guess that means we ought to update the Kconfig dependencies to avoid selecting it)

There is a replacement driver under driver/staging/vc04_services/vc-sm-cma which allocates CMA buffers and maps them into the VPU heap. It is used by VIDEO_CODEC_BCM2835 (V4L2 M2M codec driver), and should therefore be in the default configs. This should be working on 64 bit builds now #3166 is merged.

NguyenTrongThinh, 2019-08-23:
Thanks! Build done

closed 2019-08-23 by NguyenTrongThinh