raspberrypi/linux #3180: RPi kernel with support for CONFIG_NET_SCH_CAKE

kucharskim, 2019-08-24:
On 4.19.66-v7+ kernel, I see CONFIG_NET_SCH_CAKE is not set.

CAKE (Common Applications Kept Enhanced) is a shaping-capable queue discipline which uses both AQM and FQ.

CAKE is new network packet scheduler and is enabled in 4.19 in Debian Buster. It's new in kernel 4.19, as it was merged in that version into official Linux sources, but was in development for longer. For example Turris Omnia with their custom kernel 4.4 has CAKE module available. Main advantage which I'm looking for with CAKE is autorate-ingress capability, where scheduler tries to optimize packet scheduling based on dynamic bandwidth of the link. I'm targeting use of Raspberry Pi and Raspian Buster with LTE modems, where bandwidth is dynamic. Buster is needed to have fresh enough iproute2 package.

Could you please add CONFIG_NET_SCH_CAKE=m to the kernel config to make available sch_cake.ko module?

6by9, 2019-08-24:
As requested in the previous issue (https://github.com/raspberrypi/linux/issues/3006#issuecomment-500340074), what are the implications of building this module? Is it only the one .ko module that is added, or are there additional dependencies?

wu-yue, 2019-09-15:
I guess only the .ko module is enough, as we have Buster now.

I tried CAKE on my Pi 2B 1.1 router myself.

It would boot fast in 1 or 2 mintues, as CAKE is an all-in-one style module. My old setup was hfsc + qfq + pie , which needs three modules, and the qfq module need a bash "for" statement to construct the structure, which would take about 5 minutes.

CAKE support DSCP diffserv, so Torrent boxes runs better.

CAKE has a TCP ACK filter in it. So on a link in which upload is slow but download is fast, CAKE would give more performance. This would also give a better wireless performance, because mostly, wireless AP have a better antenna  than clients.

JamesH65, 2019-12-20:
@pelwell As its a module, would adding this be a an issue?

pelwell, 2019-12-20:
An hour before taking an extended break is acknowledged to be the best time to make changes to software, so see https://github.com/raspberrypi/linux/commit/8408353fec8a07f322db1ab3e2960050a2c86946.

yutayu, 2020-01-22:
I am waiting for cake at raspbian ,too :)

kucharskim, 2020-01-22:
Seems to lib released to general public. Example from one of my RPi boxes, running official Raspbian Buster:

```
# uname -rv
4.19.93-v7+ #1290 SMP Fri Jan 10 16:39:50 GMT 2020

# find /lib/modules/ -name '*cake*' | sort
/lib/modules/4.19.93+/kernel/net/sched/sch_cake.ko
/lib/modules/4.19.93-v7+/kernel/net/sched/sch_cake.ko
/lib/modules/4.19.93-v7l+/kernel/net/sched/sch_cake.ko
/lib/modules/4.19.93-v8+/kernel/net/sched/sch_cake.ko
```

Thank you @pelwell for making this happen!

yutayu, 2020-01-22:
@kucharskim @pelwell 
Thank, I confirmed this on raspbian :)
Great and Happy.