raspberrypi/linux #3181: SND_BCM2835 audio not working on latest kernel version

NguyenTrongThinh, 2019-08-24:
**Describe the bug**
Build 64 bit kernel with these configuration 
CONFIG_SND_BCM2835=y
CONFIG_VIDEO_BCM2835=y
CONFIG_VIDEO_CODEC_BCM2835=y
but the on board audio (Jack 3.5) was not working, when run command $ aplay -l; it shows no audio card

**To reproduce**
Build 64 bit kernel with the configuration above

**Expected behaviour**
Audio on Jack 3.5 can work

**Actual behaviour**
No audio out and no sound card found when run command aplay -l

**System**
Raspberry Pi 3 B+
Host Machine: Ubuntu 18.04

6by9, 2019-08-24:
Which defconfig did you start from before seting those 3 modules as built in?

bcmrpi3_defconfig already has those drivers defined as being modules. Is there a specific need to make them built in? Did it work for you with them as modules?
Is there anything useful in the kernel log? Please provide the log.

NguyenTrongThinh, 2019-08-24:
bcmrpi3_defconfig already has those drivers defined as being modules.
=> Yes. default they are "m"

Is there a specific need to make them built in? 
=> No, I have changed them to test; because "m" not work and aplay -l shows very strange output. I am sure that I did not connect raspberry PI to any HDMI monitor; And No HDMI cable plugin
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Did it work for you with them as modules?
=> No, below are dmesg log and I also change the config to "m" in this log

[dmesg.txt](https://github.com/raspberrypi/linux/files/3537109/dmesg.txt)

[defconfig_rpi3b_plus.txt](https://github.com/raspberrypi/linux/files/3537105/defconfig_rpi3b_plus.txt)

sakaki-, 2019-08-24:
You appear to be using `vc4-kms-v3d` (in `/boot/config.txt`): does the problem still arise with `vc4-fkms-v3d`?

asavah, 2019-08-24:
Out of curiosity I tested this on my own build:

NguyenTrongThinh, 2019-08-25:
@sakaki- @asavah 
Thank you for your answer! It is correct.

So now the on board audio card appear but HDMI audio is still there. I am sure that no HDMI cable is plugged in

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
**card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]**
  Subdevices: 1/1
  Subdevice #0: subdevice #0
**card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]**
  Subdevices: 1/1
  Subdevice #0: subdevice #0

floion, 2019-08-29:
I confirm the same bug on 4.19.66. On 4.19.58 the bug is not present.
Using `vc4-fkms-v3d` instead of `vc4-kms-v3d` as @sakaki- and @asavah pointed out above, the problem is fixed on 4.19.66

6by9, 2019-08-29:
vc4-hdmi is the upstream audio driver provided as part of the full DRM/KMS driver (```vc4-kms-v3d```).
The HDMI peripheral only accepts IEC958 formatted data, hence your output
```
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 []
Subdevices: 1/1
Subdevice #0: subdevice #0
```
In theory you can get ALSA to do the relevant repacking of the data, but upstream never seemed to fully resolve the issues. See #2988 

Using snd_bcm2835 (```dtparam=audio=on```) will expose one audio device that can be switched between multiple outputs (audio, HDMI0, and HDMI1 (on a Pi4 only)), and also HDMI passthrough subdevices. The HDMI passthrough devices are not runtime checked - they will always be present.

lategoodbye, 2019-08-29:
There is no relation between vc4-kms-v3d and snd_bcm2835. But maybe there is some kind of regression in the VCHIQ driver which prevent probing of snd_bcm2835 in case vc4-fkms-v3d is not probed.

asavah, 2019-08-29:
@lategoodbye With no *kms overlay at all snd_bcm2835 is probed and works correctly.

floion, 2019-09-05:
@6by9 so is this considered a bug after all?

6by9, 2019-09-05:
64 bit kernels are in the early stages still. I haven't had a chance to look as most of my time is on Pi4 at present.

rpi-update does now include a 64bit kernel, admittedly aimed more at Pi4 than Pi3, so retesting with a standard kernel would be useful. See https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=250730 for more details.

lategoodbye, 2019-09-05:
Looking at the dmesg from @NguyenTrongThinh shows there is no attempt to load snd_bcm2835.
Can someone how is affected verify that the status of snd_bcm2835 devicetree node is really set to "okay"?

6by9, 2019-09-05:
I've rpi-updated by Pi3, added ```arm_64bit=1```, ```dtparam=audio=on```, and ```dtoverlay=vc4-kms-v3d``` to config.txt.
```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 1: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

- omxplayer plays audio happily, so the VPU can produce sound (bypasses alsa).
- VLC plays audio happily if directed at [bcm2835 ALSA] default device (it can't support the IEC958 format required by vc4-hdmi).
- ```aplay -v -D plug:iec958  foo.wav``` plays happily through vc4-hdmi, but having done that snd_bcm2835 is distorted, probably because the two drivers are now fighting for the audio part of the HDMI peripheral.

Disable vc4-hdmi audio (```dtoverlay=vc4-kms-v3d,audio=off```) with ```dtparam=audio=on```, and we have the full KMS driver with snd_bcm2835.

I fail to see any form of bug there. Back to those reporting it to give more details of what they are doing differently.

floion, 2019-09-05:
I have the following:

```
vcgencmd version
Nov  4 2018 16:31:07 
Copyright (c) 2012 Broadcom
version ed5baf9520a3c4ca82ba38594b898f0c0446da66 (clean) (release)
```

My config.txt is the following (I'm on 32 bits):
```
dtoverlay=vc4-kms-v3d,audio=off
dtparam=i2c_arm=on
dtparam=spi=on
dtparam=audio=on
disable_splash=1
avoid_warnings=1
```

and `aplay -l` is:
```
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

6by9, 2019-09-05:
> vcgencmd version
Nov  4 2018 16:31:07 

Why such an ancient firmware?
What kernel version? You previously mentioned 4.19.66 which was 9th Aug 2019. 4.19.69 is the latest on this branch.

And if that is what ```aplay -l``` is reporting, then the ```audio=off``` part of ```dtoverlay=vc4-kms-v3d``` is not working. Where have your device tree files come from? They obviously don't include https://github.com/raspberrypi/linux/commit/e8e84b725b2d9048f34c57c85893e4c35d94ea9e

floion, 2019-09-05:
Updated the kernel again to 4.19.66 (I had temporarily downgraded) and indeed, I do have that commit included. So now if I'm disabling hdmi audio as you suggested, I do have the devices listed by aplay and audio playback works as expected.

Thanks for all the input, much appreciated.

floion, 2019-09-06:
Sorry to revisit this, seems there is something strange going on. Can't make it work today anymore. Wondering if there was any obscure config.txt setting I had that made it work.

```
vcgencmd version
Aug 28 2019 15:08:27 
Copyright (c) 2012 Broadcom
version 31098ceed842ca10c8ffd02ffe3640bf6c0190b4 (clean) (release) (start)
```

```
uname -a
Linux 68d00bf 4.19.66 #1 SMP Thu Sep 5 21:09:51 UTC 2019 armv7l GNU/Linux
```

```
config.txt:

avoid_warnings=1
disable_splash=1
dtoverlay=vc4-kms-v3d,audio=off
dtparam=i2c_arm=on
dtparam=spi=on
dtparam=audio=on
gpu_mem=396
```

I do have https://github.com/raspberrypi/linux/commit/e8e84b725b2d9048f34c57c85893e4c35d94ea9e applied.

```
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

Wondering if somebody else can reproduce this

6by9, 2019-09-06:
If you have vc4-hdmi ALSA device then somewhere you haven't disabled audio. Check for the file /proc/device-tree/soc/hdmi@7e902000/dmas.

It should be 0 bytes in size if ```audio=off``` has applied correctly.

If still enabled then I'd expect it to be 8 bytes
```
pi@raspberrypi:~ $ xxd /proc/device-tree/soc/hdmi@7e902000/dmas
00000000: 0000 000a 0000 0011
```

For snd_bcm2835, check /proc/device-tree/soc/audio/status contains the string "okay" (as lategoodbye commented yesterday).

warpme, 2019-11-12:
Hi,
I'm using current rpi kernel [link](https://github.com/raspberrypi/linux/releases/tag/raspberrypi-kernel_1.20190925-1)) with mainline U-boot and no matter what I can't get sound on RPI3 as I'm always ending with vc4-hdmi is the upstream audio driver. I already tried [this](https://github.com/raspberrypi/linux/issues/3181#issuecomment-528778727) but I'm constantly getting  [this](https://github.com/raspberrypi/linux/issues/3181#issuecomment-528785815) + there is no audio in /proc/device-tree/soc
It looks like entries in config.txt are ignored...
May somebody hint me how to move forward with this?

pelwell, 2019-11-12:
Unless U-boot is configured to pass through the firmware-supplied DTB, the DTB it will be loaded from scratch by U-boot. This will lose any overlays and parameters set in config.txt.

warpme, 2019-11-12:
hmm,
here is what i done:

1. build my OS for armv7 arch 
2. built mainline u-boot with rpi_3_32b_defconfig
3. copy u-boot.bin to /boot 
4. add kernel=u-boot.bin to config.txt
5. prepare boot.scr with this:

```
fatload mmc 0:1 ${kernel_addr_r} kernel7.img
setenv bootargs earlyprintk dwc_otg.lpm_enable=0 console=serial0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
bootz ${kernel_addr_r} - ${fdt_addr}
```
6. copy boot.scr to /boot
7. use [kernel7.img](https://github.com/raspberrypi/firmware/raw/master/boot/kernel7.img) from here

powering-on rpi3 gives me u-boot initial u-boot log entries on screen and then all hangs on `"Loading kernel...."`

probably is still missing something :-(

pelwell, 2019-11-12:
The firmware treats U-boot as a kernel and passes it a pointer to the DTB it has constructed. If U-boot and boot.scr are correctly configured I believe the firmware's DTB can be passed on to the Linux kernel, but you are more likely to find help for your problem on a U-boot forum.

closed 2019-12-23 by NguyenTrongThinh