raspberrypi/linux #3183: issue with realtek 8188cu

jrversteegh, 2019-08-25:
The wlan0 device created for realtek rtl8188cu makes the pi hang when used as a hotspot. dmesg shows:
`
[ 2472.241418] usb 1-1.5: new high-speed USB device number 22 using dwc_otg
[ 2472.373244] usb 1-1.5: New USB device found, idVendor=0bda, idProduct=8176, bcdDevice= 2.00
[ 2472.373266] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2472.373278] usb 1-1.5: Product: 802.11n WLAN Adapter
[ 2472.373290] usb 1-1.5: Manufacturer: Realtek
[ 2472.373302] usb 1-1.5: SerialNumber: 00e04c000001
[ 2472.726049] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2473.408748] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2473.421670] ------------[ cut here ]------------
[ 2473.422249] WARNING: CPU: 0 PID: 1926 at net/wireless/sme.c:1060 __cfg80211_disconnected+0x4dc/0x530 [cfg80211]
[ 2473.422259] Modules linked in: 8192cu fuse rtc_ds1307 squashfs sg cdc_acm bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem v4l2_common bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common raspberrypi_hwmon hwmon snd_bcm2835(C) snd_pcm videodev snd_timer media snd vc_sm_cma(C) i2c_bcm2835 arc4 rt2800usb rt2800lib rt2x00usb rt2x00lib mac80211 sha256_generic cfg80211 uio_pdrv_genirq fixed uio rfkill crc_ccitt uas i2c_dev ip_tables x_tables ipv6
[ 2473.422534] CPU: 0 PID: 1926 Comm: kworker/u8:3 Tainted: G        WC        4.19.66-v7+ #1253
[ 2473.422540] Hardware name: BCM2835
[ 2473.422806] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[ 2473.422852] [<80111f38>] (unwind_backtrace) from [<8010d4b0>] (show_stack+0x20/0x24)
[ 2473.422875] [<8010d4b0>] (show_stack) from [<808191e0>] (dump_stack+0xd4/0x118)
[ 2473.422898] [<808191e0>] (dump_stack) from [<801209c8>] (__warn+0x104/0x11c)
[ 2473.422918] [<801209c8>] (__warn) from [<80120b18>] (warn_slowpath_null+0x50/0x58)
[ 2473.423173] [<80120b18>] (warn_slowpath_null) from [<7f1341c8>] (__cfg80211_disconnected+0x4dc/0x530 [cfg80211])
[ 2473.423636] [<7f1341c8>] (__cfg80211_disconnected [cfg80211]) from [<7f1033a0>] (cfg80211_process_wdev_events+0x13c/0x160 [cfg80211])
[ 2473.424094] [<7f1033a0>] (cfg80211_process_wdev_events [cfg80211]) from [<7f103404>] (cfg80211_process_rdev_events+0x40/0x98 [cfg80211])
[ 2473.424547] [<7f103404>] (cfg80211_process_rdev_events [cfg80211]) from [<7f0fd2d4>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[ 2473.424788] [<7f0fd2d4>] (cfg80211_event_work [cfg80211]) from [<8013bf0c>] (process_one_work+0x170/0x458)
[ 2473.424810] [<8013bf0c>] (process_one_work) from [<8013c250>] (worker_thread+0x5c/0x5a4)
[ 2473.424831] [<8013c250>] (worker_thread) from [<8014253c>] (kthread+0x138/0x168)
[ 2473.424852] [<8014253c>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[ 2473.424861] Exception stack(0xae2e1fb0 to 0xae2e1ff8)
[ 2473.424874] 1fa0:                                     00000000 00000000 00000000 00000000
[ 2473.424888] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2473.424900] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 2473.424911] ---[ end trace d6659d838156b427 ]---
[ 2473.527901] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2473.667966] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2473.681289] ------------[ cut here ]------------
[ 2473.681918] WARNING: CPU: 0 PID: 1926 at net/wireless/sme.c:1060 __cfg80211_disconnected+0x4dc/0x530 [cfg80211]
[ 2473.682105] Modules linked in: 8192cu fuse rtc_ds1307 squashfs sg cdc_acm bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem v4l2_common bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common raspberrypi_hwmon hwmon snd_bcm2835(C) snd_pcm videodev snd_timer media snd vc_sm_cma(C) i2c_bcm2835 arc4 rt2800usb rt2800lib rt2x00usb rt2x00lib mac80211 sha256_generic cfg80211 uio_pdrv_genirq fixed uio rfkill crc_ccitt uas i2c_dev ip_tables x_tables ipv6
[ 2473.682403] CPU: 0 PID: 1926 Comm: kworker/u8:3 Tainted: G        WC        4.19.66-v7+ #1253
[ 2473.682410] Hardware name: BCM2835
[ 2473.682767] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[ 2473.682816] [<80111f38>] (unwind_backtrace) from [<8010d4b0>] (show_stack+0x20/0x24)
[ 2473.682848] [<8010d4b0>] (show_stack) from [<808191e0>] (dump_stack+0xd4/0x118)
[ 2473.682872] [<808191e0>] (dump_stack) from [<801209c8>] (__warn+0x104/0x11c)
[ 2473.682892] [<801209c8>] (__warn) from [<80120b18>] (warn_slowpath_null+0x50/0x58)
[ 2473.683169] [<80120b18>] (warn_slowpath_null) from [<7f1341c8>] (__cfg80211_disconnected+0x4dc/0x530 [cfg80211])
[ 2473.683646] [<7f1341c8>] (__cfg80211_disconnected [cfg80211]) from [<7f1033a0>] (cfg80211_process_wdev_events+0x13c/0x160 [cfg80211])
[ 2473.684107] [<7f1033a0>] (cfg80211_process_wdev_events [cfg80211]) from [<7f103404>] (cfg80211_process_rdev_events+0x40/0x98 [cfg80211])
[ 2473.684594] [<7f103404>] (cfg80211_process_rdev_events [cfg80211]) from [<7f0fd2d4>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[ 2473.684838] [<7f0fd2d4>] (cfg80211_event_work [cfg80211]) from [<8013bf0c>] (process_one_work+0x170/0x458)
[ 2473.684868] [<8013bf0c>] (process_one_work) from [<8013c250>] (worker_thread+0x5c/0x5a4)
[ 2473.684889] [<8013c250>] (worker_thread) from [<8014253c>] (kthread+0x138/0x168)
[ 2473.684911] [<8014253c>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[ 2473.684920] Exception stack(0xae2e1fb0 to 0xae2e1ff8)
[ 2473.684933] 1fa0:                                     00000000 00000000 00000000 00000000
[ 2473.684946] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2473.684958] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 2473.685023] ---[ end trace d6659d838156b428 ]---
[ 2473.788524] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 2474.108133] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
`

JamesH65, 2019-12-20:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

jrversteegh, 2019-12-22:
Why would it be closed? I've had no response on the initial report?

JamesH65, 2019-12-23:
Have you tried everything in this thread? The realtek drivers in the kernel tree can be a bit shonky (these are upstream drivers - we do not have anything to do with them), you may need to go for out of tree drivers. 

https://www.raspberrypi.org/forums/viewtopic.php?f=36&t=250911

JamesH65, 2020-02-13:
@jrversteegh Did you have chance to examine the linked thread, and see if it fixed you problem?

jrversteegh, 2020-02-16:
I worked around the issue immediately by using a different driver. I just meant to report an issue for the 8188cu driver, but this is apparently not the right place.

closed 2020-02-16 by jrversteegh