raspberrypi/linux #3184: CM3+ / 2019-07-10-raspbian-buster-lite, can't setup Wi-fi

valtteric, 2019-08-27:
**Describe the bug**
Can't enable Wi-fi using CM3+ and an adapter.

**To reproduce**
1. Flash Official 2019-07-10-raspbian-buster-lite.
2. Connect Wi-fi adapter to the USB.
3. Start and try to enable Wi-fi using raspi-config.

Did also update & upgrade, no difference.

**Expected behaviour**
The same setup works with no problems on 2019-04-08-raspbian-stretch-lite. Same expected with buster.

**Actual behaviour**
Raspi-config prompts
`Could not communicate with wpa_supplicant`
`There was an error running option N2 Wi-fi`
Can't setup Wi-fi.

**System**
Hardware:
CM3+, "standard" development io-board (tested with Waveshare and Geppetto).
Tried with D-Link DWA-121 and Edimax EW-7811Un (apparently both are using RTL8188CUS chipset).

Software:
Official 2019-07-10-raspbian-buster-lite. 

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi Compute Module 3 Plus Rev 1.0

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2

* Which firmware version (`vcgencmd version`)?
(After upgrade, same situation than with after clean 2019-07-10-raspbian-buster-lite flash)
Aug 15 2019 12:06:42 
Copyright (c) 2012 Broadcom
version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start)

* Which kernel version (`uname -a`)?
(After upgrade, same situation than with after clean 2019-07-10-raspbian-buster-lite flash)
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

**Logs & Additional context**

- systemctl status wpa_supplicant
```
wpa_supplicant.service - WPA supplicant
   Loaded: loaded (/lib/systemd/system/wpa_supplicant.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-08-27 10:47:24 BST; 1min 33s ago
 Main PID: 424 (wpa_supplicant)
    Tasks: 1 (limit: 1604)
   Memory: 2.7M
   CGroup: /system.slice/wpa_supplicant.service
           └─424 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant

Aug 27 10:47:23 raspberrypi systemd[1]: Starting WPA supplicant...
Aug 27 10:47:24 raspberrypi wpa_supplicant[424]: Successfully initialized wpa_supplicant
Aug 27 10:47:24 raspberrypi systemd[1]: Started WPA supplicant.
```

- Tried to manually add a valid wpa_supplicant.conf to /etc/wpa_supplicant/ didn't make any difference.

- iwconfig
```
wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```
- Nothing obvious seen on dmesg, but not sure what to look for.
https://pastebin.com/ZpZDigWw

GeorgeSapkin, 2019-08-30:
This solution (downgrading wpasupplicant) worked for me: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=244731#p1498661

JamesH65, 2020-02-13:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2020-02-13 by JamesH65

rsaralegui, 2020-03-15:
Hello, please reopen this issue, at least to publish this comment. I had a similar issue, Wi-Fi not working after update from stretch to buster. I realised that in this version "predictable network interface names" are used, but wicd network manager still pointed to the old ethernet interface. After changing the old name "eth0" to the new one (enxb827eb8 etc. etc), Wireless started working again. No idea why the wrong reference to ethernet had an effect in this...