raspberrypi/linux #3194: 4.14.y softirq fix missing in 4.19.y branch

astir13, 2019-08-30:
**Is this the right place for my bug report?**
The softirq.c file is a part of the kernel and therefore this is the right place.

**Describe the bug**
The patch https://github.com/raspberrypi/linux/commit/3eb86ff32eb54c4345b723ae8dd03bd7487d35bd done by Linus is missing on the 4.19.y branch, leading to the (earlier) described issues with dropouts in USB audio and others, see https://github.com/raspberrypi/linux/issues/2215 and https://github.com/raspberrypi/linux/commit/3eb86ff32eb54c4345b723ae8dd03bd7487d35bd

**To reproduce**
See https://github.com/raspberrypi/linux/issues/2215

**Expected behaviour**
No droupouts in USB Audio with external USB soundcard such as UMC202HD or UMC404HD.

**Actual behaviour**
see https://github.com/raspberrypi/linux/issues/2215

**System**
see https://github.com/raspberrypi/linux/issues/2215;

for me:
* Pi3B+
* raspbian 
* Which firmware version (`vcgencmd version`)?
Aug 15 2019 12:06:42 
Copyright (c) 2012 Broadcom
version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux xxxxxx 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

lategoodbye, 2019-08-30:
The patch is there, but has a different hash 3c53776e29f81719efcf8f7a6e30cdf753bee94d.

astir13, 2019-08-30:
You proofed me wrong.

closed 2019-08-30 by astir13