raspberrypi/linux #3195: Ethernet sometimes fails to start (no carrier) on Raspberry Pi 4 on kernel 4.19.66-v7l+

ralphrmartin, 2019-08-31:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Add a clear and concise description of what you think the bug is.

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

ralphrmartin, 2019-08-31:
This has only started happening recently. My Pi4 was working fine as a server for quite a time until recently (I got it more or less at the time they were announced). It was connected to a BT Smart Hub for some time, and then has been connected to the EdgeRouter for at least a month. It has been restarted quite a few times since then (we get frequent power outages here), but its only in the last few days I have started seeing the problem.

Possible causes:
- failing hardware?
- something changed in recent kernel builds? 
- a change in other software involved in booting?
- a timing problem caused by me changing some other configuration?
- marginal power supply (but I am using an official one, and my SSD / Keyboard are connected via a powered USB3 hub)?

heziegl, 2019-09-01:
Having the same problem (same bcmgenet kernel log messages) with latest kernel 4.19.69-v7l+

pelwell, 2019-09-03:
These messages are expected and not a sign of an error:
```
Aug 31 10:47:32 www kernel: bcmgenet fd580000.genet: failed to get enet clock
Aug 31 10:47:32 www kernel: bcmgenet fd580000.genet: GENET 5.0 EPHY: 0x0000
Aug 31 10:47:32 www kernel: bcmgenet fd580000.genet: failed to get enet-wol clock
Aug 31 10:47:32 www kernel: bcmgenet fd580000.genet: failed to get enet-eee clock
Aug 31 10:47:32 www kernel: bcmgenet: Skipping UMAC reset
```
There is a known (but not yet understood) problem, seemingly triggered by resetting the Ethernet interface during the boot process, that leaves the interface in a state where packets can be received but nothing is transmitted (although the stats make it look as though the transmissions occur). The "Skipping UMAC reset" message is an indication that a workaround is active. You can disable the workaround by adding the following to cmdline.txt (as part of the single line - don't start another line):
```
genet.skip_umac_reset=n
```

hansj10, 2019-09-19:
I'm hitting the same issue, HPE ProCurve Switch and H3C.
somehow I'm still struggling to get one of the workarounds to work. 

Booted from buster-lite install media (7/10) and some some updates later, eth0 fails as described.
Kernel 4.19.66-v7l+

hansj10, 2019-09-19:
my cmdline.txt:
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait plymouth.enable=0 genet.force_reneg=y genet.skip_umac_reset=n

still the same issue..

kraj, 2020-03-13:
seeing same issue with 4.19.108-v8 and suggested workaround does not fix it

```
[    0.445211] bcmgenet fd580000.genet: failed to get enet clock
[    0.445233] bcmgenet fd580000.genet: GENET 5.0 EPHY: 0x0000
[    0.445254] bcmgenet fd580000.genet: failed to get enet-wol clock
[    0.445274] bcmgenet fd580000.genet: failed to get enet-eee clock
[    0.456334] libphy: bcmgenet MII bus: probed
[    7.212833] bcmgenet fd580000.genet: configuring instance for external RGMII (no delay)
[    8.224466] bcmgenet fd580000.genet eth0: Link is Down
[   12.320500] bcmgenet fd580000.genet eth0: Link is Up - 1Gbps/Full - flow control rx/tx

```

pelwell, 2020-03-13:
```
[   12.320500] bcmgenet fd580000.genet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
```
I don't understand - the log shows that there is a link. You might get one quicker without the `genet.skip_umac_reset=n`.

kraj, 2020-03-13:
@pelwell I am looking at those `failed` messages are they harmless ?

pelwell, 2020-03-13:
Yes, they're harmless - the driver can manage perfectly happily without those clocks.

ytakio, 2020-07-25:
I've had the same problem at every time when I tried to transfer a big file in local network via NFS (v4).
My RPi4 is booted by networkboot and nfsroot, so system freezes after link is down.
It's fatal even if down time is short.

Are there any workaround?

- Here are some log in last of dmesg.
```
[   38.231494] vcc-sd: disabling
[  431.112612] bcmgenet fd580000.ethernet eth0: Link is Down
[  434.232685] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
```

- Kernel version
```
$ uname -a
Linux raspberrypi 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux
```

marchom, 2020-07-28:
I'm experiencing this on latest 5.4.51-l7 kernel.

marchom, 2020-07-30:
I wonder if it's the same issue I'm having.
I get the same messages in the kernel log.
When the device restarts I am able to ping it, and it responds fine, but no other traffic goes in or out. I'm using a static IP.
I figure ICMP must be working ok, but no TCP/UDP.

Everything starts working again if I unplug and plug back in the Ethernet. Link goes down and up and it works again.

marchom, 2020-07-30:
Found out my issue was actually #3404 , #3108

djlambert, 2020-08-17:
Fairly certain I'm having the same issue. No NIC after boot on both Raspberry Pi OS and Fedora 32. If I reboot it comes up the second time. I'm going to be running it headless with PoE though, and the NIC never comes up after power cycling. More than happy to collect any debug information.

marchom, 2020-09-06:
As a workaround I've added:
`
@reboot (sleep 5; /sbin/mii-tool -r eth0)&
`
to `sudo crontab -e`

djlambert, 2020-09-09:
After installing the latest Fedora 32 updates yesterday (including kernel 5.8.6-201) mine now reboots after:
```
bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
bcmgenet fd580000.ethernet eth0: Link is down
```
Never gets past that point, just keeps rebooting.

SkyeFire, 2021-01-25:
I'm encountering an almost identical issue, albeit with a different root cause.  And so far, this is the only thread I've found anywhere in all my Googling that even approaches the issue I have.  Hence, I'm risking a bit of threadcromancy.

I'm running a RasPi 4 with the current version of RasPiOS:
```
pi@raspberrypi:/ $ uname -a
Linux raspberrypi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux
```
And I've set eth0 to a static IP address using /etc/dhcpcd.conf:
```
pi@raspberrypi:/ $ sudo cat /etc/dhcpcd.conf
# A sample configuration for dhcpcd.
# See dhcpcd.conf(5) for details.

# Allow users of this group to interact with dhcpcd via the control socket.
#controlgroup wheel

# Inform the DHCP server of our hostname for DDNS.
hostname

# Use the hardware address of the interface for the Client ID.
clientid
# or
# Use the same DUID + IAID as set in DHCPv6 for DHCPv4 ClientID as per RFC4361.
# Some non-RFC compliant DHCP servers do not reply with this set.
# In this case, comment out duid and enable clientid above.
#duid

# Persist interface configuration when dhcpcd exits.
persistent

# Rapid commit support.
# Safe to enable by default because it requires the equivalent option set
# on the server to actually work.
option rapid_commit

# A list of options to request from the DHCP server.
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
# Respect the network MTU. This is applied to DHCP routes.
option interface_mtu

# Most distributions have NTP support.
#option ntp_servers

# A ServerID is required by RFC2131.
require dhcp_server_identifier

# Generate SLAAC address using the Hardware Address of the interface
#slaac hwaddr
# OR generate Stable Private IPv6 Addresses based from the DUID
slaac private

# Example static IP configuration:
interface eth0
static ip_address=192.168.1.10/24
#static ip6_address=fd51:42f8:caae:d92e::ff/64
#static routers=192.168.0.1
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1

# It is possible to fall back to a static IP if DHCP fails:
# define static profile
#profile static_eth0
#static ip_address=192.168.1.23/24
#static routers=192.168.1.1
#static domain_name_servers=192.168.1.1

# fallback to static profile on eth0
#interface eth0
#fallback static_eth0
```
This works perfectly, with no issues, across multiple reboots.  _Until_ I start running my CodeSys project.  For those who don't know it, CodeSys is a software-based industrial PLC control system that can run on a regular PC, and has an "educational use only" version for the Pi.

When I start my CodeSys executable, within about 60sec eth0 will "die" -- dmesg shows an "eth0 down" message, and running ifconfig shows eth0 with no IP address.  Status check shows eth0 "waiting for carrier."  This happens even when I have an active connection running ssh over eth0.

If I reboot the Pi after this, eth0 never comes up -- dmesg simply reports "eth0 down."  It also shows the same string of bcmgenet messages that appear in the OP.  My dhcpcd.conf is unchanged -- it still has my static IP set.  Using ifconfig up/down commands on eth0 do nothing -- no messages, no feedback, just... nothing.

So far, the only way I've found to recover from this is to re-burn RaspiOS from the original download image, and re-do my dhcpcd.conf changes.  This works, right up until I install CodeSys and start running it (only running it seems to kill eth0 -- just installing the binary and downloading the executable project seems to be harmless).

I've eliminated hardware issues (except perhaps on the Pi itself) by swapping ethernet cables and switches.   I've also tried re-burning the OS and deliberately _not_ performing any updates.  I also tried rebuilding my Codesys project from scratch without _any_ drivers that access Eth0, on the theory that the Ethernet/IP or ModBus drivers might be causing this issue -- no dice there, either.

I tried adding the "skipreset=no" to my cmdline.txt as suggested upthread.  No change, dmesg just shows 'eth0 down' as the only eth0-related message:
`[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:8D:A1:92 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=249298d6-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles genet.skip_umac_reset=n`

```
pi@raspberrypi:~ $ dmesg | grep bcmg
[    1.033665] bcmgenet fd580000.ethernet: failed to get enet clock
[    1.033685] bcmgenet fd580000.ethernet: GENET 5.0 EPHY: 0x0000
[    1.033709] bcmgenet fd580000.ethernet: failed to get enet-wol clock
[    1.033730] bcmgenet fd580000.ethernet: failed to get enet-eee clock
[    1.044648] libphy: bcmgenet MII bus: probed
[    9.528502] bcmgenet fd580000.ethernet: configuring instance for external RGMII
[    9.535175] bcmgenet fd580000.ethernet eth0: Link is Down
```

jonas-schievink, 2021-02-22:
I'm also seeing this issue on kernel version [20201201](https://github.com/raspberrypi/linux/releases/tag/raspberrypi-kernel_1.20201201-1) (5.4.79) with an 8 GB Raspberry Pi 4, while my older Pi 4 with 1 GB RAM works flawlessly (unfortunately I really need more memory than that).

```
good:

Hardware        : BCM2835
Revision        : a03111
Serial          : 10000000d7aa48f5
Model           : Raspberry Pi 4 Model B Rev 1.1

bad:

Hardware        : BCM2835
Revision        : d03114
Serial          : 100000001b3b9f2d
Model           : Raspberry Pi 4 Model B Rev 1.4
```

No combination of workaround flags fixes the issue for me, the most usable configuration I found was with `genet.skip_umac_reset=n genet.force_reneg=y`, which *sometimes* allows network to stay up for a few minutes at best, until it results in the kernel warning below:

[image]

Is there anything else I could try here or should I just RMA the device?

SkyeFire, 2021-02-25:
Well, to follow up on my own issue two posts up, I was in contact with CodeSys tech support, who said that they were able to re-create the problem.  Their recommendation was to update to the latest version of CodeSys, and that _does_ appear to have resolved whatever the problem was.  I took the same Pi4, built a "clean" RasPi OS SD card, did a complete "clean" install of the latest patch of CodeSys, and tried the same actions that killed eth0 previously.  Got several hours of run time today without any eth0 hiccups.

So it would appear that my specific issue was rooted in CodeSys.

However:  in the intervening weeks, I picked up _another_ Pi4 (same 8GB model), and installed the "older" version of CodeSys on it _without_ any eth0 issues.  So, two supposedly-identical Pis, running identical builds, one with the eth0 issue, and one without.  I have to wonder if the "bad" Pi had some earlier hardware revision that something in the unpatched CodeSys was tripping over.

SkyeFire, 2021-03-02:
Well, I need to backtrack -- it turns out, it wasn't the CodeSys version -- it was the GPIO.

There are two "versions" of the GPIO package available in CodeSys, one for older Pis, and one for newer.  The default is for older Pis.  Not knowing this, I was using the outdated version for my Pi4.  And it _worked_, mostly.  But one thing it also did was give me configuration control (Input vs Output vs Not Used) for GPIOs 28 and 29.

I'm not entirely sure what CodeSys does for controlling GPIO pins at the driver level, but after some very painstaking trial-and-error testing, I found that setting GPIO 28 or 29 to Output would kill eth0, as described upthread, _even without ever **setting** either of those pins_.  Just setting the configuration was enough.  I failed to catch this for some time because I was _also_ trying to use Ethernet/IP, ProfiNet, and ModBus on the Pi, and thought that _those_ were the problem, since they use eth0 heavily.  I was only using the GPIOs (at that point) for some blinking LEDs, and (rather foolishly) just set _all_ the GPIOs to Output, even the ones not broken out on the header.

After looking at the Broadcom docs for the SoC, I noticed that 28 and 29 are the only pins within the range CodeSys had access to that had "primary" I/O mode, only the "alt" modes.  I'm assuming this has something to do with the behavior I encountered.

jonas-schievink, 2021-03-02:
My issue reproduces with a freshly downloaded version of the official Raspberry Pi OS too, so it is probably the original issue reported here.

@pelwell is there anything else I can try to help you debug this issue or should I return my hardware?

pelwell, 2021-03-03:
@SkyeFire On Pi4B, GPIOs 28 and 29 are used for the MDIO link to the Ethernet PHY - Alt function 5. I'm not surprised that interfering with them kills your Ethernet.

@jonas-schievink Looking at your screenshot above I see that your WiFi interface is also complaining. Multiple, independent problems like this is usually caused by undervoltage - how are you powering your 4B, and what other equipment do you have attached? If the power supply (including any cabling) is a bit marginal then some Pis will suffer more than others - there are small variations between instances of the same chip that affect the maximum clock speed and power consumption.

jonas-schievink, 2021-03-03:
I was powering it with the official power supply, with nothing plugged in except a microSD, TV, Ethernet, and a USB keyboard

pelwell, 2021-03-03:
I think you have reasonable grounds to RMA it.

tacuna, 2021-03-15:
(did no yet try any of the workarounds, yet)

**In case it helps**, Raspberry Pi 4 v1.2 4GB (eeprom 000138a1). Official power supply.

I'm suffering from the same issue, journal/dmesg is full of:
```
systemd-networkd[230]: eth0: Lost carrier
kernel: bcmgenet fd580000.ethernet eth0: Link is Down
systemd-networkd[230]: eth0: DHCPv6 lease lost
kernel: bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
systemd-networkd[230]: eth0: Gained carrier
```
happens every few minutes (inconsistent interval), the LAN port of the switch (D-link) you can see the lights go off (and back on after 2-3 sec). It happens for as long as the Pi is powered on/running.

(tried resetting router, switch, changed cable; no dice)

**Started** 6 Feb 2021 in the middle of the night (Pi was running since the 3th Feb) (before there were no eth0 issues what so ever, device is from mid 2020), at that time running: `Linux version 5.10.13-3-ARCH (builduser@leming) (aarch64-unknown-linux-gnu-gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35) #1 SMP PREEMPT Fri Feb 5 14:00:49 MST 2021` and `Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:74:8C:0F vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  root=PARTUUID=45a68f79-02 rw rootwait console=ttyAMA0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=ttyAMA0,115200 fsck.repair=yes mitigations=off`. From then onwards it happens every few minutes; **very annoying**.

Using a static IP (`/etc/systemd/network/eth0.network`).

Not using any GPIO connected devices at the moment (i tried some I2C sensors but disabled I2C again. No effect).

**Current**: `5.10.22-3-ARCH #1 SMP PREEMPT Fri Mar 12 16:15:46 MST 2021 aarch64 GNU/Linux` (Arch Linux ([linux-raspberrpy4 kernel](https://archlinuxarm.org/packages/aarch64/linux-raspberrypi4)))