raspberrypi/linux #3198: Some audio overlays do not apply to device tree

kwohlfahrt, 2019-08-31:
**Describe the bug**
Some overlays for audio devices fail to apply to any Raspberry Pi device tree. The `justboom-dac` and `hifiberry-dacplus` for example fail, while the `hifiberry-dac` applies.

**To reproduce**
This applies to both the binary (`raspberrypi/firmware`) and source repositories. To reproduce in the fimrware repository:

```
cd boot/
fdtoverlay -i bcm2710-rpi-3-b.dtb -o test.dtb overlays/justboom-dac.dtbo
```

To reproduce in this repository, the `.dts` files must first be compiled into `.dtb`:
```
cd arch/arm/boot/dts/
cpp -nostdinc -undef -isystem ../../../../include/ -x assembler-with-cpp bcm2710-rpi-3-b.dts | grep -v \^# > bcm2710-rpi-3-b.dts.tmp
dtc -@ -I dts -O dtb -o bcm2710-rpi-3-b.dtb bcm2710-rpi-3-b.dts.tmp
dtc -@ -I dts -O dtb -o overlays/justboom-dac.dtbo overlays/justboom-dac-overlay.dts
```

Then, apply the overlay as above.

**Expected behaviour**
I expect `test.dtb` to be produced, with no errors.

**Actual behaviour**
`fdtoverlay` exits with an error code, and prints the following:

```
Failed to apply overlays/justboom-dac-overlay.dtbo (-1)
```

**System**

This issue is regarding compilation of a device-tree overlay with `dtc`/`fdtoverlay`, not the system on which it is running. It occurs with both versions `1.4.7` and `1.5.0` of `dtc`.

I intend to use the resulting `dtb` file on a Raspberry Pi 3B, but haven't yet got that far.

kwohlfahrt, 2019-09-01:
I think I need to apply `i2s-gpio28-31` before the audio overlays. At least with this the complete device-tree builds, now to see whether this is enough for the hardware to work.

pelwell, 2019-09-01:
Even with "Verbose messages" enabled, the diagnostic output from `fdtoverlay` is woeful. Use `dtmerge` (installed by default on Raspbian) instead and a) you'll get more meaaningful error messages, and b) it will work.

kwohlfahrt, 2019-09-02:
`dtmerge` does apply the overlay successfully, and as far as I can tell the appropriate changes are made to my device tree. The hardware doesn't work (yet) but I think the problem lies elsewhere now.

Do you know why `fdtoverlay` doesn't work on these inputs? No worries if not, I'll try and puzzle it out.

Apologies for all the spam regarding commits above, I didn't realize GitHub would scrape that link.

pelwell, 2019-09-02:
I don't know the reason `fdtoverlay` fails, but if I were to investigate the problem I'd either rebuild the tool from source with debugging enabled and step through it, or delete bits of the overlay until it applies, reducing the size of the deleted parts until the problem is pinpointed.

pelwell, 2019-09-03:
The problem with the justboom-dac overlay (to pick one) is the label on the `__overlay__` sub-node of `fragment@2`. The [commit](https://github.com/dgibson/dtc/commit/b99353474850969a1f856d344d273325ddb069c7) in the dtc repo that fixes the problem explains it better than I could:
```
fdtoverlay: Allow adding labels to __overlay__ nodes in overlays

When applying overlays, we merge symbols from the overlay into the target
tree.  At the moment the logic for this assumes all symbols in the overlay
are attached to a node of the form:
	/fragment@XXX/__overlay__/relative/path

And will end up applied to the relative/path node under the fragment's
target.

However, this disallows the case of a symbol in the form just:
	/fragment@XXX/__overlay__

This does have a pretty obvious sensible meaning: attach the new symbol
directly to the fragment's target, but we don't currently do that.

It's pretty easy to workaround this limitation in one's overlays, but it's
also easy to handle in the overlay applying code, so we might as well
extend it to cover this case.

Reported-by: Christophe Braillon

```
Building the dtc utilities is simply a matter of cloning the repo and typing `make`, but you'd still be left with the issue that error messages are going to be lacking. Also, `fdtoverlay` merges overlay symbols into the base tree (dtmerge and the RPi firmware treat overlay symbols as being local to the overlay), which could be a problem.

pelwell, 2019-09-06:
Given that the failure is caused by syntax that is required by the overlays in question, and that the issue is fixed in the current dtc source, I'm closing this.

closed 2019-09-06 by pelwell

flokli, 2020-12-27:
@pelwell that patch never made it into https://git.kernel.org/pub/scm/utils/dtc/dtc.git/log/, did it?