raspberrypi/linux #3201: system reports ARMv7 even though CPU is ARMv8

tessus, 2019-09-02:
**Describe the bug**
I installed Raspbian Buster Lite on my sd card and put it into my Raspberry Pi 4 Model B.

`uname -a` reports the following:

```
Linux rasp4 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l GNU/Linux
```
and `cat /proc/cpuinfo` the following:
```
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: c03111
Serial		: 100000001e20b2d7
```

This is rather strange, since my Raspberry 4 has a 64-bit quad-core Cortex-A72 processor, so the output should say ARMv8 and uname should show armv8.

**To reproduce**
- buy raspberry pi 4 model B
- install Raspbian Buster Lite release date 2019-07-10
- run `uname -a` and `cat /proc/cpuinfo`

**Expected behaviour**
I should see that my cpu is in fact a 64-bit CPU. 

**Actual behaviour**
It shows that my CPU is 32-bit and the wrong processor.

**System**
```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2

Linux rasp4 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l GNU/Linux
Hardware	: BCM2835
Revision	: c03111
Serial		: 100000001e20b2d7
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Aug 15 2019 12:03:51
Copyright (c) 2012 Broadcom
version 9f8431fb7839c7f00f52b81f5822ddab2b31d0db (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0
```

**Additional context**
I really hope that this issue belongs here. But since the kernel is responsible for retrieving the proper CPU info, I think it does. Unless they sent me the wrong Raspberry PI in the packaging that says `Raspberry Pi 4 Computer Model B 4GB RAM`.

P33M, 2019-09-02:
This isn't a bug. The reported architecture is ARMv7 because the default is to boot in AARCH32 mode.

There is a community-supported 64-bit experimental kernel build here: https://www.raspberrypi.org/forums/viewtopic.php?p=1507435#p1507435

closed 2019-09-02 by P33M

tessus, 2019-09-02:
hmm, I am currently not looking for a 64-bit image, unless a Lite one is available. I only need a headless system and furthermore the 64-bit image can only access 1GB of memory, which defies the purpose of a 64-bit image. 

The issue is still with the 32-bit image. I understand that the kernel is 32-bit, so `uname -a` might be ok showing arm7. However the `cat /proc/cpuinfo` is still broken. Even on a 32-bit kernel the correct info should be shown.

But it even shows `Hardware	: BCM2835` which is basically the CPU of a Raspberry Pi Zero.

@P33M IMO this issue should not be closed. Unless you think it is normal that a 32-bit kernel reports false hardware. As said, it's ok to show a 32-bit kernel, but not 32-bit hardware.

JamesH65, 2019-09-02:
This was a decision by the upstream kernel developers, not Raspberry Pi. It's been like this forever. You should use the revision code to determine the exact model of Pi in use. See https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md

tessus, 2019-09-02:
@JamesH65 thanks for the info. that explains it. (However, just to be clear - it is a bug. Just a "wont-fix" bug. Showing the wrong info - for whatever reason - is a bug.)

Please add a reference to that page in the [Raspbian](https://www.raspberrypi.org/documentation/raspbian/) section. Someone (like me) who is new to Raspberry and its preferred OS does not understand that information is wrongly presented. I would have never thought to look into the hardware section to get the necessary info.

The sentence: _As of the 4.9 kernel, all Pis report BCM2835, even those with BCM2836, BCM2837 and BCM2711 processors. You should not use this string to detect the processor._ should also include that the rest of the cpuinfo is wrong.

pelwell, 2019-09-02:
If it's a bug (and it isn't), it should be reported to the Linux devs upstream.

Anyway, I hope we've now cleared up any confusion about the capabilities of your Pi 4.

JamesH65, 2019-09-02:
Also in the FAQ https://www.raspberrypi.org/documentation/faqs/#pi-software, and at the end of very many Google searches.

tessus, 2019-09-02:
@pelwell you have a strange notion what a bug is. Just because it is accepted to show false information does not mean it is not a bug. If `df` were to show 50% less free space, it wouldn't be a bug? Interesting.

The issue template states the following:
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place.

So what is it? Shall kernel-related issues be opened here or upstream?

tessus, 2019-09-02:
@JamesH65 thx, I missed the entry in the FAQ.

pelwell, 2019-09-02:
> Just because it is accepted to show false information does not mean it is not a bug.

Show me the specification of what cpuinfo returns and we can decide if it is a bug or not. Or show me a 32-bit kernel built for another ARM-based platform that reports the CPU as being 64-bit capable. The tool didn't meet your expectations, but that doesn't mean it isn't doing what it was intended to do.

> Shall kernel-related issues be opened here or upstream?

This is the right place to _report_ kernel issues. Sometimes they end up being upstream problems, in which case we tell users to report it there. Nobody has said you were wrong to report it here, just that the blame doesn't lie here.

It's not that complicated.

talvdav, 2020-04-10:
> Show me the specification of what cpuinfo returns and we can decide if it is a bug or not. Or show me a 32-bit kernel built for another ARM-based platform that reports the CPU as being 64-bit [...]

It really is not that complicated.

https://en.wikipedia.org/wiki/ARM_Cortex-A72 

The wiki article states that the architecture of the ARM Cortex-A72 is ARMv8-A
but lscpu shows it as armv7l so it is a bug! unless of course lscpu is not to be expected to show the correct information of the cpu... then it is a feature... i guess...

JamesH65, 2020-04-11:
Did you even read this thread? If running a 32 bit kernel, it's will be reported as 32bit - because the processor is running as a 32bit device.

And even if this were a bug, I believe it would be an upstream one, as that is where the decision is made.