raspberrypi/linux #3204: USB audio suddendly has dropouts with 4.19.66

astir13, 2019-09-04:
**Is this the right place for my bug report?**
Changing the kernel version back to 4.14.98 makes the problem disappear. Therefore, and due to https://github.com/raspberrypi/linux/issues/2215 it seems probable that there is a kernel problem around softirq.c. This is a speculation.

**Describe the bug**
Audio generated by playing a wave file through an external USB soundcard (Behringer UMC202HD or UMC404HD) contains phase jumps, i.e. audible and measurable artefacts, where none should be.

**To reproduce**
List the steps required to reproduce the issue.
/1/ create raspberry pi 3bB+ with raspbian stretch or buster, but fully updated to kernel revision 4.19.66
/2/ create a wave file containing 5 minutes of continuous sinus tone at 997Hz with 32 bit floating point precision.
/3/ connect the UMC202HD to a Linux notebook, and use aplay command and arecord command to play and at the same time record the sinus for at least 5 minutes while having a loopback connection at the sound card from output to input. Ensure correct GAIN and OUTPUT adjustments to avoid Clipping the input signal.
/4/ analyze the recorded wave file with Audacity and observe that there are no dropout problems (quality of sound card is good, cabling is good).
/5/ do the same on the raspberry pi 3B+
/6/ analyze the recorded wav file with Audacity and observe that there are dropouts in the recorded audio. 
/7/ the dropouts can be found especially easy by using the spectrum view in Audacity, as a dropout generates a change in the spectrum
/8/ observe that there are up to 4 dropouts seen per 10 second interval, which makes the generated sinus output unusable for measurement purposes.

**Expected behaviour**
No dropouts shall occur with normal CPU load.

**Workaround**
How to get a setup that does not generate dropouts? Use raspbian strech, fully updated (to kernel 4.19.66) and then downgrade the kernel to 4.14.98 by use of the command `rpi-update a08ece3d48c3c40bf1b501772af9933249c11c5b`. This setup will not create any audible dropouts, at the same side conditions, with the exact same hardware.

**Actual behaviour**
It is not clear why that happens. It only can be suspected that the problem is related to the old issue https://github.com/raspberrypi/linux/issues/2215.

**System**
see above description.

**Logs**

**Additional context**

slowrunner, 2019-09-25:
Not just USB audio - onboard audio is showing similar distorted output (and ALSA end of stream cutout).

Thanks for the work-around.

AlloKatana, 2019-10-30:
We design audio boards using CM compute module 3+. We observe same issue. Kernel version 4.19 is creating problems with all DACs that are non native DSD. If we downgrade to Kernel 4.14 issue disappears

tvm1, 2019-11-26:
I first noticed this problem (with onboard audio) on our custom yocto build after switching from 4.14 to 4.19 kernel. Same thing is happening when i compare current 4.19 Raspbian image with older Raspbian image with 4.14. I'm somewhat surprised that this issue isn't getting more attention, because the degradation of audio output quality is quite significant.

pelwell, 2019-11-26:
If you have a problem with onboard audio then you'll need to create a new issue - this one's about USB sound cards.