raspberrypi/linux #3205: vc4-fkms: Add hotplug support

6by9, 2019-09-04:
Task tracking / proposal.
As noted in https://github.com/raspberrypi/firmware/issues/1243, routing HDMI audio sensibly is currently a bit messy due to FKMS hiding whether it is connector HDMI0 or HDMI1 into which a monitor is connected. What DRM/KMS sees as HDMI-1 could therefore be a single monitor plugged into either connector.

Bottom line is we can have two displays active at a time due to only two HVS channels being available (the third channel is dedicated to the transposer).
The supported use cases are
- HDMI0 & HDMI1
- HDMI0 & DSI1
- HDMI0 & DPI
- SDTV.

(DSI and DPI are handled in the same way and named DSI-1 via DRM/KMS.)

If we add HDMI hotplug control to FKMS, then we can always create the HDMI display outputs in a consistent way, and the connector status should do the magic within the frameworks to enable and disable the monitors at appropriate times.

JamesH65, 2019-12-20:
@6by9 ISTR hotplug was getting some love, did it get enough?

6by9, 2019-12-20:
No, still in progress.
I have the kernel side working fine, but setting up DispmanX is more tricky.