raspberrypi/linux #3206: Buggy legacy (non vc4-kms-v3d) graphics on kernel 4.19.66

floion, 2019-09-05:
**Describe the bug**
Running kernel 4.19.66 and not using the `vc4-kms-v3d` overlay makes previously apps not run as expected. For example, we are running wpe on rpi3 and on older 4.14 kernel, without using the vc4-kms-v3d overlay, and wpe ran as expected. On the 4.19 kernel, the same app in legacy mode (non-kms) yields the following error:

` failed to add service - already in use?`

Which means that there is a problem with using the VPU in the legacy mode?

**To reproduce**
List the steps required to reproduce the issue.

**Expected behavior**
wpe should run

**Actual behavior**
wpe errors out: `* failed to add service - already in use?`

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW

raspberry pi 3B+

* Which OS and version (`cat /etc/rpi-issue`)?

custom yocto OS

* Which firmware version (`vcgencmd version`)?

# vcgencmd version
Aug 28 2019 15:08:27 
Copyright (c) 2012 Broadcom
version 31098ceed842ca10c8ffd02ffe3640bf6c0190b4 (clean) (release) (start)

* Which kernel version (`uname -a`)?

4.19.66

JamesH65, 2019-09-05:
Does WPE use OpenVG/OpenGLES? I think that is the error you get when trying to use OpenVG, or perhaps even OpenGL/ES when it's not available, as is the case on the Pi4 in legacy mode. OpenVG is deprecated and OpenGL is now only available using the FKMS driver.

floion, 2019-09-05:
WPE is compiled against the proprietary broadcom egl and gles libraries.

floion, 2019-09-05:
So the legacy mode won't work anymore with the proprietary gl libs?

JamesH65, 2019-09-05:
No. The firmware support has been removed. Moving to the much more standardised Mesa/DRM/KMS system is the recommended approach, it provides much better support.

6by9, 2019-09-05:
Pi3B+ should be fine running against the firmware GLES drivers.

Do you have anything odd in /boot/config.txt?

6by9, 2019-09-05:
Where are you installing WPE from?
The closest Raspbian package is xwpe, which installs and runs fine on Buster (text based, so I'm surprised it uses GLES).

floion, 2019-09-05:
> Pi3B+ should be fine running against the firmware GLES drivers.
> 
> Do you have anything odd in /boot/config.txt?

```
disable_splash=1
dtoverlay=vc4-kms-v3d
dtparam=i2c_arm=on
dtparam=spi=on
dtparam=audio=on
enable_uart=1
gpu_mem=396
```

I am running wpe from here https://github.com/balena-io-projects/balena-wpe
Which compiles the rdk backend and uses the BCM_RPI sub-backend.  ( from https://github.com/balena-io-projects/balena-wpe/blob/master/base-image/meta-resin-wpe/conf/samples/local.conf.sample )

JamesH65, 2019-09-05:
Bit confused, you have the vc4-kms-v3d turned on in that config, but the report is for it turned off.

floion, 2019-09-05:
Yeah, sorry, I had added it in the meanwhile to test something out.

floion, 2019-09-05:
If it helps:
```
dmesg |grep -i vch
[    0.808843] vchiq: vchiq_init_state: slot_zero = (ptrval), is_master = 0
[    0.812644] vc_vchi_sm_init: failed to open VCHI service (-1)
[    7.188562] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[    7.295229] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    7.299666] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    7.336396] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[    7.400613] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
```

6by9, 2019-09-05:
Sorry, but I'm not messing around signing up with Balena to install their app.
Please test a simpler GLES application, such as hello_videocube from the userland repo (https://github.com/raspberrypi/userland/tree/master/host_applications/linux/apps/hello_pi/hello_videocube)

If vchiq is failing to initialise then something is very broken in your configuration. The fact that you are using a custom yocto OS makes it nearly impossible for us to debug.
Please test with a stock kernel and device tree, and preferably using Raspbian.

pelwell, 2019-09-05:
Posting an unfiltered dmesg log might show something useful.

floion, 2019-09-05:
https://pastebin.com/UuvEifk4

6by9, 2019-09-05:
That kernel log has vc4-kms-v3d
```
[    0.000000] cma: Reserved 256 MiB at 0x1ec00000
...
[    0.215591] simple-framebuffer 3e914000.framebuffer: framebuffer at 0x3e914000, 0x6d8c00 bytes, mapped to 0x(ptrval)
[    0.215610] simple-framebuffer 3e914000.framebuffer: format=a8r8g8b8, mode=1824x984x32, linelength=7296
[    0.245128] Console: switching to colour frame buffer device 228x61
[    0.273974] simple-framebuffer 3e914000.framebuffer: fb0: simplefb registered!
[    0.274546] simple-framebuffer 3ec88000.framebuffer: framebuffer at 0x3ec88000, 0x373800 bytes, mapped to 0x(ptrval)
[    0.274566] simple-framebuffer 3ec88000.framebuffer: format=r5g6b5, mode=1824x984x16, linelength=3648
[    0.274580] checking generic (3e914000 6d8c00) vs hw (3ec88000 373800)
[    0.274796] simple-framebuffer 3ec88000.framebuffer: fb1: simplefb registered!
...
[    7.851628] vc4_hdmi 3f902000.hdmi: 'dmas' DT property is missing or empty, no HDMI audio
[    7.851774] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops [vc4])
[    7.854838] vc4-drm soc:gpu: bound 3f806000.vec (ops vc4_vec_ops [vc4])
[    7.855075] vc4-drm soc:gpu: bound 3f004000.txp (ops vc4_txp_ops [vc4])
[    7.855193] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
[    7.855642] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops vc4_crtc_ops [vc4])
[    7.856038] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops vc4_crtc_ops [vc4])
[    7.856433] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops [vc4])
[    7.881189] vc4-drm soc:gpu: bound 3fc00000.v3d (ops vc4_v3d_ops [vc4])
```

Seeing that fb0 and fb1 are both loaded, I'm suspecting you've got uboot in there messing up your DT as diagnosed in #3139, or would have if vc4-kms didn't take over control of the screen from the firmware.

Full DT from the running system needed as something is being seriously messed with. 
```dtc -I fs -O dts -o foo.dts /proc/device-tree```

floion, 2019-09-05:
Sorry about the dmesg, this one is from a boot without the vc4-kms-v3d overlay:
https://pastebin.com/xtcPr08X

And yes, u-boot is in the bootchain indeed. And I get the same yellow tinted framebuffer.

And here it the device tree from the running board:
https://gist.github.com/floion/42fa0a30656b97ba8fc8d9cfacc73c14

floion, 2019-09-05:
Confirmed that blacklisting `bcm2708_fb` made the whole problem go away and wpe is working as expected. Wondering if I should go this way or actually disable the simplefb driver from u-boot instead.
Thanks a lot for the info and time, much appreciated.

6by9, 2019-09-05:
I suspect uboot is making significant assumptions about running with a mainline kernel, not the Pi downstream one.
Setting up a simplefb device by making the relevant framebuffer mailbox calls is therefore reasonable, but it seems to break a lot of things when run with a downstream kernel :-(

floion, 2019-09-05:
Thanks again @6by9

closed 2019-09-05 by floion