raspberrypi/linux #3208: brcm-pcie: dmabounce: could not allocate 33554432 byte DMA pool

wizeman, 2019-09-06:
**Is this the right place for my bug report?**
I believe so.

**Describe the bug**
According to #3144, the latest commit in the 4.19.y branch should support the Raspberry Pi 4 with the full 4G of RAM in 64-bit mode.

With both commits 656dbed5db34b21bc951c918b597a6fd00740059 and f91882d9208992e286bc174859d0661b74ed10e8, I can get 3G of RAM to work but with 4G I get error messages during kernel boot and USB doesn't seem to work.

**To reproduce**
Just boot the Raspberry Pi 4 in 64-bit mode with `total_mem=4096` set in the SD card's `config.txt`.

**Expected behaviour**
I expected everything to work.

**Actual behaviour**
I'm getting the following error messages when booting:
[image]

Also, USB doesn't seem to work (for example, if I insert/remove any device, no kernel messages are displayed).

However, USB works fine and no errors are displayed if I limit RAM to 3072 MB.

* Which model of Raspberry Pi? `Pi4B`
* Which OS and version? `NixOS 19.03 + patches to support the Pi4`
* Which firmware version? `Release 1.20190718`
* Which kernel version (`uname -a`)? `Linux xxxxx 4.19.69 #1-NixOS SMP Thu Jan 1 00:00:01 UTC 1970 aarch64 GNU/Linux`

**Logs**
See screenshot above.

**Additional context**
My config.txt (all of it gets written by NixOS itself, except for the last 2 lines which I added):
```
# U-Boot used to need this to work, regardless of whether UART is actually used or not.
# TODO: check when/if this can be removed.
enable_uart=1

# Prevent the firmware from smashing the framebuffer setup done by the mainline kernel
# when attempting to show low-voltage or overtemperature warnings.
avoid_warnings=1
# Boot in 64-bit mode.
arm_64bit=1
kernel=kernel.img
initramfs initrd followkernel
gpu_mem=32
total_mem=4096
```

I also tried to boot with `gpu_mem=32` commented out but the same behavior remained.

pelwell, 2019-09-06:
With `total_mem=3072`, what does `cat /proc/cmdline` show? My kernel command line includes `coherent_pool=1M` and `cma=256M`.

wizeman, 2019-09-06:
This is what mine showed:
`coherent_pool=1M 8250.nr_uarts=1 cma=64M bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:09:EB:F3 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  kaslr page_poison=1 l1tf=full,force mds=full,nosmt nohibernate earlyprintk console=tty1 root=PARTUUID=<MY-ROOT-PARTITION-UUID> rootwait cma=32M dwc_otg.lpm_enable=0 bcm2835_wdt.nowayout=1 consoleblank=0 loglevel=9 apparmor=1 security=apparmor init=<PATH-TO-NIXOS-INIT/init>`

Note that it shows `cma=32M` because I added this parameter to the sd card's `cmdline.txt` file (it was left over from my working Raspberry Pi 3 config).

I noticed that the bootloader seems to add `cma=64M` automatically so I simply disabled my custom `cma=32M` parameter and everything seems to work now!

Thanks!

closed 2019-09-06 by wizeman