raspberrypi/linux #3210: OOM issue on Pi 4 with Buster

rowanalex123, 2019-09-06:
**Describe the bug**
Writing to a USB3 SSD disk results in kernel out of memory reaper killing random processes 

**To reproduce**
Boot Pi 4 from SSD HDD on USB3
Run `sudo dd if=/dev/zero of=~/test.tmp bs=500K count=8024`
Run `dmesg`

**Expected behaviour**
No processes killed due to OOM reaper

**Actual behaviour**
`dmesg output
[222125.725243] Mem-Info:
[222125.725255] active_anon:8353 inactive_anon:8505 isolated_anon:0
                 active_file:18133 inactive_file:179255 isolated_file:84
                 unevictable:20 dirty:5438 writeback:393 unstable:0
                 slab_reclaimable:11578 slab_unreclaimable:12597
                 mapped:8368 shmem:8196 pagetables:698 bounce:0
                 free:605316 free_pcp:475 free_cma:47644
[222125.725262] Node 0 active_anon:33412kB inactive_anon:34020kB active_file:72532kB inactive_file:717020kB unevictable:80kB isolated(anon):0kB isolated(file):336kB mapped:33472kB dirty:21752kB writeback:1572kB shmem:32784kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
[222125.725272] DMA free:210020kB min:16384kB low:20480kB high:24576kB active_anon:0kB inactive_anon:0kB active_file:716kB inactive_file:244kB unevictable:0kB writepending:0kB present:434176kB managed:388116kB mlocked:0kB kernel_stack:1744kB pagetables:0kB bounce:0kB free_pcp:72kB local_pcp:0kB free_cma:190576kB
[222125.725276] lowmem_reserve[]: 0 0 3008 3008
[222125.725300] HighMem free:2211244kB min:512kB low:33016kB high:65520kB active_anon:33868kB inactive_anon:34788kB active_file:72316kB inactive_file:716976kB unevictable:80kB writepending:22908kB present:3080192kB managed:3080192kB mlocked:80kB kernel_stack:0kB pagetables:2792kB bounce:0kB free_pcp:1816kB local_pcp:252kB free_cma:0kB
[222125.725304] lowmem_reserve[]: 0 0 0 0
[222125.725323] DMA: 594*4kB (UHC) 583*8kB (UC) 315*16kB (UEHC) 148*32kB (UHC) 52*64kB (UEHC) 25*128kB (UEHC) 10*256kB (C) 6*512kB (C) 3*1024kB (C) 3*2048kB (C) 42*4096kB (C) = 210224kB
[222125.725408] HighMem: 1113*4kB (UM) 4974*8kB (UM) 5790*16kB (UM) 2307*32kB (UM) 240*64kB (UM) 10*128kB (UM) 1*256kB (M) 1*512kB (M) 85*1024kB (M) 48*2048kB (UM) 439*4096kB (M) = 2211604kB
[222125.725491] 205652 total pagecache pages
[222125.725503] 135 pages in swap cache
[222125.725507] Swap cache stats: add 609, delete 474, find 58/90
[222125.725511] Free swap  = 6954484kB
[222125.725515] Total swap = 6957052kB
[222125.725518] 878592 pages RAM
[222125.725522] 770048 pages HighMem/MovableOnly
[222125.725526] 11515 pages reserved
[222125.725529] 65536 pages cma reserved
[222125.725533] Tasks state (memory values in pages):
[222125.725537] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
[222125.725556] [    238]     0   238     4606      947    49152        1         -1000 systemd-udevd
[222125.725564] [    385]   100   385     5592     1327    57344        3             0 systemd-timesyn
[222125.725569] [    423]     0   423     1986      564    45056        1             0 cron
[222125.725575] [    425]     0   425     3276     1443    53248        0             0 systemd-logind
[222125.725580] [    431] 65534   431     1080      544    32768        4             0 thd
[222125.725586] [    440]   104   440     1715      968    36864        4          -900 dbus-daemon
[222125.725591] [    447]     0   447     2675     1012    45056        2             0 wpa_supplicant
[222125.725597] [    449]     0   449      923      175    28672        4             0 alsactl
[222125.725603] [    450]     0   450     6914       17    40960        3             0 rngd
[222125.725608] [    454]     0   454     6378     1004    53248        0             0 rsyslogd
[222125.725614] [    461]   108   461     1475      750    40960        0             0 avahi-daemon
[222125.725619] [    480]   108   480     1443       60    36864        3             0 avahi-daemon
[222125.725625] [    495]     0   495      740      502    28672       10             0 dhcpcd
[222125.725632] [    516]     0   516     2094      669    36864        2             0 vncserver-x11-s
[222125.725637] [    555]     0   555    10035     1802    73728        0             0 lightdm
[222125.725642] [    564]     0   564     2699     1400    45056        0         -1000 sshd
[222125.725647] [    565]     0   565    11426     2116    81920        3             0 polkitd
[222125.725652] [    576]     0   576     2740      813    45056        0             0 wpa_supplicant
[222125.725658] [    598]     0   598     1431      631    40960       10             0 login
[222125.725663] [    736]  1000   736     3699     1863    53248        0             0 systemd
[222125.725668] [    740]  1000   740     4231      424    61440       28             0 (sd-pam)
[222125.725674] [    768]  1000   768     1668      870    40960        0             0 dbus-daemon
[222125.725679] [    817]  1000   817    13157     1476    69632        0             0 gvfsd
[222125.725684] [    822]  1000   822    14192     1669    81920        0             0 gvfsd-fuse
[222125.725690] [    937]  1000   937    14005     1720    86016        0             0 gvfs-afc-volume
[222125.725695] [    946]  1000   946    10025     1264    69632        0             0 gvfs-goa-volume
[222125.725700] [    954]  1000   954    10031     1120    61440        0             0 gvfs-mtp-volume
[222125.725706] [    966]  1000   966    10423     1219    69632        0             0 gvfs-gphoto2-vo
[222125.725711] [   1045]  1000  1045    10895     1604    77824        0             0 gnome-keyring-d
[222125.725717] [   1085]  1000  1085     7775     1225    65536        0             0 gvfsd-metadata
[222125.725722] [   1115]  1000  1115     2601     1018    45056        1             0 zsh
[222125.725727] [   1155]     0  1155      535       31    28672        1             0 hciattach
[... 311 lines cut ...]
004247.053: brfs: File read: /mfs/sd/kernel7l.img
004247.118: Loading 'kernel7l.img' to 0x8000 size 0x55d880
004251.536: No kernel trailer - assuming DT-capable
004251.630: brfs: File read: 5625984 bytes
004263.090: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
004263.153: Loading 'bcm2711-rpi-4-b.dtb' to 0x565880 size 0x9e6f
004572.988: brfs: File read: 40559 bytes
004576.672: brfs: File read: /mfs/sd/config.txt
004577.407: dtparam: audio=on
004614.287: brfs: File read: 1747 bytes
004661.234: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d.dtbo
004718.727: Loaded overlay 'vc4-fkms-v3d'
006203.480: Device tree loaded to 0x1a7e5c00 (size 0xa3e4)
006206.128: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
008902.583: vchiq_core: vchiq_init_state: slot_zero = 0xca580000, is_master = 1
008906.030: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
008912.346: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x00000010
020783.801: brfs: File read: 1255 bytes
021485.401: TV service:host side not connected, dropping notification 0x00000008, 0x00000004, 0x00000000
1866353.078: TV service:host side not connected, dropping notification 0x00000001, 0x00000002, 0x00000000
1866407.776: TV service:host side not connected, dropping notification 0x00000002, 0x00000004, 0x00000000

dmesg log
---------
`

pelwell, 2019-09-06:
1. What does `lsusb` report?
2. How are you powering the SSD?
3. There is a new test firmware for the VL805 that seems to be very promising. You can download it here: https://drive.google.com/file/d/1nUFfKGI118bJ6DmiDm9euE6kWbOxbG7q/view?usp=sharing

rowanalex123, 2019-09-06:
The SSD is powered using the official 3A Pi 4 power supply. vl805_update_0137ab does not fix the issue. 

`lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge
Bus 002 Device 002: ID 1058:1140 Western Digital Technologies, Inc. My Book Essential (WDBACW)
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
'

After the firmware upgrade, I ran:
```
sudo ./vl805
VL805 FW version: 000137ab

sudo dd if=/dev/zero of=/home/pi/test.tmp bs=4M count=3824
```

dmesg output
```
 pi  ~/Downloads/vl805_update_0137a8  sudo dmesg
[  669.808484] device-mapper: ioctl: 4.39.0-ioctl (2018-04-03) initialised: dm-devel@redhat.com
[  669.996916] NET: Registered protocol family 38
[  670.013086] cryptd: max_cpu_qlen set to 1000
[  674.981739] EXT4-fs (dm-0): mounted filesystem with ordered data mode. Opts: (null)
[ 2690.825013] lxpanel invoked oom-killer: gfp_mask=0x7080c0(GFP_KERNEL_ACCOUNT|__GFP_ZERO), nodemask=(null), order=1, oom_score_adj=0
[ 2690.825022] lxpanel cpuset=/ mems_allowed=0
[ 2690.825040] CPU: 0 PID: 870 Comm: lxpanel Tainted: G         C        4.19.66-v7l+ #1253
[ 2690.825043] Hardware name: BCM2835
[ 2690.825071] [<c0212c7c>] (unwind_backtrace) from [<c020d49c>] (show_stack+0x20/0x24)
[ 2690.825082] [<c020d49c>] (show_stack) from [<c097f080>] (dump_stack+0xd4/0x118)
[ 2690.825095] [<c097f080>] (dump_stack) from [<c033f5b4>] (dump_header+0x80/0x250)
[ 2690.825104] [<c033f5b4>] (dump_header) from [<c033e92c>] (oom_kill_process+0x358/0x3a8)
[ 2690.825113] [<c033e92c>] (oom_kill_process) from [<c033f25c>] (out_of_memory+0x134/0x36c)
[ 2690.825122] [<c033f25c>] (out_of_memory) from [<c0345620>] (__alloc_pages_nodemask+0xfc0/0x1180)
[ 2690.825133] [<c0345620>] (__alloc_pages_nodemask) from [<c021f5d0>] (copy_process.part.5+0x1f4/0x1ad4)
[ 2690.825141] [<c021f5d0>] (copy_process.part.5) from [<c0221078>] (_do_fork+0xd8/0x438)
[ 2690.825146] [<c0221078>] (_do_fork) from [<c02214fc>] (sys_clone+0x34/0x3c)
[ 2690.825153] [<c02214fc>] (sys_clone) from [<c0201000>] (ret_fast_syscall+0x0/0x28)
[ 2690.825157] Exception stack(0xc64d3fa8 to 0xc64d3ff0)
[ 2690.825162] 3fa0:                   b57311d8 00000001 01200011 00000000 00000000 00000000
[ 2690.825168] 3fc0: b57311d8 00000001 b3d15a40 00000078 00000001 b6473000 b5731630 00000013
[ 2690.825171] 3fe0: b5731170 beec16c0 b63c8854 b63c8788
[ 2690.825175] Mem-Info:
[ 2690.825185] active_anon:24288 inactive_anon:31049 isolated_anon:1
                active_file:8763 inactive_file:360689 isolated_file:0
                unevictable:16 dirty:6541 writeback:628 unstable:0
                slab_reclaimable:13552 slab_unreclaimable:11228
                mapped:12179 shmem:19585 pagetables:1503 bounce:0
                free:393281 free_pcp:64 free_cma:46791
[ 2690.825194] Node 0 active_anon:97152kB inactive_anon:124196kB active_file:35052kB inactive_file:1442756kB unevictable:64kB isolated(anon):4kB isolated(file):0kB mapped:48716kB dirty:26164kB writeback:2512kB shmem:78340kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
[ 2690.825203] DMA free:203832kB min:16384kB low:20480kB high:24576kB active_anon:0kB inactive_anon:0kB active_file:3252kB inactive_file:344kB unevictable:0kB writepending:0kB present:434176kB managed:388116kB mlocked:0kB kernel_stack:3424kB pagetables:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:187164kB
[ 2690.825207] lowmem_reserve[]: 0 0 3008 3008
[ 2690.825232] HighMem free:1369292kB min:512kB low:33016kB high:65520kB active_anon:97300kB inactive_anon:124196kB active_file:31800kB inactive_file:1442292kB unevictable:64kB writepending:28704kB present:3080192kB managed:3080192kB mlocked:64kB kernel_stack:0kB pagetables:6012kB bounce:0kB free_pcp:384kB local_pcp:0kB free_cma:0kB
[ 2690.825235] lowmem_reserve[]: 0 0 0 0
[ 2690.825254] DMA: 317*4kB (UEC) 258*8kB (UE) 220*16kB (UEC) 102*32kB (UEH) 64*64kB (UE) 22*128kB (UEHC) 2*256kB (C) 2*512kB (C) 3*1024kB (C) 3*2048kB (C) 43*4096kB (C) = 203908kB
[ 2690.825336] HighMem: 4872*4kB (UM) 995*8kB (UM) 68*16kB (UM) 38*32kB (UM) 28*64kB (UM) 245*128kB (UM) 1453*256kB (UM) 580*512kB (UM) 149*1024kB (UM) 79*2048kB (UM) 79*4096kB (M) = 1369784kB
[ 2690.825417] 393633 total pagecache pages
[ 2690.825426] 4700 pages in swap cache
[ 2690.825430] Swap cache stats: add 84212, delete 79512, find 21231/29672
[ 2690.825434] Free swap  = 3944188kB
[ 2690.825437] Total swap = 4194300kB
[ 2690.825441] 878592 pages RAM
[ 2690.825445] 770048 pages HighMem/MovableOnly
[ 2690.825448] 11515 pages reserved
[ 2690.825451] 65536 pages cma reserved
[ 2690.825455] Tasks state (memory values in pages):
[ 2690.825459] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
[ 2690.825475] [    127]     0   127     5347      864   204800      202             0 systemd-journal
[... 568 lines cut ...]
[ 2986.631634] [   1105]  1000  1105     2601      261    45056      283             0 zsh
[ 2986.631639] [   1109]  1000  1109     7744       97    65536      113             0 gvfsd-metadata
[ 2986.631645] [   1116]     0  1116    12087      382   122880      452             0 smbd
[ 2986.631650] [   1130]     0  1130    11356      255   102400      458             0 smbd-notifyd
[... cut ...]
[ 2986.631733] [  19614]     0 19614   199847      171    49152        0          -999 containerd-shim
[ 2986.631738] [  19631]  9001 19631      352        0    28672       12             0 tini
[ 2986.631744] [  19954]     0 19954     2514      753    45056        0             0 sudo
[ 2986.631749] [  19959]     0 19959     2650     1284    49152        0             0 dd
[ 2986.631754] Out of memory: Kill process 610 (Xorg) score 5 or sacrifice child
[ 2986.631783] Killed process 610 (Xorg) total-vm:110164kB, anon-rss:14388kB, file-rss:25568kB, shmem-rss:576kB
```

pelwell, 2019-09-06:
Expecting the 4B to power an external SSD, even with a known-good power supply, is ambitious.

That particular USB<->SATA adaptor has been seen to cause problems.

yaroslavros, 2019-09-06:
@rowanalex123 Can you try removing `coherent_pool=1M` `cma=64M` `cma=256M` from your cmdline.txt?

wushilin, 2019-12-04:
On the USB-SATA adapter part - I disagree. I use Western Digital 8TB usb disk, and it has same issue.

randyoo, 2019-12-10:
@wushilin
I also had the same issue while running ZFS (and rsync), and it was gone after switching to the 64-bit kernel.

mandev, 2019-12-13:
Using the 64bit kernel did not solve the problem. At least for me.

balajisraghavan, 2019-12-22:
Tried multiple SATA USB controllers including the Asmedia ASM1153. Issue still persists. Dont think its a power supply issue as well. I'm using the official power supply with no peripherals (keyboard, mouse etc) attached. Should be more than enough power to run the SATA SSD drive.

JamesH65, 2020-02-14:
Please update to the latest release kernel and retry, please report results.

balajisraghavan, 2020-02-14:
What's changed James?

bkev, 2020-08-20:
I had this happen to my Pi last night too. I was doing a DD to a USB 3 hard drive and got the OOM errors in the syslog file.

Lipown, 2020-08-24:
Having same issue. Firstly caused by Transmission, then Deluge when downloading more torrents. Then just by coping files over network to PI. Tried to make huge swap without any solution. Now I am trying to disable oom killer by https://serverfault.com/questions/141988/avoid-linux-out-of-memory-application-teardown. It really driving me as I cannot use pi as a backup network storage. I have latest kernel and  Raspberry OS.

bkev, 2020-08-25:
I'm no expert, but it seems that it's just the data transfer that causes the issue, not the application then. I was just using dd to make a copy of the image to the USB 3 hard drive and I had the same issue, with OOM kicking in and killing all kinds of things in an attempt to get it to work.

I have a Pi v4 with 4Gb of RAM and I've never seen the amount used go over around 20% used.

I've not rebooted my pi since posting the memory above and it's now 

total        used        free      shared  buff/cache   available
Mem:           3776         483         165         235        3126        2970
Swap:            99          39          60

rowanalex123, 2020-08-25:
This issue is a definite deal breaker for anyone trying to use the Pi 4 as a desktop replacement or a media hub. Please look into this.

bkev, 2020-08-25:
@rowanalex123 So, are you still having the same issue, with the same things in the logs, even now?

I know that your level of detail in this post was what helped me figure out I was having the same, or a very similar issue.

I don't think it does it all the time for me, but last week whilst doing dd like you did, it slowed my Pi to a crawl and OOM starting closing all sorts of things down even though the memory seemed to be OK.

rowanalex123, 2020-08-25:
I've not tested it myself off late. Going by the above two reports though, think its safe to say the issue is not resolved with the latest kernel?

bkev, 2020-08-25:
I regularly run apt update and full-upgrade so I'm running the latest firmware, bootloader and kernal.

Lipown, 2020-08-25:
I also tried huge 80GB swap but it's not used at all. I would like to have this resolved. When I had the storage drive as NTFS and transfer speeds were <20MBs that time over network I had not this issue but now since I have storage on ext4 I have speeds ~100MBs it's crashing by oom killer.

pelwell, 2020-08-25:
Can people seeing the problem please post the output of the following commands:
```
$ grep Revision /proc/cpuinfo
$ uname -a
$ vmstat
```

pelwell, 2020-08-25:
And:
```
$ lsusb
```

bkev, 2020-08-25:
> I also tried huge 80GB swap but it's not used at all. I would like to have this resolved. When I had the storage drive as NTFS and transfer speeds were <20MBs that time over network I had not this issue but now since I have storage on ext4 I have speeds ~100MBs it's crashing by oom killer.

Actually this rings a bell with me. I have my external drive formatted as NTFS but I kept on getting all sorts of errors etc related to NTFS-3G, but I can't remember any issues with OOM. I reformatted the drive to ext4 and got the OOM error.

bkev, 2020-08-25:
> Can people seeing the problem please post the output of the following commands:
> 
> ```
> $ grep Revision /proc/cpuinfo
> $ uname -a
> $ vmstat
> ```

Hi, thanks for the info. Here are the results of the commands on my Pi:
Revision        : c03111

Linux raspberrypiv4 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

procs -----------memory---------- ---swap-- -----io---- -system-- ------cpu-----
 r  b   swpd   free   buff  cache   si   so    bi    bo   in   cs us sy id wa st
 0  0  41472 224508 130100 3006784    0    0     5    19    5   22  2  2 95  1  0

Bus 002 Device 002: ID 1058:2627 Western Digital Technologies, Inc.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 1e3d:4083 Chipsbank Microelectronics Co., Ltd
Bus 001 Device 004: ID 0781:5583 SanDisk Corp. Ultra Fit
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

pelwell, 2020-08-25:
That's a rev 1.1 4B with 4GB RAM running a recent kernel. The rev 1.1 boards where fitted with BCM2711B0 - the first production silicon - which had a few restrictions that were improved in the later C0 revision:
1. The EMMC2 bus can only directly address the first 1GB.
2. The PCIe interface can only directly address the first 3GB.

The latter restriction may be having an effect on the disk throughput. You can eliminate it as a possibility by (as an experiment!) rebooting with `total_mem=3072` in config.txt. See if that makes any difference.

andrum99, 2020-08-25:
How can I tell which silicon revision my BCM2711 is? Is it etched into the top of the package? All mine seem to be B0 - I'm guessing it's the B0 in the middle of this:

`2711ZPKFS**B0**6BOT`

Do the chips with C0 silicon say C0 instead?

pelwell, 2020-08-25:
It's the penultimate 2 characters that matter: 2711ZPKFSB06**CO**T

On a running system the easiest way to tell is to look at the declaration of the bus the SD card controller is on:
```
# B0
pi@raspberrypi:~$ od -An -tx1 /proc/device-tree/emmc2bus/dma-ranges
 00 00 00 00 c0 00 00 00 00 00 00 00 00 00 00 00
 40 00 00 00

# C0
pi@raspberrypi:~$ od -An -tx1 /proc/device-tree/emmc2bus/dma-ranges
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 fc 00 00 00
```

pelwell, 2020-08-25:
As far as I know, all production 8GB units are fitted with C0s.

bkev, 2020-08-25:
> That's a rev 1.1 4B with 4GB RAM running a recent kernel. The rev 1.1 boards where fitted with BCM2711B0 - the first production silicon - which had a few restrictions that were improved in the later [...]

Thanks for the info.

So, my board not only has the USB-C power issue, it also can't access all of the memory?

Will the total_mem=3072 limit the board to 3Gb instead of 4Gb of memory? 

If so, is there any kind of hardware replacement program for these boards if the issues can't be fixed in software without limiting the memory?

pelwell, 2020-08-25:
It's a test - I'm trying to ascertain whether or not SWIOTLB is a factor or not. Yes it will limit the system to 3GB _during the test_.

bkev, 2020-08-25:
Hi @pelwell thanks for the info.

My Pi will be doing a dd tomorrow night as part of a cron job. I'll let it do it without me changing anything to see if it occurs again. If it does, I'll make the change and then let it do the dd again to see if it happens again and I will report back. This will probably be the end of next week though. 

Thanks again.

Lipown, 2020-08-26:
Here is my output of the commands:

Revision        : c03111

Linux raspberrypi 5.4.59-v7l+ #1336 SMP Wed Aug 19 16:09:04 BST 2020 armv7l GNU/Linux

procs -----------memory---------- ---swap-- -----io---- -system-- ------cpu-----
 r  b   swpd   free   buff  cache   si   so    bi    bo   in   cs us sy id wa st
 1  0      0 738352  25536 1797696    0    0    11     8   12   76  2  2 95  1  0

Lipown, 2020-08-26:
Since I am impatient I tried
total_mem=3072 neither 2048 value solves the issue unfortunatelly :(

Lipown, 2020-08-26:
Something came up to my mind reacting on others person post on another forum. Me neither the person having those problems when coping small files. Maybe files larger than ram doing that? I am a noob so I am just guessing....

wushilin, 2020-08-26:
I didn't have this solved but bought another x64 based mini pc running
freebsd as Nas, it did work well.

I could have saved the 300 bucks if pi4 is okay. Worst thing is I bought
two of them...

bkev, 2020-08-26:
Could it be that with large file transfers that if there is an issue with the throughput to the usb hub, it starts to back up into the ram causing the oom? And that might be why it only affects large files? 

Just a complete stab in the dark.

Lipown, 2020-09-02:
Has anyone looked on this issue please? I am going to buy another 4 4GB but I need that the issue is no more on the latest revision.

bkev, 2020-09-04:
I had the issue occur again for me. 
The load on the Pi was at about 7 and it was quite unresponsive with errors in the log with I think are maybe related to this
 xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
 xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
 usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd

I have added quirks to my cmdline.txt now to see if it makes any difference:

https://www.raspberrypi.org/forums/viewtopic.php?t=245931#:~:text=The%20kernel%20can%20be%20told,needs%20to%20go%20into%20cmdline.

bkev, 2020-09-05:
The quirks above haven't helped me. I've started to get the xhci_hcd errors again less than 24 hours later.

Lipown, 2020-09-06:
Another log of oom killer just by using Chromium with google spreadsheet triggered this:

https://pastebin.com/WEEAh97X

The pi running few days straight. What is wrong :/

pelwell, 2020-09-06:
As well as chromium running a spreadsheet you also have Hassio, containerd, mysqld, and others - this isn't a vanilla system. And it was dockerd that triggered the OOM killer.

Lipown, 2020-09-07:
> As well as chromium running a spreadsheet you also have Hassio, containerd, mysqld, and others - this isn't a vanilla system. And it was dockerd that triggered the OOM killer.

Indeed, that's why I shut down the services after this and found I/O errors were caused by long USB Cables from my drive. But shutting down docker (and hassio), mysqld and all unnecessities I did 200 GB copy test and it ended up oom killing. The pi has 4GB of memory. I remember copying that with far less memory and did copied eight years ago similar amount of data from network to external drive connectted to Samsung tablet without issues. I believe all of that pi can make if another mini pc can...Nevetherless that 4 is currently most powerful...

rowanalex123, 2020-09-07:
The issue still exists for me. Can copy or download a 6gb file without the oom reaper kicking in. This is with the latest updates installed on the 4gb Pi 4

Lipown, 2020-09-07:
> The issue still exists for me. Can copy or download a 6gb file without the oom reaper kicking in. This is with the latest updates installed on the 4gb Pi 4

Can or can not?

I did again a test with log to Home Assistant. When copy from PI over network its perfectly fine with CPU usage about 25% percent. I can work on pi during that.
When I copying to PI processor go to 40%. Everything slows down. Just starting Chromium triggers the OOM killer, tha ram is about 40% used, then it crash totaly with login screen in few minutes,

balajisraghavan, 2020-09-08:
> Can or can not?

Sorry typo. Should have been `can not`. Writing a 6GB file seems to trigger the OOM reaper.

rowanalex123, 2020-09-08:
> Can or can not?

Sorry typo. Should have been `can not`. Writing a 6GB file seems to trigger the OOM reaper.

jozsefDevs, 2020-10-08:
Joining to the issue list. 

Revision	: c03112
Linux BreGu 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
procs -----------memory---------- ---swap-- -----io---- -system-- ------cpu-----
 r  b   swpd   free   buff  cache   si   so    bi    bo   in   cs us sy id wa st
 0  0    512 365132  13768 2964392    0    0    18     2   46   46  6  0 94  0  0

Interesting that I can NOT reproduce the issue by using the `dd` command to my external drive. I can write out 6 or 8GB file there without any issues. 

However if I start up `rtorrent` the issue is kicking in almost immediately. At the time when the issue is kicking in the free command shows the following (I started another terminal with `watch -n2 free -h`:

              total        used        free      shared  buff/cache   available
Mem:          3.3Gi       132Mi       356Mi        33Mi       2.8Gi       3.0Gi
Swap:         1.8Gi       0.0Ki       1.8Gi

So it's should NOT be a memory issue... but it seems it is somehow. It's really frustrating.

Lipown, 2020-10-22:
Anything done on this?

bkev, 2020-10-23:
I'm still getting the same. Every few days and it's so frustrating. For me, my Pi really slows down and the load goes up when I start seeing this in the log...

usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd

And it never recovers apart from having to do a reboot.

jozsefDevs, 2020-10-26:
Okay, so for me this issue is resolved. I just set `arm_64bit=1` into /boot/config.txt and my Pi runs since 8 days without any OOM reaping. Again, this is the Pi4 / 4GB version.

randyoo, 2020-10-26:
For what it's worth, I had a similar experience. Regular OOM reaper issues before switching to 64 bit kernel, and never happened since.

Lipown, 2020-10-26:
> Okay, so for me this issue is resolved. I just set `arm_64bit=1` into /boot/config.txt and my Pi runs since 8 days without any OOM reaping. Again, this is the Pi4 / 4GB version.

I did a test of copying of 150 GB into and outside the pi external USB drive without oom killer so I consider it success. How you came to this solution please?

jozsefDevs, 2020-10-28:
@Lipown: There is unfortunately no reasonable explanation from my side. I just browsed through all related topics from the [forums](https://www.raspberrypi.org/forums/) and did some trial-and-error tests.

balajisraghavan, 2020-10-28:
Using `arm_64bit=1` breaks vlc hardware h264 decoding. It's not a viable option for me until that's fixed. But, good to know we can address this issue in at least some cases.

vk5tu, 2020-11-01:
Phil, I have seen similar symptoms on another platform. With a 32-bit kernel a high rate of I/O lead to fragmentation of kernel lowmem (memory with a direct mapping to kernel address space, I think on ARM 32-bit this is a contiguous 512MB). That fragmentation lead to longer CPU and elapsed times to obtain memory for buffers, which is accounted towards the requesting process. That in turn lead to selection of the process by the OOM killer. Moving to a 64-bit kernel (with its massively greater lowmem) 'solved' our problem.  I haven't investigated your particular issue, but thought sharing this experience might be helpful. All the best, Glen.

bkev, 2020-11-02:
> Phil, I have seen similar symptoms on another platform. With a 32-bit kernel a high rate of I/O lead to fragmentation of kernel lowmem (memory with a direct mapping to kernel address space, I think on ARM 32-bit this is a contiguous 512MB). That fragmentation lead to longer CPU and elapsed times as the kernel worked to obtain memory for buffers, which is accounted towards the process doing the large amount of I/O. That in turn lead to selection of the process by the OOM killer. Moving to a 64-bit kernel (with its massively greater lowmem) 'solved' our problem. I haven't investigated your particular issue, but thought sharing this experience might be helpful. All the best, Glen.

That's interesting. I think my issues are initially caused by these errors:

xhci_hcd 0000:01:00.0: WARN Cannot submit Set TR Deq Ptr
xhci_hcd 0000:01:00.0: A Set TR Deq Ptr command is pending.
usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd

Which then impacts everything else as the system starts to cache more and more as it isn't able to write to the USB device for some reason. I still haven't figured out why this is. It will be fine for days and then all of a sudden get those errors and I don't know any way of recovering it other that rebooting the Pi. 

It's a shame that there isn't any kind of reset to the ports as I think this would fix it, but all the automated stuff the OS tries doesn't work!

I've swapped to 64 bit now too, just to see if this helps.

vovan-, 2020-11-09:
I can confirm I have similar issue with Raspberry Pi 4 8GB.
If I run deluged and download speed reaches smth higher than 5MB/s it will suddenly cause OOM killer to kill deluged or some other process. USB3 SATA SSD is connected
Enabled swap 1GB to SSD with no luck.

WIll give it a try with arm_64bit=1, do you know any issues with this, besides the one mentioned above regarding VLC hardware decoding?

rtsperry, 2020-11-15:
I have what appears to be a similar issue, so FWIW here are some slightly different data points. 

Scenario: prepping a new USB hard drive via command line luksFormat, luksOpen, and dd'ing /dev/zero until the drive cries 'No mas!' Encrypting a 4TB drive will take several days and hence it's perfect task for a Pi 4 4G I have on the network. I can connect via VNC to set things up and then occasionally check the progress on a display off to the side. 

This afternoon -- a few hours shy of the anticipated conclusion -- I glance over at the display to see the Raspbian OS login box. This can't be good, I think. But the VNC server is working fine and `lsblk -f` still shows the encrypted partition as previously mapped. What's going on? Checking the last few lines of /var/log/kern.log reveals the dreaded oom_reaper:
```
Nov 14 15:59:49 BlackBoxPi kernel: [471926.178931] [   5997]     0  5997     2487      804    36864        0             0 sudo
Nov 14 15:59:49 BlackBoxPi kernel: [471926.178938] [   6002]     0  6002     1882      630    40960        0             0 dd
Nov 14 15:59:49 BlackBoxPi kernel: [471926.178946] [   2829]     0  2829     2487      810    45056        0             0 sudo
Nov 14 15:59:49 BlackBoxPi kernel: [471926.178953] [   2830]     0  2830     2016      736    40960        0             0 htop
Nov 14 15:59:49 BlackBoxPi kernel: [471926.178963] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=Xorg,pid=545,uid=0
Nov 14 15:59:49 BlackBoxPi kernel: [471926.179006] Out of memory: Killed process 545 (Xorg) total-vm:147712kB, anon-rss:4612kB, file-rss:13852kB, shmem-rss:17212kB, UID:0 pgtables:296kB oom_score_adj:0
Nov 14 15:59:49 BlackBoxPi kernel: [471926.189788] oom_reaper: reaped process 545 (Xorg), now anon-rss:0kB, file-rss:8100kB, shmem-rss:17272kB
Nov 14 15:59:51 BlackBoxPi kernel: [471929.001990] broken atomic modeset userspace detected, disabling atomic
```
and there's a similar report at the end of dmesg:
```
[471926.178946] [   2829]     0  2829     2487      810    45056        0             0 sudo`
[471926.178953] [   2830]     0  2830     2016      736    40960        0             0 htop`
[471926.178963] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=Xorg,pid=545,uid=0
[471926.179006] Out of memory: Killed process 545 (Xorg) total-vm:147712kB, anon-rss:4612kB, file-rss:13852kB, shmem-rss:17212kB, UID:0 pgtables:296kB oom_score_adj:0
[471926.189788] oom_reaper: reaped process 545 (Xorg), now anon-rss:0kB, file-rss:8100kB, shmem-rss:17272kB
[471929.001990] broken atomic modeset userspace detected, disabling atomic
```
Hardware: known good supply, electronic load tested to 5.2 V @ 3.0 A. Only peripheral attached is a powered USB3 hub, which the new USB3 Toshiba drive is plugged into.
Software: Up to date OS and apps. Only application open is LXTerminal with tabs for dd and htop.

Per previous messages, here's the data dump:
```
Bus 002 Device 009: ID 0480:0820 Toshiba America Inc Canvio Advance Disk
Bus 002 Device 002: ID 05e3:0616 Genesys Logic, Inc. hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

od -An -tx1 /proc/device-tree/emmc2bus/dma-ranges:
 00 00 00 00 c0 00 00 00 00 00 00 00 00 00 00 00
 40 00 00 00

Revision	: c03111

Linux BlackBoxPi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux

procs -----------memory---------- ---swap-- -----io---- -system-- ------cpu-----
 r  b   swpd   free   buff  cache   si   so    bi    bo   in   cs us sy id wa st
 6  1  11008 3197780 167708 250456    0    0    18    21    0    7  0 14 76 10  0

              total        used        free      shared  buff/cache   available
Mem:        3919788      150556     3508824       82076      260408     3555096
Swap:        102396       11008       91388
```
If I get particularly curious/motivated I'll test things in 64bit mode; before reading this thread I simply punted by creating a new partition for the last part of the drive and repeating the previous procedure.

Lipown, 2020-11-15:
The 64 bit mode really works. I am finally running a NAS setup on PI
without issues.

rtsperry, 2020-11-26:
A follow-up (anecdotal) report on encrypting my 4T Toshiba hard drive with 32-bit software: I downloaded the latest 32-bit Ubuntu server image and performed the same procedure as described above on the same hardware and encrypted the 4T drive without incident. I next tried using a Pi 4 8G model configured to boot into 32-bit CLI mode. Although the Alt-F1 dd process concluded successfully, the htop instance on Alt-F2 fell victim to the oom_reaper. Finally, I tested things with my new Pi 400 and it didn't even make it half-way before aborting things with oom-killer messages. Reviewing the Pi 400's journal output I noticed it was a full half hour from the RNG reporting no problems to the OOM errors, with no messages between indicating problems.

My conclusion is the unremarkable observation that this isn't a hardware issue as originally postulated. Using the c03130 revision of the BCM2711 in my Pi 400 didn't fix things. I'd be happy to post the relevant journal messages if someone thinks it will help but at this point it doesn't seem a lack of journal logs is the issue. The problem is known, even if the timing of the OOM explosions is somewhat unpredictable. 

While I look forward to retesting things when the 64-bit version of Raspian OS makes it out of beta and into gamma I'm also aware that the developers have limited time and a long list of things to work on. Given a shift to 64-bit mode means I'm not going to have access to all of the Pi software ecosystem anyway my next step will be testing an Arch image for my NAS. It's my preferred OS in x86 land and I'm already familiar with it.

popcornmix, 2020-11-27:
> While I look forward to retesting things when the 64-bit version of Raspian OS makes it out of beta 

You don't need to move to 64-bit RpiOS to try this. Just add arm_64bit=1 to config.txt and reboot.

rtsperry, 2020-11-28:
Good point, popcornmix. Better is often the enemy of good enough. And it's true that adding arm_64bit=1 to config.txt eliminated oom messages from my logs. 

Alas, after a few hours dd froze up completely. htop shows one of the cores at 100%. And my log is littered with thousands of new complaints along the lines of:
```Nov 28 06:25:01 BlackBoxPi kernel: sdhci-iproc fe340000.emmc2: swiotlb buffer is full (sz: 8192 bytes), total 32768 (slots), used 27103 (slots)
Nov 28 06:25:01 BlackBoxPi kernel: sdhci-iproc fe340000.emmc2: overflow 0x00000001b258e000+8192 of DMA mask ffffffff bus limit ffffffff
```
A quick web search shows swiotlb problems in aarch64 are neither new nor unique to Pis. Given my firmware and kernel are current as of Oct 22 yet other users haven't been bitten by swiotlb, I don't know what to say except that my issues with large USB transfers continue. I don't want to clutter a thread with off topic complaints, so I guess I'm back to a strategy of awaiting further developments.

[47 more replies]