raspberrypi/linux #3213: On-board WiFi EAPOL key exchange not responding

jantheman, 2019-09-06:
Trying to connect to WPA-EAP Enterprise with phase2=MSCHAPV2 network using Raspberry Pi Zero W revision 1.1, image reference 2019-07-10. 

When I plug-in an external WiFi USB dongle, "RTL8188CUS 802.11n WLAN Adapter" everything works as expected, it connects to the network and receives IP address from DHCP and internet works.

With exactly the same configuration, the on-board WiFi connects, but does not receive any IP. If I use static IP address, I could not even ping the gateway. Here is a packet trace from bringing the interfaces up (I changed the MAC address using macchanger to match the external adapter to rule out any MAC filtering).:

[image]

Nisca_ is the Raspberry Pi, Cambridg_ is the AP. The right one contains the working connection attempt through external dongle. Starting with packet 28, EAPOL key exchange is initiated, exchanging 4 key messages and 2 group key messages. The built-in non-working one on the left side has the same connecting trace, the first key message from AP being packet 28, but the Raspberry Pi never responds to that message. The AP tries a couple of times and then gives up.

I am left with no other options than blaming the wireless driver. 

**System**

* Which model of Raspberry Pi? PiZeroW 1.1
* Which OS and version (`cat /etc/rpi-issue`)? 2019-07-10 ...9a4a56c, stage2
* Which firmware version (`vcgencmd version`)? 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start)
* Which kernel version (`uname -a`)? 4.19.66+ #1253 Aug 15

lategoodbye, 2019-09-07:
Please provide your dmesg, so we can check this is related to the following [wpa_supplicant issue](http://lists.infradead.org/pipermail/hostap/2019-June/040288.html).

mitch-davenport, 2019-09-16:
Same issue on Raspberry Pi 3 B+ using MSChapv2. Same hardware on 4.14.79-v7+ image works fine with PEAP. WLAN Controller shows it passes radius auth but doesn't respond so I believe it's failing step 3 of EAP process. This one has no issue connecting a PSK network.
[dmesg.log](https://github.com/raspberrypi/linux/files/3618191/dmesg.log)

Which model of Raspberry Pi? Pi 3 B+
Which OS and version (cat /etc/rpi-issue)? 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2
Which firmware version (vcgencmd version)? version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start)
Which kernel version (uname -a)? Linux raspberrypi 4.19.66-v7+ 
Attaching the dmesg log.

Here's my supplicant.
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
        ssid="SecureNet"
        key_mgmt=WPA-EAP
        identity="piuser"
        password="pass"
        eap=PEAP
        phase1="peaplabel=0"
        phase2="auth-MSCHAPV2"
}

lategoodbye, 2019-09-17:
According to the dmesg this looks like the mentioned wpa_supplicant 2.8 regression. So in this case it isn't a kernel issue.

Try to use another wpa_supplicant version (2.7 or 2.9).

jantheman, 2019-09-24:
Any luck @mitch-davenport with newer wpa_supplicant?

I confirm I had 2.8 too, I tried 2.9, but it got worse. Previously, it exchanged certificates at least, but with 2.9, it won't even associate. wpa_cli writes:
```
CTRL-EVENT-SCAN-STARTED
CTRL-EVENT-SCAN-RESULTS
Trying to associate with SSID 'mynetwork'
CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code 16
... few repeats ...
CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid='mynetwork' auth_failures=4 duration=77 reason=CONN_FAILED
```

When I use the external wifi dongle, it still works with the same config:
```
CTRL-EVENT-SCAN-STARTED
CTRL-EVENT-SCAN-RESULTS
SME: Trying to authenticate with 9c:50:ee:xx:xx:xx (SSID='mynetwork', freq=2312 MHz)
Trying to authenticate with 9c:50:ee:xx:xx:xx (SSID='mynetwork', freq=2312 MHz)
Associated with 9c:50:ee:xx:xx:xx
CTRL-EVENT-EAP-STARTED EAP authentication started
...
```

(by the way, the wifi dongle completely hangs the pi, requiring power cycling)