raspberrypi/linux #3214: Unable to reboot Raspberry Pi 3 B+ with kernel 64 bit

NguyenTrongThinh, 2019-09-07:
**Describe the bug**
Raspberry Pi 3b+ cannot reboot when I use command "$ reboot"
I have checked somethings below:
 * Watchdog Driver
CONFIG_WATCHDOG=y
CONFIG_WATCHDOG_CORE=y
CONFIG_WATCHDOG_HANDLE_BOOT_ENABLED=y
CONFIG_GPIO_WATCHDOG=m
CONFIG_BCM2835_WDT=y
**=> I think it is already enabled**
* Watchdog daemon
Loaded: loaded (/lib/systemd/system/watchdog.service; enabled; vendor preset:
 enabled)
   Active: [[0;1;32mactive (running)[[0m since Sat 2019-09-07 13:33:39 +07; 9min
 ago
**=> it is running**
Still cannot reboot

**To reproduce**
- Build 64 bit kernel for Raspberry PI 3 B+
- Boot up the board
- Use command "$ reboot" to reboot raspberry

**Expected behaviour**
System reboot 

**Actual behaviour**
System hang 

**System**
* Raspberry Pi 3B+
* Kernel 64 bit on branch 4.19-y with latest commit.

**Logs**
[  OK  ] Unmounted /boot.
[  OK  ] Unmounted /var/volatile.
[  OK  ] Stopped target Local File Systems (Pre).
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Stopped File System Check on Root Device.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped target Swap.
[  OK  ] Reached target Shutdown.
[  OK  ] Reached target Final Step.
[  OK  ] Started Reboot.
[  OK  ] Reached target Reboot.
[   94.601114] systemd-shutdow: 35 output lines suppressed due to ratelimiting
[   94.639062] systemd-shutdown[1]: Syncing filesystems and block devices.
[   94.690724] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[   94.713668] systemd-journald[100]: Received SIGTERM from PID 1 (systemd-shutdow).
**Stop here and no reboot happend**
[config.txt](https://github.com/raspberrypi/linux/files/3586321/config.txt)

[rpi3b_plus_defconfig.txt](https://github.com/raspberrypi/linux/files/3586322/rpi3b_plus_defconfig.txt)

pelwell, 2019-09-09:
`$ reboot` on a 3B+ with a 64-bit kernel works for me on 4.19.71, both from the console and from the desktop.

Why have you attached a 32-bit config (it's not a _defconfig) when you are reporting a problem with a 64-bit kernel?

NguyenTrongThinh, 2019-09-09:
@pelwell github do not allow me update a file without extension so i change it to .txt. By the way, I extract the config file from raspberry PI 3 B+ (raspian Image) by command below:
$ sudo modprobe configs
$ zcat /proc/config.gz > .config

So, what wrong in my kernel configuration? How to know it is a 32 bit config?
Please teach!

pelwell, 2019-09-09:
A 64-bit config includez `CONFIG_ARM64=y` and `CONFIG_64BIT=y`, whereas a 32-bit config has `CONFIG_ARM=y` instead.

Try building your 64-bit kernel with `bcmrpi3_defconfig`.

NguyenTrongThinh, 2019-09-09:
@pelwell I will try and update the result.

NguyenTrongThinh, 2019-09-10:
i used bcmrpi3_defconfig to rebuild kernel, but the same issues . Debug log would like below

...
[  OK  ] Unmounted Temporary Directory (/tmp).
[  OK  ] Unmounted /boot.
[  OK  ] Unmounted /var/volatile.
[  OK  ] Stopped target Local File Systems (Pre).
[  OK  ] Stopped Create Static Device Nodes in /dev.
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Stopped File System Check on Root Device.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped target Swap.
[  OK  ] Reached target Shutdown.
[  OK  ] Reached target Final Step.
[  OK  ] Started Reboot.
[  OK  ] Reached target Reboot.
[ 4047.060800] systemd-shutdow: 32 output lines suppressed due to ratelimiting
[ 4047.106165] systemd-shutdown[1]: Syncing filesystems and block devices.
[ 4047.164595] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[ 4047.190656] systemd-journald[95]: Received SIGTERM from PID 1 (systemd-shutdow).

**It stops here for about 50s and then reboot.** Is this the right approach? Why does it stop too long?

pelwell, 2019-09-10:
Try working through this diagnostic process: https://freedesktop.org/wiki/Software/systemd/Debugging/#index2h1

NguyenTrongThinh, 2019-09-10:
$ reboot -f works for me. maybe a wrong bug report 
Thanks

closed 2019-09-10 by NguyenTrongThinh

pelwell, 2019-09-10:
Feel free to reopen if it starts to look like a kernel issue after all.