raspberrypi/linux #3221: Distortion on new gles drivers on 1366x768

carlonluca, 2019-09-11:
**Is this the right place for my bug report?**
It is very difficult to say what component is causing the issue, but maybe someone here can help.

**Describe the bug**
I ported a system including Qt apps based on eglfs/dispmanx to fkms in order to support rpi4. Apps are also used on Windows, rpi3, mac os etc... with small variations. I have issues with animations that I cannot explain: some scaling animations seem to cause distortion of the scene. The distortion appears a bit like a vsync issue (just trying to describe it, not sure if it is related to vsync). The distortion often persists after the animation completes. If I set the animation to have duration 0, no issue appear.
The distortion seems to only appear when resolution is set to values "near" 1366x768 (I could also reproduce on 1360x768) **and** on fkms. 1920x1080 seems to be ok with both drivers. 1280x720 seems to be ok with both drivers. 1366x768 was ok on old dispmanx drivers but is not on fkms.

In some cases, when the animation starts on 1366x768, I get this from the kernel:

```
[  262.072366] [drm] Resetting GPU.
[  270.952406] [drm] Resetting GPU.
[  276.952461] [drm] Resetting GPU.
[  278.952417] [drm] Resetting GPU.
[  281.112422] [drm] Resetting GPU.
[  283.032194] [drm] Resetting GPU.
```

**To reproduce**
I tried to write a minimal example showing the problem, but I failed. Even in the same software, not every scaling animations show the problem.
What I see is that I can reproduce by running some scaling animations on fkms by setting the resolution to 1366x768.

**Expected behaviour**
No distortion, like with the legacy drivers.

**Actual behaviour**
Distortion appears.

**System**
* Which model of Raspberry Pi?

Pi3B+

* Which OS and version (`cat /etc/rpi-issue`)?

```
Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage2
```

* Which firmware version (`vcgencmd version`)?

```
Aug 15 2019 12:08:48
Copyright (c) 2012 Broadcom
version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start_x)
```

* Which kernel version (`uname -a`)?

`Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux`

**Logs**
Dmesg output:

```
[  262.072366] [drm] Resetting GPU.
[  270.952406] [drm] Resetting GPU.
[  276.952461] [drm] Resetting GPU.
[  278.952417] [drm] Resetting GPU.
[  281.112422] [drm] Resetting GPU.
[  283.032194] [drm] Resetting GPU.
```

Not sure if other logs exist somewhere. I tried to get logs with vcdbg but I got an error:

```
sudo vcdbg log msg
Failed to allocate -505734863 bytes for message buffer
```

**Additional context**
My application runs without X11. Difficult to say if this is a problem in Qt, my app, drivers etc... Any advice on how to investigate further?

carlonluca, 2019-09-23:
I'm not getting any answer here. May I have more luck in the firmware repo?

JamesH65, 2019-09-23:
I doubt anyone has had time to look at this. Note that 1366x768 is a very strange resolution that does cause problems in the HDMI system due to having odd numbered timings. Might be related.

carlonluca, 2019-09-23:
Thank you for answering. That is the resolution of the screen I'm targeting. To temporarily "fix", I simply forced a resolution of 720p when auto detection sets fb to 1366x768 or 1360x768, but I'm afraid people may hit the same issue with other resolutions.

Do you think someone will have time to look into this (I perfectly understand you have priorities to respect)? Do you suggest to leave the message here or in the firmware repo? I'm testing a rpi3b+ atm.
Thanks again.

6by9, 2019-09-23:
This would appear to be related to KMS and/or Mesa, so this is the right place.

Is the behaviour the same with both vc4-kms-v3d(*) and vc4-fkms-v3d on your Pi3? That would imply that it is more in the 3D rendering side than the composition and output side.
```[drm] Resetting GPU.``` would seem to say that it is 3D related anyway.

The behaviour may be slightly different on a Pi4 anyway, as the 3D block is 2 generations further on than Pi3.

(*) NB That vc4-kms-v3d is currently NOT available on Pi4.

carlonluca, 2019-09-25:
To test I simply switched from dtoverlay=vc4-fkms-v3d to dtoverlay=vc4-kms-v3d, is this sufficient to switch from fkms to kms?

I ran the same test and the issue is there in both cases. However, there is a slight difference: with fkms the scale animation stutters and then, for some time, I get a corruption similar to horizontal tearing, which is very noticeable as there is a scrollable content in my UI.
On kms, the same scale animation stutters but then I can see no noticeable "corruption" when scrolling the content. So it seems the only issue is the scaling animation on kms.

carlonluca, 2020-02-07:
I could reproduce the issue also on other resolutions, like 1080p and 720p only on rpi3. Rpi4 seems to work properly.

mbtronics, 2021-12-24:
I got this exact problem when overclocking from 1.2Ghz to 1.3Ghz (in batocera)

popcornmix, 2023-01-24:
Is this still an issue?

mbtronics, 2023-01-24:
I have not tried a recent version in half a year, so no idea actually

jcea, 2023-07-21:
Maybe related: https://discourse.osmc.tv/t/osmcs-june-update-is-here-with-kodi-v20-osmc/95094/90?u=jcea

pelwell, 2023-07-21:
This issue was marked for closure in January, and that's what I'm going to do. I suggest you disable your screensaver (or try a different one), and wait for another OSMC update - ideally to a 6.1 kernel.

closed 2023-07-21 by pelwell

jcea, 2023-07-21:
If you read the OSMC thread, you will see that this is not related to a screensaver or to me, specifically. It is a pretty widespread issue with current OSMC release, using kernel 5.15.83-3.

If you are closing this issue, where should I report the problem?

Thanks.

jcea, 2023-07-21:
I guess this issue was marked as "to be closed" because no more report were coming in. Reports are coming in now :-).

Thanks.

popcornmix, 2023-07-21:
The 5.15 kernel is no longer updated. If any fixes are required they will appear on the current 6.1 kernel tree.

So we'd need to know if the issue still affects the 6.1 kernel.

You could test an RPiOS image (which will use the 6.1 kernel). Install kodi and the screensaver from apt, and test if the issue occurs. If it does, we can investigate.

pizdjuk, 2023-11-10:
> The 5.15 kernel is no longer updated. If any fixes are required they will appear on the current 6.1 kernel tree.
>
> So we'd need to know if the issue still affects the 6.1 kernel.
>
> You could test an [...]

I catch the issue on RPi 3B+, LibreELEC (official): 11.0.3, Kernel 6.1.38. The resolution was standard 1920x1080.

pizdjuk, 2023-11-12:
@pelwell  please, reopen.

arren-ru, 2023-11-19:
Got the same issue on RPi3B+ with Linux 6.1.38 armv7l GNU/Linux, stuck with [drm] Resetting GPU.

pizdjuk, 2023-11-24:
Other time got the same issue. But this time only one time appeared the line [drm] Resetting GPU after some kernel stack dump
[kodi_screen_shifted.log](https://github.com/raspberrypi/linux/files/13454175/kodi_screen_shifted.log)

The screen was then shifted horizontally on ~0.1 screenwidth to the left. On the right side was the part that should be no the left. Restarting kodi didnt help, reboot of device helped.