raspberrypi/linux #3222: rpi4 64bit  Internal error: Oops: 96000005 [#1] PREEMPT SMP

yrzr, 2019-09-11:
**Describe the bug**
```
[  402.108994] Unable to handle kernel paging request at virtual address 0000000800000022
[  402.108999] Mem abort info:
[  402.109001]   ESR = 0x96000005
[  402.109004]   Exception class = DABT (current EL), IL = 32 bits
[  402.109006]   SET = 0, FnV = 0
[  402.109008]   EA = 0, S1PTW = 0
[  402.109009] Data abort info:
[  402.109011]   ISV = 0, ISS = 0x00000005
[  402.109012]   CM = 0, WnR = 0
[  402.109017] user pgtable: 4k pages, 39-bit VAs, pgdp = 000000005f5d0f95
[  402.109019] [0000000800000022] pgd=0000000000000000, pud=0000000000000000
[  402.109025] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[  402.114597] Modules linked in: nfsd xt_nat xt_tcpudp veth ipt_MASQUERADE nf_conntrack_netlink nfnetlink iptable_nat nf_nat_ipv4 xt_addrtype iptable_filter ip_tables xt_conntrack x_tables nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 br_netfilter bridge stp llc overlay brcmfmac vc4 bcm2835_codec(C) bcm2835_v4l2(C) cfg80211 v4l2_mem2mem drm_kms_helper videobuf2_dma_contig brcmutil cec v3d videobuf2_vmalloc videobuf2_memops v4l2_common videobuf2_v4l2 syscopyarea sysfillrect videobuf2_common gpu_sched sysimgblt videodev fb_sys_fops drm media drm_panel_orientation_quirks fixed
[  402.133956] Unable to handle kernel paging request at virtual address 000000080000000a
[  402.165658] Process emerge (pid: 4270, stack limit = 0x00000000bc91f879)
[  402.173571] Mem abort info:
[  402.180257] CPU: 3 PID: 4270 Comm: emerge Tainted: G         C        4.19.71-yrzr+ #40
[  402.180259] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[  402.183041]   ESR = 0x96000005
[  402.191035] pstate: 60000005 (nZCv daif -PAN -UAO)
[  402.191046] pc : anon_vma_interval_tree_insert+0x44/0xb0
[  402.196859]   Exception class = DABT (current EL), IL = 32 bits
[  402.199902] lr : anon_vma_fork+0x108/0x180
[  402.199904] sp : ffffff8010d9bc50
[  402.204684]   SET = 0, FnV = 0
[  402.209982] x29: ffffff8010d9bc50 x28: 0000000000000001 
[  402.215893]   EA = 0, S1PTW = 0
[  402.223284] Data abort info:
[  402.226326] x27: ffffffc05b1c6280 
[  402.231628]   ISV = 0, ISS = 0x00000005
[  402.234757] x26: ffffff8008c9ec48 
[  402.237627]   CM = 0, WnR = 0
```

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
rpi4b
* Which OS and version (`cat /etc/rpi-issue`)?
```
~  cat /etc/gentoo-release
Gentoo Base System release 2.6
```
* Which kernel version (`uname -a`)?
```
~ uname -a
Linux rpi4 4.19.71-yrzr+ #40 SMP PREEMPT Mon Sep 9 21:55:52 HKT 2019 aarch64 GNU/Linux
```

JamesH65, 2019-12-20:
@yrzr Is this issue present on the latest Raspbian? For Gentoo support you should use the Gentoo forums.

JamesH65, 2020-02-13:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2020-02-13 by JamesH65

mwatermolen, 2022-05-27:
I think we just got this today.
PXE boot NFSv3 Raspbian x64. rpi4b

```
Linux lnk-z01-tv-3 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```

```
Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage4
```

```
[32185.008731] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000018
[32185.008765] Mem abort info:
[32185.008770]   ESR = 0x96000005
[32185.008776]   EC = 0x25: DABT (current EL), IL = 32 bits
[32185.008783]   SET = 0, FnV = 0
[32185.008788]   EA = 0, S1PTW = 0
[32185.008792]   FSC = 0x05: level 1 translation fault
[32185.008798] Data abort info:
[32185.008802]   ISV = 0, ISS = 0x00000005
[32185.008807]   CM = 0, WnR = 0
[32185.008813] user pgtable: 4k pages, 39-bit VAs, pgdp=00000000427aa000
[32185.008820] [0000000000000018] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[32185.008840] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[32185.008846] Modules linked in: rfcomm cmac algif_hash aes_arm64 algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc snd_soc_hdmi_codec brcmfmac brcmutil pwm_raspberrypi_poe raspberrypi_hwmon v3d gpu_sched cfg80211 bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem rfkill vc_sm_cma(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common vc4 8021q videodev i2c_brcmstb snd_bcm2835(C) garp stp mc llc cec drm_kms_helper snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer rpivid_mem snd syscopyarea sysfillrect sysimgblt fb_sys_fops rpi_poe_power nvmem_rmem pwm_fan uio_pdrv_genirq uio i2c_dev drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[32185.008972] CPU: 3 PID: 725 Comm: Xorg Tainted: G         C        5.15.32-v8+ #1538
[32185.008979] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[32185.008985] pstate: 00000005 (nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[32185.008991] pc : clk_request_done+0x70/0xe0
[32185.009004] lr : clk_request_done+0x2c/0xe0
[32185.009008] sp : ffffffc00a8eb900
[32185.009011] x29: ffffffc00a8eb900 x28: ffffff805f246a00 x27: 0000000000000003
[32185.009020] x26: ffffffc000f5a698 x25: ffffff80725da770 x24: ffffff80725da580
[32185.009027] x23: ffffff80725da700 x22: ffffff8041e7a480 x21: ffffff804360b000
[32185.009035] x20: ffffff80725dac00 x19: ffffff8041ca3c80 x18: 0000000000000000
[32185.009042] x17: 0000000000000465 x16: 0000000000008007 x15: 0438000008980804
[32185.009051] x14: ffffff8099d000e8 x13: ffffff808273d700 x12: 0000000001db8c00
[32185.009058] x11: 0000000000000000 x10: 0000000000000009 x9 : ffffffc0086d687c
[32185.009065] x8 : ffffff80725dac80 x7 : ffffff8042705ac0 x6 : ffffff8042705ac0
[32185.009073] x5 : 0000000000000000 x4 : dead000000000100 x3 : dead000000000122
[32185.009080] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff80725dacd8
[32185.009089] Call trace:
[32185.009092]  clk_request_done+0x70/0xe0
[32185.009097]  vc4_atomic_commit_tail+0x4e4/0x730 [vc4]
[32185.009136]  commit_tail+0xac/0x1a8 [drm_kms_helper]
[32185.009202]  drm_atomic_helper_commit+0x168/0x378 [drm_kms_helper]
[32185.009238]  drm_atomic_commit+0x54/0x68 [drm]
[32185.009561]  drm_atomic_helper_disable_plane+0xb8/0x148 [drm_kms_helper]
[32185.009673]  __setplane_atomic+0x134/0x140 [drm]
[32185.009760]  drm_mode_cursor_universal+0x12c/0x258 [drm]
[32185.009826]  drm_mode_cursor_common+0x100/0x218 [drm]
[32185.009892]  drm_mode_cursor_ioctl+0x58/0x80 [drm]
[32185.009960]  drm_ioctl_kernel+0xc8/0x118 [drm]
[32185.010026]  drm_ioctl+0x21c/0x450 [drm]
[32185.010091]  __arm64_sys_ioctl+0xb0/0xf0
[32185.010100]  invoke_syscall+0x4c/0x110
[32185.010108]  el0_svc_common.constprop.3+0xfc/0x120
[32185.010114]  do_el0_svc+0x2c/0x90
[32185.010119]  el0_svc+0x24/0x60
[32185.010125]  el0t_64_sync_handler+0x90/0xb8
[32185.010131]  el0t_64_sync+0x1a0/0x1a4
[32185.010138] Code: d2800001 eb00005f 54000100 d503201f (f9400c43)
[32185.010145] ---[ end trace 76f28b713bfad878 ]---
```