raspberrypi/linux #3231: Driver sc16is7xx is not enabling IRQ in sc16is752 chip?

kaamil1984, 2019-09-14:
I'm trying to use sc16is752 chip connected to spi1 (and INT to gpio24) and it looks like **chip does not generate IRQ when it receives some data**.

I have checked this with logic analyzer. SC16IS752 INT pin is always HIGH (pulled up to Vdd by 1kOhm resistor).

I'm using `screen /dev/ttySC0 9600` to check if port works. It does not receive data until I "manually" generate falling edge on gpio24 (by shorting it to GND for a moment). When I do that - looks like whole buffer is being read, so interrupt mechanism is working fine.

I also tried to disconnect INT pin from Raspberry to be sure, that Raspberry is not holding this pin in HIGH state somehow.

I saw driver source here: [sc16is7xx.c line 1042](https://github.com/torvalds/linux/blob/master/drivers/tty/serial/sc16is7xx.c#L1042), it looks that it should enable interrupts in chip, but it doesn't.

```
static int sc16is7xx_startup(struct uart_port *port)
{
	struct sc16is7xx_port *s = dev_get_drvdata(port->dev);
	unsigned int val;

	sc16is7xx_power(port, 1);

	/* Reset FIFOs*/
	val = SC16IS7XX_FCR_RXRESET_BIT | SC16IS7XX_FCR_TXRESET_BIT;
	sc16is7xx_port_write(port, SC16IS7XX_FCR_REG, val);
	udelay(5);
	sc16is7xx_port_write(port, SC16IS7XX_FCR_REG,
			     SC16IS7XX_FCR_FIFO_BIT);

	/* Enable EFR */
	sc16is7xx_port_write(port, SC16IS7XX_LCR_REG,
			     SC16IS7XX_LCR_CONF_MODE_B);

	regcache_cache_bypass(s->regmap, true);

	/* Enable write access to enhanced features and internal clock div */
	sc16is7xx_port_write(port, SC16IS7XX_EFR_REG,
			     SC16IS7XX_EFR_ENABLE_BIT);

	/* Enable TCR/TLR */
	sc16is7xx_port_update(port, SC16IS7XX_MCR_REG,
			      SC16IS7XX_MCR_TCRTLR_BIT,
			      SC16IS7XX_MCR_TCRTLR_BIT);

	/* Configure flow control levels */
	/* Flow control halt level 48, resume level 24 */
	sc16is7xx_port_write(port, SC16IS7XX_TCR_REG,
			     SC16IS7XX_TCR_RX_RESUME(24) |
			     SC16IS7XX_TCR_RX_HALT(48));

	regcache_cache_bypass(s->regmap, false);

	/* Now, initialize the UART */
	sc16is7xx_port_write(port, SC16IS7XX_LCR_REG, SC16IS7XX_LCR_WORD_LEN_8);

	/* Enable the Rx and Tx FIFO */
	sc16is7xx_port_update(port, SC16IS7XX_EFCR_REG,
			      SC16IS7XX_EFCR_RXDISABLE_BIT |
			      SC16IS7XX_EFCR_TXDISABLE_BIT,
			      0);

	/* Enable RX, TX interrupts */
	val = SC16IS7XX_IER_RDI_BIT | SC16IS7XX_IER_THRI_BIT;
	sc16is7xx_port_write(port, SC16IS7XX_IER_REG, val);

	return 0;
}
```

I'm using RPi CM3 module (with "Waveshare Compute Module IO Board Plus" development board).

**System**

```
# cat /etc/rpi-issue

Raspberry Pi reference 2019-04-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e910bf55ac92e27c6e77043e8995bcb5e0a2d742, stage2
```

```
# vcgencmd version
Aug 15 2019 12:06:42
Copyright (c) 2012 Broadcom
version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start)
```

```
# uname -a
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
```

pelwell, 2019-09-25:
Are we supposed to know/assume/guess that pin 6 on CON28 is wired to GPIO24? Many people are using this device, so I'd be very surprised if the IRQ function doesn't work.

JamesH65, 2019-12-20:
@kaamil1984 Is this issue still present?

kaamil1984, 2019-12-27:
Well, it looks like that I had some other issue with device tree or drivers.

After few tries with diffrent GPIO mappings I managed to get it work.

I also discovered, that IRQ functionality is not used by UART driver when it receives data. On logic analyzer I see that linux driver continuously checks if there is some data in buffer.

I'm not completely satisfied how this driver (or sc16is752 device) works, because SPI bus is continuously busy, but **IT WORKS**.

closed 2019-12-27 by kaamil1984