raspberrypi/linux #3232: arm64 compile error for 5.2.y

laroche, 2019-09-15:
**Describe the bug**
Compiling an arm64 5.2.14 kernel on Debian 10 gives an error.
This is due to the patch 97741883e26a7936cd1122e4f566d886bb5a17f5
"Ported pcie-brcmstb bounce buffer implementation to ARM64. (#3144)"
Leaving out this patch, everything compiles ok.

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
/home/flaroche/data/kernel/linux-5/arch/arm64/mm/dma-mapping.c: In function <E2><80><98>__iommu_mmap_attrs<E2><80><99>:
/home/flaroche/data/kernel/linux-5/arch/arm64/mm/dma-mapping.c:298:10: error: implicit declaration of function <E2><80><98>__swiotlb_mmap_pfn<E2><80><99>; did you mean <E2><80><98>swiotlb_map<E2><80><99>? [-Werror=implicit-function-declaration]
   return __swiotlb_mmap_pfn(vma, pfn, size);
          ^~~~~~~~~~~~~~~~~~
          swiotlb_map
/home/flaroche/data/kernel/linux-5/arch/arm64/mm/dma-mapping.c: In function <E2><80><98>__iommu_get_sgtable<E2><80><99>:
/home/flaroche/data/kernel/linux-5/arch/arm64/mm/dma-mapping.c:326:10: error: implicit declaration of function <E2><80><98>__swiotlb_get_sgtable_page<E2><80><99>; did you mean <E2><80><98>__iommu_get_sgtable<E2><80><99>? [-Werror=implicit-function-declaration]
   return __swiotlb_get_sgtable_page(sgt, page, size);
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
          __iommu_get_sgtable
/home/flaroche/data/kernel/linux-5/arch/arm64/mm/dma-mapping.c: In function <E2><80><98>__iommu_map_page<E2><80><99>:
/home/flaroche/data/kernel/linux-5/arch/arm64/mm/dma-mapping.c:380:7: error: implicit declaration of function <E2><80><98>iommu_dma_mapping_error<E2><80><99>; did you mean <E2><80><98>pci_dma_mapping_error<E2><80><99>? [-Werror=implicit-function-declaration]
  if (!iommu_dma_mapping_error(dev, dev_addr) &&
       ^~~~~~~~~~~~~~~~~~~~~~~
       pci_dma_mapping_error

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

laroche, 2019-09-15:
The patch removes __swiotlb_mmap_pfn, but this function is still called from the same
file if CONFIG_IOMMU_DMA is defined.

pelwell, 2019-09-16:
Sorry about that - it got broken (by me) back in 4.20 while pulling in all the recent work on 4.19 (there's been enough change in this area to make the merges interesting), then propagated with further changes to 5.0-5.2. I think it should be good now.

laroche, 2019-09-17:
Hello Phil Elwell,

Big thanks for fixing this so fast. I've started a new kernel compile and assume this will
work out ok, so closing this issue.

Thanks a lot,
regards from Germany,

Florian La Roche

closed 2019-09-17 by laroche