raspberrypi/linux #3233: NULL pointer dereference in drm_mode_object_get_properties

ickromwerk, 2019-09-16:
```
$ uname -a
Linux pi64 4.19.67-v8-174fcab91765-p4-bis+ #2 SMP PREEMPT Tue Aug 27 13:58:09 GMT 2019 aarch64 GNU/Linux
```

I'm trying to use RPi4B with SDTV video output on clean install of gentoo64( arm64 kernel ).

To enable SDTV output added to config.txt:
```
sdtv_mode=2
enable_tvout=1
```

System starts. Shows rainbow splash and then boots until black screen.

dmesg shows:
```
[   20.329498] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000030
[   20.329504] Mem abort info:
[   20.329508]   ESR = 0x96000005
[   20.329513]   Exception class = DABT (current EL), IL = 32 bits
[   20.329516]   SET = 0, FnV = 0
[   20.329520]   EA = 0, S1PTW = 0
[   20.329523] Data abort info:
[   20.329526]   ISV = 0, ISS = 0x00000005
[   20.329529]   CM = 0, WnR = 0
[   20.329535] user pgtable: 4k pages, 39-bit VAs, pgdp = 00000000403ca293
[   20.329539] [0000000000000030] pgd=0000000000000000, pud=0000000000000000
[   20.329551] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[   20.329560] Modules linked in: lz4 lz4_compress hci_uart btbcm serdev bnep bluetooth ecdh_generic ipv6 brcmfmac bcm2835_codec(C) brcmutil vc4 sha256_generic bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) v4l2_common drm_kms_helper cfg80211 videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops cec videobuf2_v4l2 snd_soc_core videobuf2_common rfkill snd_bcm2835(C) snd_compress videodev snd_pcm_dmaengine v3d syscopyarea snd_pcm media sysfillrect sysimgblt fb_sys_fops snd_timer gpu_sched vc_sm_cma(C) spidev snd drm vchiq(C) raspberrypi_hwmon drm_panel_orientation_quirks spi_bcm2835 hwmon rpivid_mem uio_pdrv_genirq uio fixed
[   20.329683] Process X (pid: 2467, stack limit = 0x00000000ece3d3a6)
[   20.329691] CPU: 0 PID: 2467 Comm: X Tainted: G         C        4.19.67-v8-174fcab91765-p4-bis+ #2
[   20.329695] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[   20.329701] pstate: 20000005 (nzCv daif -PAN -UAO)
[   20.329740] pc : drm_mode_object_get_properties+0x8c/0x1c0 [drm]
[   20.329769] lr : drm_mode_getconnector+0x220/0x490 [drm]
[   20.329773] sp : ffffff800fd53ac0
[   20.329777] x29: ffffff800fd53ac0 x28: 0000000000000000
[   20.329785] x27: ffffff800fd53cfc x26: 0000000000000001
[   20.329793] x25: ffffff8008de8688 x24: 0000000000000000
[   20.329800] x23: 0000000000000000 x22: ffffffc0f20cb0a8
[   20.329807] x21: 0000000000000000 x20: 0000000000000005
[   20.329814] x19: 0000000000000005 x18: ffffffc0f20cc2b0
[   20.329821] x17: 0000000000000000 x16: 0000000000000000
[   20.329828] x15: 0000000000000000 x14: ffffffc0f3a9e1c0
[   20.329835] x13: ffffffc0f3a9dff0 x12: 0000000000000000
[   20.329842] x11: 0000000000000035 x10: 0000000000000000
[   20.329849] x9 : 0000000000000040 x8 : 0000000000000001
[   20.329856] x7 : 0000000000000228 x6 : 0000000000000035
[   20.329863] x5 : ffffffc0f23e1400 x4 : ffffff800fd53cfc
[   20.329870] x3 : 0000000000000000 x2 : 0000000000000005
[   20.329877] x1 : ffffffc0f20cb200 x0 : ffffffc0f20cb1d8
[   20.329884] Call trace:
[   20.329912]  drm_mode_object_get_properties+0x8c/0x1c0 [drm]
[   20.329940]  drm_mode_getconnector+0x220/0x490 [drm]
[   20.329968]  drm_ioctl_kernel+0xc0/0x150 [drm]
[   20.329994]  drm_ioctl+0x2d4/0x560 [drm]
[   20.330004]  do_vfs_ioctl+0xc4/0x860
[   20.330008]  ksys_ioctl+0x50/0xa0
[   20.330013]  __arm64_sys_ioctl+0x28/0x40
[   20.330020]  el0_svc_common+0xa4/0x1a0
[   20.330025]  el0_svc_handler+0x38/0x80
[   20.330030]  el0_svc+0x8/0xc
[   20.330036] Code: 5400088d 8b34cc01 f9400435 2a1303e2 (b94032a1)
[   20.330043] ---[ end trace 62fab3833404709e ]---
```

ickromwerk, 2019-09-16:
As far as I understand some application( PID 2467 ) is trying to make drm_ioctl call and it fails.

I don't have a clues how to figure out what this application is. Because PID 2467 not shows in processes. How to catch it? Do I need to trace all syscalls during boot?

6by9, 2019-09-16:
Already fixed. https://github.com/raspberrypi/linux/pull/3223

JamesH65, 2019-09-16:
Fixed on Raspbian, so closing.

closed 2019-09-16 by JamesH65

ickromwerk, 2019-09-16:
Confirm. it works now