raspberrypi/linux #3234: Raspberry Pi 2 Model B Rev 1.2 only sees 1 CPU with 64bit kernel

digitallyelite, 2019-09-16:
**Describe the bug**
Upon boot, only 1 CPU is available for use.

**To reproduce**
Install 64bit kernel on a Raspberry Pi 2 Model B Rev 1.2

**Expected behaviour**
All 4 CPUs should be available

**Actual behaviour**
Only 1 CPU is available (32bit kernel does not have this issue)

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
RaspberryPi2 v1.2
* Which OS and version (`cat /etc/rpi-issue`)?
Raspbian Buster
* Which firmware version (`vcgencmd version`)?
Sep 12 2019 16:32:35
Copyright (c) 2012 Broadcom
version 4b30e8dcf91177a0be822c4c8d3bac8ed65f7ea5 (clean) (release) (start)
* Which kernel version (`uname -a`)?
4.19.71
**Logs**
`dmesg | grep cpu` spits out the following that I think is pertient...
[0.000000] /cpus/cpu@1: missing enable-method property
[0.000000] /cpus/cpu@2: missing enable-method property
[0.000000] /cpus/cpu@3: missing enable-method property

**Additional context**
Both 32bit and 64bit /proc/cpuinfo list 
Hardware: 2835
Revision: a02042
Model: Raspberry Pi 2 Model B Rev 1.2

pelwell, 2019-09-16:
The reason will be that Pi 2B didn't used to be a 64-bit device, but the rev 1.2 iteration comes with at 2837. The Device Tree for the 2B hasn't changed, hence the lack of enable-methods. The correct solution would be to introduce bcm2710-rpi-2-b.dtb, but it would require changes to the kernel and firmware repos.

digitallyelite, 2019-09-16:
@popcornmix it happened with both a self built as well as rpi-update with arm_64bit=1

full uname -a is
`Linux raspberrypi 4.19.71-v8+ #1263 SMP PREEMPT Thu Sep 12 16:28:48 BST 2019 aarch64 GNU/Linux`

@pelwell i'm assuming that that work is very low on the totem pole/won't be done at all?  I could probably mash together a dts file, but if the firmware never recognizes it (Which revision is incorrect, the matrix at the revision-codes URL above, or the one on the board?) would it even be worth doing?

pelwell, 2019-09-16:
You can force the firmware to use any DTB using the `device_tree=whatever.dtb` setting in config. txt. I was referring to the automatic file selection based on revision codes. If you can hack together and share a working dts it will make it much more likely the rest of the work will get done.

pelwell, 2019-09-16:
PR #3235 shouldn't be too far wrong.

pelwell, 2019-09-17:
A small patch has been dropped into the firmware mill and may pop out of the other side in the next few days.

pelwell, 2019-09-17:
The latest firmware (available from rpi-update) should attempt to load bcm2710-rpi-2-b.dtb on a rev 1.2 Pi 2, falling back to bcm2709-rpi-2-b.dtb if it isn't found.

digitallyelite, 2019-09-17:
Just tested the firmware here and it does seem to be working as intended, 2710 if it exists, and 2709 if it does not.  I haven't noticed anything super out of the ordinary on booting it 64bit - it sees all 1GB of ram, rng tools seems to be starting, kernel uname is
`Linux raspberrypi 4.19.73-v8+ #1266 SMP PREEMPT Tue Sep 17 21:08:24 BST 2019 aarch64 GNU/Linux`

The only thing out of the ordinary to me, is that it still reports a revision of `a02042` instead of `a22042` - unless I have one from Sony UK and not Embest and it just hasn't been added to the matrix yet?

pelwell, 2019-09-18:
If the board has a small QR code sticker on the bottom then it's built by Sony.

digitallyelite, 2019-09-18:
There is a very tiny one, with what could be a QR code and a "19/05" on it so I guess Sony UK are doing them and it's just not on the matrix yet :)

pelwell, 2019-09-18:
Yes, that's the one.

Where is this matrix you keep referring to?

digitallyelite, 2019-09-18:
> Where is this matrix you keep referring to?
https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md 

Or maybe better, should I send a PR to https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/revision-codes/README.md to add the Sony UK revision?

pelwell, 2019-09-18:
> Or maybe better, should I send a PR

Yes - go for it. I'd merge it myself if I had the privileges on that repo.

digitallyelite, 2019-09-18:
Done, thank you so much for the work to get 64bit support in for the RaspberryPi2 v1.2, it's greatly appreciated!

digitallyelite, 2019-09-18:
Actually... something isn't quite right, but I don't think it's specific to *these* changes.  Basically, all of my testing happened on the minimal Buster image, which is console only.  But if I enable 64bit (with or without the dtb, which is why I believe this issue isn't specific to these changes) all I get is a blank screen, but it won't let me even switch to a console.  X just seems to complain about a missing g2d_23 driver not found in /lib/modules/4.19.73-v8+ (??) and then there's an undefined symbol in fbturbo_drv.so... is this something I should take to the forums?

pelwell, 2019-09-18:
Do you have "dtoverlay=vc4-fkms-v3d" in config.txt?

digitallyelite, 2019-09-18:
Only in the [pi4] section

pelwell, 2019-09-18:
I suggest trying it, at least as a workaround.

digitallyelite, 2019-09-18:
that worked, thanks for the help.  Since the dts is in now, I believe we can close this issue?

pelwell, 2019-09-18:
I won't deny you the pleasure.

closed 2019-09-18 by digitallyelite