raspberrypi/linux #3236: V4L2 problem with 32bit and 64bit

peel70, 2019-09-16:
Hello,
I've two different problems with v4l2 codec on 32 and 64 bit raspberry linux

I've build 32 and 64 bit image for RPI4 with bitbake poky master branch because there is a problem with MESA on Warrior.

RPI4-32bit master GStreamer 1.16 Qt 5.13 v4l2 decoder is active but I'm not able to play mp4 files, work on .mov with gst-launch and some particular mp4 codec but not in QMultimedia. It play only mkv on QMultimedia and Gstreamer but with 2fps

RPI4-64bit master GStreamer 1.16 Qt5.13 v4l2 decoder is not enabled (and don't know why) all video play in both GStreamer and QMultimedia but via avdec_h264 (ffmpeg libav) software decode CPU over 250% for the same video file that play on RPI3 at 18-20% CPU

RPI-32bit burster light with Qt 5.12.3 cross compiled and Gstreamer 1.16 (not compiled by me) same problem ad RPI4-32bit master

There something wrong in my configuration?

JamesH65, 2019-12-20:
Do these issues happen on the standard 32 and 6 bit Raspbian kernels?

peel70, 2019-12-22:
> Reply to: raspberrypi/linux <reply@reply.github.com>

Hi James,

thank you for your email.

The problem seems to be on both 32 and 64 bit versions.

Recently I’ve rebuild images from the latest master (ZEUS?) and the problem on 64bit seems to be fixed (not tested on 32 yet), I’m able to play .mp4 video with gst-launch snd v4l decoder

I think so because the CPU load is near the 20% during the video play (I think 1 copy render of the video)

On QML the load is near twice (I think 2 copy render of the video)

 

At the moment is not possible to play mkv (H265) video; is there the h265 hw decoder plugin or it will be available in the future?

If it possible to play h265 using the pi4 hw decoder with v4l and gstreamer, could you give me the correct pipeline to test?

 

Best regards

Enrico