raspberrypi/linux #3240: USB Host dwcotg:irq 39 disabled in 32 bit mode

imgaolong, 2019-09-18:
Compiled in 32 bit mode with default config file bcm2709_defconfig, the usb host initialization failed. And the usb mouse and keyboard will not work.

 I am using Raspberry Pi 3B+, and the distribution of Ubuntu Mate 18.04, and gcc 5.4 for arm-linux-gnueabihf-. The kernel branch is rpi-4.19.y.

dmesg shows:
[   12.706587] irq 39: nobody cared (try booting with the "irqpoll" option)
[   12.706592] CPU: 0 PID: 31 Comm: kworker/0:1 Not tainted 4.15.0-1032-raspi2 #34-Ubuntu
[   12.706594] Hardware name: BCM2835
[   12.706609] Workqueue: events deferred_probe_work_func
[   12.706633] [<80113394>] (unwind_backtrace) from [<8010e0bc>] (show_stack+0x20/0x24)
[   12.706646] [<8010e0bc>] (show_stack) from [<809d6d30>] (dump_stack+0xcc/0x110)
[   12.706655] [<809d6d30>] (dump_stack) from [<8019d194>] (__report_bad_irq+0x3c/0xdc)
[   12.706661] [<8019d194>] (__report_bad_irq) from [<8019d594>] (note_interrupt+0x278/0x2c4)
[   12.706669] [<8019d594>] (note_interrupt) from [<8019a334>] (handle_irq_event_percpu+0x6c/0x8c)
[   12.706677] [<8019a334>] (handle_irq_event_percpu) from [<8019a39c>] (handle_irq_event+0x48/0x6c)
[   12.706684] [<8019a39c>] (handle_irq_event) from [<8019dfec>] (handle_level_irq+0xbc/0x13c)
[   12.706691] [<8019dfec>] (handle_level_irq) from [<801991c4>] (generic_handle_irq+0x34/0x44)
[   12.706700] [<801991c4>] (generic_handle_irq) from [<805b9d1c>] (bcm2836_chained_handle_irq+0x48/0x4c)
[   12.706707] [<805b9d1c>] (bcm2836_chained_handle_irq) from [<801991c4>] (generic_handle_irq+0x34/0x44)
[   12.706714] [<801991c4>] (generic_handle_irq) from [<8019982c>] (__handle_domain_irq+0x6c/0xc4)
[   12.706722] [<8019982c>] (__handle_domain_irq) from [<801015fc>] (bcm2836_arm_irqchip_handle_irq+0xac/0xb0)
[   12.706729] [<801015fc>] (bcm2836_arm_irqchip_handle_irq) from [<809f34b0>] (__irq_svc+0x70/0xb0)
[   12.706732] Exception stack(0xb9eebae0 to 0xb9eebb28)
[   12.706738] bae0: 8013345c 00000000 60010193 60010193 b9eea000 00000000 00000000 00000001
[   12.706743] bb00: b9cc5780 ffffe000 00000002 b9eebb94 b9eebb30 b9eebb30 801016e8 801016ec
[   12.706746] bb20: 60010113 ffffffff
[   12.706752] [<809f34b0>] (__irq_svc) from [<801016ec>] (__do_softirq+0xec/0x460)
[   12.706761] [<801016ec>] (__do_softirq) from [<8013345c>] (irq_exit+0xd8/0x104)
[   12.706769] [<8013345c>] (irq_exit) from [<80199830>] (__handle_domain_irq+0x70/0xc4)
[   12.706775] [<80199830>] (__handle_domain_irq) from [<801015fc>] (bcm2836_arm_irqchip_handle_irq+0xac/0xb0)
[   12.706781] [<801015fc>] (bcm2836_arm_irqchip_handle_irq) from [<809f34b0>] (__irq_svc+0x70/0xb0)
[   12.706783] Exception stack(0xb9eebbf0 to 0xb9eebc38)
[   12.706787] bbe0:                                     394ac000 00000000 60010093 60010013
[   12.706792] bc00: 60010013 60010013 806429b8 00000075 b9eea000 80fb8318 80fb7de0 b9eebc9c
[   12.706796] bc20: b9eebc40 b9eebc40 8019739c 801973ac 60010013 ffffffff
[   12.706803] [<809f34b0>] (__irq_svc) from [<801973ac>] (console_unlock+0x3e8/0x59c)
[   12.706810] [<801973ac>] (console_unlock) from [<801980d8>] (register_console+0x23c/0x410)
[   12.706817] [<801980d8>] (register_console) from [<80639f78>] (uart_add_one_port+0x480/0x538)
[   12.706825] [<80639f78>] (uart_add_one_port) from [<80644540>] (pl011_register_port+0x6c/0xbc)
[   12.706831] [<80644540>] (pl011_register_port) from [<80644700>] (pl011_probe+0x170/0x190)
[   12.706838] [<80644700>] (pl011_probe) from [<805f66d4>] (amba_probe+0xe8/0x15c)
[   12.706845] [<805f66d4>] (amba_probe) from [<806690b8>] (driver_probe_device+0x314/0x488)
[   12.706853] [<806690b8>] (driver_probe_device) from [<806693e8>] (__device_attach_driver+0xac/0x130)
[   12.706859] [<806693e8>] (__device_attach_driver) from [<80666e60>] (bus_for_each_drv+0x68/0xc8)
[   12.706866] [<80666e60>] (bus_for_each_drv) from [<80668c04>] (__device_attach+0xe0/0x168)
[   12.706872] [<80668c04>] (__device_attach) from [<806694c8>] (device_initial_probe+0x1c/0x20)
[   12.706879] [<806694c8>] (device_initial_probe) from [<80667ed4>] (bus_probe_device+0x94/0x9c)
[   12.706885] [<80667ed4>] (bus_probe_device) from [<8066845c>] (deferred_probe_work_func+0x70/0x188)
[   12.706893] [<8066845c>] (deferred_probe_work_func) from [<80149b6c>] (process_one_work+0x20c/0x588)
[   12.706902] [<80149b6c>] (process_one_work) from [<8014aef8>] (worker_thread+0x2bc/0x5c4)
[   12.706909] [<8014aef8>] (worker_thread) from [<801505b4>] (kthread+0x16c/0x174)
[   12.706916] [<801505b4>] (kthread) from [<80109174>] (ret_from_fork+0x14/0x20)
[   12.706918] handlers:
[   12.706923] [<(ptrval)>] dwc_otg_common_irq
[   12.706928] [<(ptrval)>] dwc_otg_pcd_irq
[   12.706933] [<(ptrval)>] usb_hcd_irq
[   12.706936] Disabling IRQ #39

[   17.635063] Indeed it is in host mode hprt0 = 00001101
[   17.827038] usb 1-1: device descriptor read/64, error -110
[   33.251035] Indeed it is in host mode hprt0 = 00001101
[   33.443031] usb 1-1: device descriptor read/64, error -110
[   33.555067] Indeed it is in host mode hprt0 = 00001101
[   33.747033] usb 1-1: new high-speed USB device number 3 using dwc_otg
[   38.883036] Indeed it is in host mode hprt0 = 00001101
[   39.075032] usb 1-1: device descriptor read/64, error -110
[   54.499043] Indeed it is in host mode hprt0 = 00001101
[   54.691032] usb 1-1: device descriptor read/64, error -110
[   54.803049] usb usb1-port1: attempt power cycle

6by9, 2019-09-18:
> The kernel branch is rpi-4.19.y.

> [ 12.706592] CPU: 0 PID: 31 Comm: kworker/0:1 Not tainted 4.15.0-1032-raspi2

So 4.19 or 4.15?

imgaolong, 2019-09-18:
It is a mistake, it is log message of Ubuntu Mate, I forget to switch to the new kernel of 4.19.

With dwc otg, the usb of P3B+ is working well, just as in 64 bit mode.

closed 2019-09-18 by imgaolong