raspberrypi/linux #3242: Unable to handle kernel paging request

mstekker, 2019-09-19:
Hi,

I am getting a "Unable to handle kernel paging request" every several hours. The system reboots after the message but stays frozen. Only power off / on brings it up.

====== system ======
Original PI power supply

Pi3B+

Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2

Aug 15 2019 12:06:42 
Copyright (c) 2012 Broadcom
version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start)

Linux sensor100 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

===== syslog =====
Sep 19 00:04:27 sensor100 kernel: [11576.822720] Unable to handle kernel paging request at virtual address f2004000
Sep 19 00:04:27 sensor100 kernel: [11576.823675] pgd = c7bd51ce
Sep 19 00:04:27 sensor100 kernel: [11576.824580] [f2004000] *pgd=00000000
Sep 19 00:04:27 sensor100 kernel: [11576.825489] Internal error: Oops: 5 [#1] SMP ARM
Sep 19 00:04:27 sensor100 kernel: [11576.826408] Modules linked in: cmac rfcomm bnep 8021q garp stp llc arc4 rt2800usb rt2800lib rt2x00usb rt2x00lib mac80211 sha256_generic btusb cfg80211 btrtl btintel btbcm crc_ccitt bluetooth ecdh_generic rfkill bcm2835_codec(C) snd_bcm2835(C) bcm2835_v4l2(C) raspberrypi_hwmon v4l2_mem2mem snd_pcm bcm2835_mmal_vchiq(C) videobuf2_dma_contig v4l2_common snd_timer videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common snd vc_sm_cma(C) i2c_bcm2835 videodev media uio_pdrv_genirq uio fixed rtc_ds1307 hwmon i2c_dev ip_tables x_tables ipv6
Sep 19 00:04:27 sensor100 kernel: [11576.834020] CPU: 3 PID: 9884 Comm: python Tainted: G         C        4.19.66-v7+ #1253
Sep 19 00:04:27 sensor100 kernel: [11576.836392] Hardware name: BCM2835
Sep 19 00:04:27 sensor100 kernel: [11576.837586] PC is at __kmalloc+0xac/0x288
Sep 19 00:04:27 sensor100 kernel: [11576.838791] LR is at __kmalloc+0x60/0x288
Sep 19 00:04:27 sensor100 kernel: [11576.839945] pc : [<80294db8>]    lr : [<80294d6c>]    psr: a0000093
Sep 19 00:04:27 sensor100 kernel: [11576.841120] sp : 9fc9dbd0  ip : 9fc9dbd0  fp : 9fc9dc14
Sep 19 00:04:27 sensor100 kernel: [11576.842283] r10: b9c01e40  r9 : 80656e28  r8 : 00488020
Sep 19 00:04:27 sensor100 kernel: [11576.843448] r7 : b9c01e40  r6 : 00000038  r5 : 00480020  r4 : f2004000
Sep 19 00:04:27 sensor100 kernel: [11576.844626] r3 : 00000000  r2 : 80c97a60  r1 : 396f0000  r0 : 9fc9dbd0
Sep 19 00:04:27 sensor100 kernel: [11576.845771] Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
Sep 19 00:04:27 sensor100 kernel: [11576.846923] Control: 10c5383d  Table: 37ca806a  DAC: 00000055
Sep 19 00:04:27 sensor100 kernel: [11576.848097] Process python (pid: 9884, stack limit = 0xf898a26b)
Sep 19 00:04:27 sensor100 kernel: [11576.849288] Stack: (0x9fc9dbd0 to 0x9fc9e000)
Sep 19 00:04:27 sensor100 kernel: [11576.850449] dbc0:                                     afb8dcb0 afb8dcac 00000000 0011d188
Sep 19 00:04:27 sensor100 kernel: [11576.852689] dbe0: 00000009 00000001 0000000b 00000038 00480020 00000000 00000001 b96b0200
Sep 19 00:04:27 sensor100 kernel: [11576.854896] dc00: 00000002 00000000 9fc9dc24 9fc9dc18 80656e28 80294d18 9fc9dc4c 9fc9dc28
Sep 19 00:04:27 sensor100 kernel: [11576.857094] dc20: 8064b398 80656e10 00000000 afb8dc80 b8f83100 00480020 80d04d48 b9de0400
Sep 19 00:04:27 sensor100 kernel: [11576.859320] dc40: 9fc9dcac 9fc9dc50 8064bb78 8064b36c 9fc9dc84 9fc9dc60 80116f9c 80116e08
Sep 19 00:04:27 sensor100 kernel: [11576.861677] dc60: 8011ad6c 9fc9dcd0 00000000 b8d07100 00000000 b8d07120 9fc9dcac 66dda9d3
Sep 19 00:04:27 sensor100 kernel: [11576.864215] dc80: 8061c9f4 b8f83100 b96b0200 80d04d48 00000000 00480020 b8f83108 00000000
Sep 19 00:04:27 sensor100 kernel: [11576.866850] dca0: 9fc9dd5c 9fc9dcb0 8061d17c 8064bad8 9fc9dd14 9fc9dcc0 80168298 808367b4
Sep 19 00:04:27 sensor100 kernel: [11576.869636] dcc0: 00000000 9fc9dcd0 20000013 80d04d48 00000000 00000000 00000000 0000001c
Sep 19 00:04:27 sensor100 kernel: [11576.872484] dce0: 00000000 a383cb00 00000000 00000000 00000003 a4300780 00000019 00000000
Sep 19 00:04:27 sensor100 kernel: [11576.875403] dd00: 9fc9dcd8 802c083c 00000100 00000200 ae184014 a4300e40 00000019 80831344
Sep 19 00:04:27 sensor100 kernel: [11576.878403] dd20: 00000000 006000c0 9fc9dd44 66dda9d3 80831344 b8f83100 00480020 00000002
Sep 19 00:04:27 sensor100 kernel: [11576.881431] dd40: 00000200 b8d65000 00000040 00000000 9fc9dda4 9fc9dd60 8061ed60 8061d0b0
Sep 19 00:04:27 sensor100 kernel: [11576.884457] dd60: 9fc9dd8c 9fc9dd70 80201ed0 80202f48 00000002 00000000 afb8dc80 9fc9de78
Sep 19 00:04:27 sensor100 kernel: [11576.887489] dd80: afb8dc80 80e3ff68 b9675580 afb8dc94 afb8dc90 00000000 9fc9de1c 9fc9dda8
Sep 19 00:04:27 sensor100 kernel: [11576.890518] dda0: 8062b51c 8061eab4 808367e4 b8d650ac a4300180 0000000c 80e3ff68 afb8dc80
Sep 19 00:04:27 sensor100 kernel: [11576.893546] ddc0: 9fc9ddec afb8dc90 00000000 00000000 00000000 00000001 00000000 00000000
Sep 19 00:04:27 sensor100 kernel: [11576.896571] dde0: 00000000 00000001 73bb4438 b8d07100 b8d650ac 80d04d48 b8d65078 b8d650ac
Sep 19 00:04:27 sensor100 kernel: [11576.899598] de00: b8d65000 0000000c 802c550a afb8dc80 9fc9ded4 9fc9de20 8062bdf0 8062a6c8
Sep 19 00:04:27 sensor100 kernel: [11576.902625] de20: 9fc9de94 9fc9de30 80154fa8 66dda9d3 b6c5ca00 80d04d48 80d04d48 b8f20f60
Sep 19 00:04:27 sensor100 kernel: [11576.905651] de40: 73bb4438 00000040 5d82a9eb 00000000 00000000 af1c8e40 9fc9de8c 9fc9de68
Sep 19 00:04:27 sensor100 kernel: [11576.908676] de60: 8082a25c 80822b18 00000003 af1c8e40 af1c8e40 00000000 00008303 00000000
Sep 19 00:04:27 sensor100 kernel: [11576.911701] de80: 00000000 748009f0 00000040 00000000 00000000 00000000 00000000 00000000
Sep 19 00:04:27 sensor100 kernel: [11576.914724] dea0: 748efe38 66dda9d3 8019a718 80d04d48 73bb4438 b8f20f60 a4300180 0000000c
Sep 19 00:04:27 sensor100 kernel: [11576.917752] dec0: 9fc9c000 00000036 9fc9df6c 9fc9ded8 802bf870 8062b540 af1c8e40 802fe8e0
Sep 19 00:04:27 sensor100 kernel: [11576.920775] dee0: 00000001 af1c8e40 9fc9dfa4 9fc9def8 802fe8e0 802ac868 00000000 00000000
Sep 19 00:04:27 sensor100 kernel: [11576.923803] df00: 00000000 00000000 00000001 00000000 00000000 00000001 00000000 00000000
Sep 19 00:04:27 sensor100 kernel: [11576.926827] df20: 00000000 00000000 a4300e40 802cbb44 73bb4438 73bb4438 00000000 66dda9d3
Sep 19 00:04:27 sensor100 kernel: [11576.929852] df40: 9fc9df5c a4300181 73bb4438 a4300180 802c550a 0000000c 9fc9c000 00000036
Sep 19 00:04:27 sensor100 kernel: [11576.932880] df60: 9fc9df94 9fc9df70 802bfffc 802bf7c0 00db8200 73bb4438 748efe38 00000000
Sep 19 00:04:27 sensor100 kernel: [11576.935902] df80: 00000036 801011c4 9fc9dfa4 9fc9df98 802c003c 802bffc4 00000000 9fc9dfa8
Sep 19 00:04:27 sensor100 kernel: [11576.938924] dfa0: 80101000 802c0030 73bb4438 748efe38 0000000c 802c550a 73bb4438 00db8200
Sep 19 00:04:27 sensor100 kernel: [11576.941948] dfc0: 73bb4438 748efe38 00000000 00000036 00000001 00000001 00000000 00000000
[... 1346 lines cut ...]
Sep 18 23:51:59 sensor100 systemd[711]: Reached target Paths.
Sep 18 23:51:59 sensor100 systemd[711]: Listening on GnuPG cryptographic agent and passphrase cache.
Sep 18 23:51:59 sensor100 systemd[711]: Reached target Sockets.
Sep 18 23:51:59 sensor100 systemd[711]: Reached target Basic System.
Sep 18 23:51:59 sensor100 systemd[711]: Reached target Default.
Sep 18 23:51:59 sensor100 systemd[711]: Startup finished in 380ms.
Sep 18 23:51:59 sensor100 systemd[1]: Started User Manager for UID 1000.
Sep 18 23:51:59 sensor100 systemd[1]: Started Session c1 of user pi.
Sep 18 23:51:59 sensor100 dhcpcd[450]: eth0: no IPv6 Routers available
Sep 18 23:52:13 sensor100 systemd[1]: systemd-fsckd.service: Succeeded.

==== system not responding to anything anymore, manual power off ====

Sep 19 04:43:34 sensor100 systemd-timesyncd[264]: Synchronized to time server for the first time 207.180.204.206:123 (2.debian.pool.ntp.org).
Sep 19 04:43:34 sensor100 systemd[1]: Starting Daily man-db regeneration...
Sep 19 04:43:34 sensor100 systemd[1]: Starting Rotate log files...
Sep 19 04:43:34 sensor100 systemd[1]: Starting Clean php session files...
Sep 19 04:43:35 sensor100 systemd[1]: logrotate.service: Succeeded.
Sep 19 04:43:35 sensor100 systemd[1]: Started Rotate log files.

Thanks for any clues.

Regards,

Mart

JamesH65, 2019-09-19:
What do you have plugged in to the USB ports, or anywhere else?

pelwell, 2019-09-19:
This appears to be memory corruption, which invariably has one of two causes - a bad driver, or a low voltage supply. Answering James's question will help to narrow down the former. For the latter, while (or after) running a typical workload, run `vcgencmd get_throttled` and report the output.

mstekker, 2019-09-19:
Thanks for your answers. The UBS  dongles are:

Bus 001 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 005: ID 148f:5572 Ralink Technology, Corp. RT5572 Wireless Adapter
Bus 001 Device 007: ID 0451:16ae Texas Instruments, Inc. 
(This is a CC2531 ZigBee USB dongle)

Integrated Bluetooth and WiFi are disabled (config.txt / overlays)

vcgencmd get_throttled result:
throttled=0x80000

No voltage warnings in syslog.

pelwell, 2019-09-23:
> No voltage warnings in syslog.

That's no guarantee that voltage isn't an issue if the load changes rapidly - this many dongles should be on a powered hub - but let's assume for now that it isn't. Is it feasible to remove the devices one at a time and test the stability of the remaining system, or does that reduce the functionality so much as to completely break your use cases?

mstekker, 2019-09-23:
Hi pelwell,

In the mean time I have tried a powered hub. But with no result. It appears that the issue/exception is not happening with the CC2531 dongle disconnected.  Even when the other devices are directly connected to the PI.

If the CC2531 is connected but no software is polling its data, the system also runs very stable. So I am very sure the CC2531 is causing the troubles - which for me is an issue because the unit should monitor ZigBee devices (for example Philips HUE).

For monitoring ZigBee I use the "KillerBee" Python framework which on it's turn is using "PyUSB" to poll the CC2531.  PyUSB is using / is a wrapper for  libusb. So I am afraid there could be an issue in this chain of use regarding to the kernel crash.

pelwell, 2019-09-23:
That's a useful result. We can't spend much time looking into an individual driver that (so far) only affects one person (that we know of), but if you can narrow down the issue any further we can see if there are any changes to the driver in newer kernels that migjt help.

JamesH65, 2019-12-20:
@mstekker Did you have any further results or find any solutions?

real3856, 2020-01-10:
Hi @pelwell,
I was wondering if you could tell me why a low voltage supply may cause this problem?
I also get this problem during power on/off testing too.

pelwell, 2020-01-10:
Because below a certain voltage the internal logic of the SoC starts to fail because the operations don't complete in time for the next clock tick.

mstekker, 2020-03-08:
> @mstekker Did you have any further results or find any solutions?
Hi JamesH65,

After testing with several ac adapters, several Raspberries, several active USB hubs,  several CC2531 sticks and the newest raspbian kernel updates the problem still exists as today. An Intel NUC with Debian works fine, so I am stuck regarding not enough knowledge about the kernel.

Best regards,

Mart

mstekker, 2020-03-11:
Hi,
After updating to the newest kernel version, same hardware, the log stays clean from error and raspbian is working fine for several days now. I do not know what changed, but the problem is solved as by today with the latest kernel / firmware.

closed 2020-03-11 by mstekker

pelwell, 2020-03-11:
I suspect that a change in dynamic power management in the firmware has fixed it for you.

mstekker, 2020-03-11:
> I suspect that a change in dynamic power management in the firmware has fixed it for you.

Thanks for the information!

IIIdefconIII, 2024-03-29:
for me was because i had the wrong image to be honest, mybad