raspberrypi/linux #3243: dwcotg: USB Host corrupted the kernel in 32 bit mode

imgaolong, 2019-09-19:
For 4.19.y-rt branch, in 32 bit mode, if I choose full rt schedule policy, the kernel will stop booting. Dmesg information is shown below.
On the other hand, if I disable Synopsis DWC host support options in device drivers, the kernel will boot OK, but the usb mouse and keyboard will not work, and the graphic user interface will not show. The problem is related with dwcotg?
 
I am using pi3b+, and the same problem on pi3b, the distribution is ubuntu mate 18.04 for armhf. Everything seemed ok if I run 4.19.y-rt in 64 bit mode of full rt schedule policy on a ubuntu mate 18.04 for arm64.

dmesg show like :
[   15.024735] ------------[ cut here ]------------
[   15.024738] ------------[ cut here ]------------
[   15.024765] WARNING: CPU: 3 PID: 82 at kernel/sched/core.c:7256 migrate_disable+0x224/0x24c
[   15.024770] Modules linked in:
[   15.024791] CPU: 3 PID: 82 Comm: irq/56-dwc_otg_ Not tainted 4.19.71-rt24-v7+ #142
[   15.024795] Hardware name: BCM2835
[   15.024826] [<80111ea0>] (unwind_backtrace) from [<8010d3f4>] (show_stack+0x20/0x24)
[   15.024843] [<8010d3f4>] (show_stack) from [<8083ac20>] (dump_stack+0xbc/0x100)
[   15.024861] [<8083ac20>] (dump_stack) from [<80120c6c>] (__warn+0x104/0x11c)
[   15.024876] [<80120c6c>] (__warn) from [<80120dbc>] (warn_slowpath_null+0x50/0x58)
[   15.024890] [<80120dbc>] (warn_slowpath_null) from [<8014e8e0>] (migrate_disable+0x224/0x24c)
[   15.024910] [<8014e8e0>] (migrate_disable) from [<80183604>] (vprintk_emit+0x260/0x2b8)
[   15.024928] [<80183604>] (vprintk_emit) from [<80183840>] (vprintk_default+0x44/0x74)
[   15.024945] [<80183840>] (vprintk_default) from [<801845c4>] (vprintk_func+0x120/0x238)
[   15.024961] [<801845c4>] (vprintk_func) from [<80183d30>] (printk+0x40/0x5c)
[   15.024977] [<80183d30>] (printk) from [<80120d9c>] (warn_slowpath_null+0x30/0x58)
[   15.024991] [<80120d9c>] (warn_slowpath_null) from [<8014eafc>] (migrate_enable+0x1f4/0x478)
[   15.025007] [<8014eafc>] (migrate_enable) from [<80127384>] (__local_bh_enable+0x84/0xb0)
[   15.025024] [<80127384>] (__local_bh_enable) from [<80186e48>] (irq_forced_thread_fn+0x78/0xac)
[   15.025038] [<80186e48>] (irq_forced_thread_fn) from [<801871f8>] (irq_thread+0x138/0x22c)
[   15.025051] [<801871f8>] (irq_thread) from [<80143c2c>] (kthread+0x140/0x170)
[   15.025065] [<80143c2c>] (kthread) from [<801010b4>] (ret_from_fork+0x14/0x20)
[   15.025071] Exception stack(0xaea87fb0 to 0xaea87ff8)
[   15.025080] 7fa0:                                     00000000 00000000 00000000 00000000
[   15.025091] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.025099] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   15.025106] ---[ end trace 0000000000000002 ]---
[   15.025121] WARNING: CPU: 3 PID: 82 at kernel/sched/core.c:7290 migrate_enable+0x1f4/0x478
[   15.025125] Modules linked in:
[   15.025141] CPU: 3 PID: 82 Comm: irq/56-dwc_otg_ Tainted: G        W         4.19.71-rt24-v7+ #142
[   15.025144] Hardware name: BCM2835
[   15.025159] [<80111ea0>] (unwind_backtrace) from [<8010d3f4>] (show_stack+0x20/0x24)
[   15.025172] [<8010d3f4>] (show_stack) from [<8083ac20>] (dump_stack+0xbc/0x100)
[   15.025184] [<8083ac20>] (dump_stack) from [<80120c6c>] (__warn+0x104/0x11c)
[   15.025197] [<80120c6c>] (__warn) from [<80120dbc>] (warn_slowpath_null+0x50/0x58)
[   15.025211] [<80120dbc>] (warn_slowpath_null) from [<8014eafc>] (migrate_enable+0x1f4/0x478)
[   15.025225] [<8014eafc>] (migrate_enable) from [<80127384>] (__local_bh_enable+0x84/0xb0)
[   15.025240] [<80127384>] (__local_bh_enable) from [<80186e48>] (irq_forced_thread_fn+0x78/0xac)
[   15.025252] [<80186e48>] (irq_forced_thread_fn) from [<801871f8>] (irq_thread+0x138/0x22c)
[   15.025263] [<801871f8>] (irq_thread) from [<80143c2c>] (kthread+0x140/0x170)
[   15.025275] [<80143c2c>] (kthread) from [<801010b4>] (ret_from_fork+0x14/0x20)
[   15.025280] Exception stack(0xaea87fb0 to 0xaea87ff8)
[   15.025288] 7fa0:                                     00000000 00000000 00000000 00000000
[   15.025298] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   15.025307] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   15.025312] ---[ end trace 0000000000000003 ]---

imgaolong, 2019-09-20:
when booting in a full rt schedule policy, the kernel stopped with such error:

[   49.754351] mmc1: Timeout waiting for hardware interrupt.
[   50.225627] Sending NMI from CPU 3 to CPUs 1:
[   59.994351] mmc1: Timeout waiting for hardware interrupt.
[   60.226866] kworker/2:0     D    0    34      2 0x00000000
[   60.226893] Workqueue: usb_hub_wq hub_event
[   60.226908] kworker/2:0     D    0    34      2 0x00000000
[   60.226924] Workqueue: usb_hub_wq hub_event
[   60.226938] rcu: rcu_preempt kthread starved for 2002 jiffies! g437 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x0 ->cpu=3
[   60.226941] rcu: RCU grace-period kthread stack dump:
[   60.226944] rcu_preempt     R  running task        0    12      2 0x00000000
[   60.226971] [<808529cc>] (__schedule) from [<80852dc8>] (schedule+0x64/0x10c)
[   60.226988] [<80852dc8>] (schedule) from [<80855ca4>] (schedule_timeout+0x228/0x4ec)
[   60.227007] [<80855ca4>] (schedule_timeout) from [<801981d8>] (rcu_gp_kthread+0x690/0xb30)
[   60.227023] [<801981d8>] (rcu_gp_kthread) from [<80143c2c>] (kthread+0x140/0x170)
[   60.227038] [<80143c2c>] (kthread) from [<801010b4>] (ret_from_fork+0x14/0x20)
[   60.227044] Exception stack(0xb9d55fb0 to 0xb9d55ff8)
[   60.227052] 5fa0:                                     00000000 00000000 00000000 00000000
[   60.227062] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   60.227071] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   60.227181] mmc0: timeout waiting for hardware interrupt.
[   60.457814] [029349a0] TCM> aec9f3b0 0
[   60.466585] [029349a1] FDA> aec9f2d0 0
[   60.475226] [029349a1] DMA> 0 0
[   60.483141] [029349a6] TSK< aec9f2d0 0
[   60.491666] [029349bc] TSK> aec9f2d0 0
[   60.500047] [029349fd] PRD< aec9f7b0 0
[   60.508403] [02934a01] PRD1 b9ebfc40 0
[   60.516659] [02934a43] PRD2 38 0
[   60.524388] [02934b63] PRD3 aeb70004 0
[   60.532584] [02934b66] PDM> aec9f7b0 0
[   60.540698] [02934b6a] REQ< aec9f6d0 10801
[   60.548990] [02934b6c] CMD< 17 200
[   60.556529] [02934b74] FCM< aec9f6d0 aec9f714
[   60.565020] [02934b77] RSP  900 0
[   60.572433] [02934b78] CMD< 12 a00
[   60.579891] [02934b7b] CMDD 200 200
[   60.587325] [02934b7c] SDMA aec9f7b0 b9ebfc40
[   60.595587] [02934b80] FCM< aec9f6d0 aec9f748
[   60.603725] [02934b82] RSP  900 0
[   60.610694] [02934b83] FCM> aec9f6d0 0
[   60.618003] [02934b84] FCM> aec9f6d0 0
[   60.625160] [02934b86] CMD  12 200
[   60.631832] [02934b87] REQ> aec9f6d0 0
[   60.638847] [029378da] DMA< aec9f7b0 1
[   60.645725] [029378db] DMA  52 10834
[   60.652340] [02937924] FDA< aec9f6d0 0
[   60.658946] [02937926] TCM< aec9f7b0 0
[   60.665272] [02937927] WTC< 10804 0
[   60.671163] [02937928] WTC> 10804 0
[   60.676966] [0293792a] TCM> aec9f7b0 0
[   60.683012] [0293792b] FDA> aec9f6d0 0
[   60.689071] [0293792c] DMA> 0 0
[   60.694491] [02937931] TSK< aec9f6d0 0
[   60.700540] [02937948] TSK> aec9f6d0 0
[   60.706589] [02937978] PRD< aec9f3b0 0
[   60.712617] [0293797b] PRD1 b9ebfc40 0
[   60.718666] [02937982] PRD2 1 0
[   60.724066] [02937992] PRD3 aec10984 0
[   60.730085] [02937995] PDM> aec9f3b0 0
[   60.736096] [02937999] REQ< aec9f2d0 10801
[   60.742455] [0293799b] CMD< 17 8
[   60.747900] [029379a2] FCM< aec9f2d0 aec9f314
[   60.754531] [029379a4] RSP  900 0
[   60.760094] [029379a6] CMD< 12 65000
[   60.765924] [029379a8] CMDD 8 200
[   60.771503] [029379aa] SDMA aec9f3b0 b9ebfc40
[   60.778150] [029379ac] FCM< aec9f2d0 aec9f348
[   60.784791] [029379ad] RSP  900 0
[   60.790356] [029379ae] FCM> aec9f2d0 0
[   60.796364] [029379af] FCM> aec9f2d0 0
[   60.802370] [029379b0] CMD  12 200
[   60.808040] [029379b1] REQ> aec9f2d0 0
[   60.814015] [02937c49] DMA< aec9f3b0 1
[   60.820026] [02937c4a] DMA  52 10834
[   60.825847] [02937c4f] FDA< aec9f2d0 0
[   60.831808] [02937c50] TCM< aec9f3b0 0
[   60.837803] [02937c51] WTC< 10804 0
[   60.843522] [02937c52] WTC> 10804 0
[   60.849236] [02937c54] TCM> aec9f3b0 0
[   60.855224] [02937c54] FDA> aec9f2d0 0
[   60.861202] [02937c55] DMA> 0 0
[   60.866562] [02937c5a] TSK< aec9f2d0 0
[   60.872538] [02937c73] TSK> aec9f2d0 0
[   60.878521] [02938ba5] PRD< aec9f3b0 0
[   60.884500] [02938ba9] PRD1 b9ebfc40 0
[   60.890478] [02938bde] PRD2 1a 0
[   60.895923] [02938c7b] PRD3 aed82204 0
[   60.901921] [02938c81] PDM> aec9f3b0 0
[   60.907926] [02938c85] REQ< aec9f2d0 10801
[   60.914259] [02938c87] CMD< 17 200
[   60.919918] [02938c8e] FCM< aec9f2d0 aec9f314
[   60.926539] [02938c91] RSP  900 0
[   60.932080] [02938c92] CMD< 12 63c00
[   60.937917] [02938c95] CMDD 200 200
[   60.943653] [02938c97] SDMA aec9f3b0 b9ebfc40
[   60.950287] [02938c9a] FCM< aec9f2d0 aec9f348
[   60.956915] [02938c9b] RSP  900 0

imgaolong, 2019-09-20:
I wonder why dwcotg of Synopsis DWC host option in Kconfig is set to bool, not a tristate? I thought I might be a little easier if  I could debug it as a module.

JamesH65, 2019-12-20:
@imgaolong IIRC, there have been some recent fixes for the Rt kernel, do they fix your problem? You may need to rpi-update.

guysoft, 2020-03-01:
I am stuck building too for other reasons: https://github.com/raspberrypi/linux/issues/3487