raspberrypi/linux #3245: eth_led0 for the RPI4

satmandu, 2019-09-19:
Earlier Raspberry Pi boards have had device tree options for adjusting the behavior of the ethernet port LEDs, as mentioned for instance here: https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/README

Turning the LED off would for instance require this:

```dtparam=eth_led0=14```

That functionality doesn't appear to be configured for the BCM54213PE on the Raspberry Pi 4B, though configuring the LEDs is mentioned in commits such as https://github.com/raspberrypi/linux/commit/af502b2acafb23364b5757f01ae2c43e2943f195

Could we have the same functionality hooked up for the 4B?

satmandu, 2019-10-11:
As per https://www.raspberrypi.org/forums/viewtopic.php?p=1540845#p1540845 it appears that this can disable the LEDs on the RPI4 ethernet port thus:

https://github.com/raspberrypi/linux/commit/fef235ba38269f198cb93383f7b4a6b675dbf7e3
```
diff --git a/drivers/net/phy/broadcom.c b/drivers/net/phy/broadcom.c
index a919a6a4c970f..52d0582688104 100644
--- a/drivers/net/phy/broadcom.c
+++ b/drivers/net/phy/broadcom.c
@@ -61,9 +61,14 @@ static void bcm54213pe_config_init(struct phy_device *phydev)
 	val |= BCM54XX_SHD_LEDCTL_ACTLINK_EN;
 	bcm_phy_write_shadow(phydev, BCM54XX_SHD_LEDCTL, val);
 
-	/* Set ACTIVITY on LED "1" output, LINKSPD[1] on LED "3" output */
-	val = BCM5482_SHD_LEDS1_LED1(BCM_LED_SRC_ACTIVITYLED) |
-		BCM5482_SHD_LEDS1_LED3(BCM_LED_SRC_LINKSPD1);
+	/* Some day there may be a DT wired toggle for this. Disable for now.
+	 * Set ACTIVITY on LED "1" output, LINKSPD[1] on LED "3" output
+	 * val = BCM5482_SHD_LEDS1_LED1(BCM_LED_SRC_ACTIVITYLED) |
+	 *	BCM5482_SHD_LEDS1_LED3(BCM_LED_SRC_LINKSPD1);
+	 */
+	/* Turn off LED "1" and LED "3" unconditionally */
+	val = BCM5482_SHD_LEDS1_LED1(BCM_LED_SRC_OFF) |
+		BCM5482_SHD_LEDS1_LED3(BCM_LED_SRC_OFF);
 	bcm_phy_write_shadow(phydev, BCM5482_SHD_LEDS1, val);
 }
```

JamesH65, 2019-11-14:
Fixed by #3316

closed 2019-11-14 by JamesH65