raspberrypi/linux #3248: /usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz points to a private git commit

agsha, 2019-09-20:
I recently bought a rpi 4. I'm trying to get and compile the kernel source code.
I am using these commands to identify the git commit for my kernel

```
[sharath@raspberrypi /mnt/SEAGATE/pi/linux-src/linux]$ FIRMWARE_HASH=$(zgrep "* firmware as of" /usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz | head -1 | awk '{ print $5 }')
[sharath@raspberrypi /mnt/SEAGATE/pi/linux-src/linux]$ echo $FIRMWARE_HASH
e5b11a9d32cf4388b09097c33ab7bffbe1ca721c
```

But it seems like the commit `e5b11a9d32cf4388b09097c33ab7bffbe1ca721c` is a private commit. The commit is not present in github. How can I obtain this commit?

pelwell, 2019-09-20:
Our bootloader contains lots of Broadcom IP, so it's not Open Source.

popcornmix, 2019-09-20:
For me, I'm getting e10b2880e26708afc0b32a485c1231207037fc45 which is a valid commit here:
https://github.com/raspberrypi/firmware/commit/e10b2880e26708afc0b32a485c1231207037fc45

You can get the linux git hash from:
https://github.com/raspberrypi/firmware/blob/e10b2880e26708afc0b32a485c1231207037fc45/extra/git_hash

Where did you install raspbian from? Have you updated it?

agsha, 2019-09-20:
the rspbian came pre-installed on the sd-card in the [desktop-kit](https://www.raspberrypi.org/products/raspberry-pi-4-desktop-kit/)

agsha, 2019-09-20:
I did a `sudo apt-get dist-upgrade` and now my changelog is pointing to the correct versio `e10b2880e26708afc0b32a485c1231207037fc45`

Thank you @popcornmix

closed 2019-09-20 by agsha