raspberrypi/linux #3249: where is mkknlimg?

jeffersonchua, 2019-09-21:
Can't find mkknlimg after latest 5.3.0 pull yesterday. Was still in rpi-5.3.0-rc8.

Thanks,
Jeff

pelwell, 2019-09-21:
It's been deprecated for a long time, and the current firmware makes no use of the trailer it used to add, so I removed it as part of the continuous drive to converge with upstream.

jeffersonchua, 2019-09-21:
> It's been deprecated for a long time, and the current firmware makes no
use of the trailer it used to add, so I removed it as part of the
continuous drive to converge with upstream.

Hi Phil,

I could be doing something wrong then. If I just copy arch/arm/boot/zImage
to /boot/kernel7.img, rpi3 would boot but complained about invalid ELF
header and all sorts of errors (even gcc gets stuck complying), but with
mkknlimg, all errors went away ...

This is one sample error ... but no issue with mkknlimg ...

01-Jan-1970 07:30:31.974 error:25066067:DSO support
routines:DLFCN_LOAD:could not load the shared
library:dso_dlfcn.c:187:filename(/local/ssl/lib/engines/libgost.so):
/local/ssl/lib/engines/libgost.so: invalid ELF header

01-Jan-1970 07:30:31.974 error:25070067:DSO support routines:DSO_load:could
not load the shared library:dso_lib.c:233:

01-Jan-1970 07:30:31.975 error:260B6084:engine routines:DYNAMIC_LOAD:dso
not found:eng_dyn.c:467:

01-Jan-1970 07:30:31.975 error:2606A074:engine routines:ENGINE_by_id:no
such engine:eng_list.c:390:id=gost

My best.,
Jeff

pelwell, 2019-09-23:
Jeff,

I was a bit surprised that anyone was still using mkknlimg, but I guess once you have a build script that uses it there would no obvious reason to change it until somebody breaks it for you.

It's not at all clear what would be going wrong in your case. Our standard kernel build instructions (https://www.raspberrypi.org/documentation/linux/kernel/building.md) just copy the zImage:
```
sudo cp arch/arm/boot/zImage /boot/$KERNEL.img
```
and
```
sudo cp arch/arm/boot/zImage mnt/fat32/$KERNEL.img
```

The one thought I have is that it might be related to kernel file length. A side effect of adding the trailer is to round the kernel file length up to a multiple of 4, and some odd-length issues have been found and fixed in the firmware in the past. Which firmware are you running (`vcgencmd version`)?

jeffersonchua, 2019-09-23:
On Mon, Sep 23, 2019 at 5:30 PM Phil Elwell <notifications@github.com>
wrote:

> I was a bit surprised that anyone was still using mkknlimg, but I guess
> once you have a build script that uses it there would no obvious reason to
> change it until somebody breaks it for you.
>
> It's [...]
I tried that .. and zImage just won't work for my case. but after using
mkknlimg, it works fine.

> The one thought I have is that it might be related to kernel file length.
> A side effect of adding the trailer is to round the kernel file length up
> to a multiple of 4, and some odd-length issues [...]
# vcgencmd version
Sep 17 2019 21:15:22
Copyright (c) 2012 Broadcom
version d07eb79fbb9b871c8d4a695fbe048e4eae7b33bd (clean) (release) (start_x)

Could it be that I'm cross-compiling (crosstool-ng-1.23 bmc2709)?
   gcc version 6.3.0 (crosstool-NG crosstool-ng-1.23.0)

But the strange thing is mkknlimg works.

Thanks,
Jeff.

pelwell, 2019-09-23:
Can you post the output of `sudo vcdbg log msg` for a working kernel image, and one for a broken image if you can get it? Alternatively, if you have a serial port hooked up, put the following in config.txt and reboot:
```
enable_uart=1
uart_2ndstage=1
# the new name for pi3-disable-bt
dtoverlay=disable-bt 
```
The firmware debug output should appear on the serial port.

ultibohub, 2019-09-24:
>It's been deprecated for a long time, and the current firmware makes no use of the trailer it used to add, so I removed it as part of the continuous drive to converge with upstream.

Just wanted to add that while the kernel trailer might be deprecated for Linux kernels (and the original Pi 4 firmware failed to boot if it was present) the trailer is very important in allowing non Linux kernels such as bare metal to make full use of the Pi.

A big thanks to whoever recently fixed the Pi 4 firmware to properly support the kernel trailer, without it we would need to consider implementing device tree support which is a big task.

pelwell, 2019-09-24:
> the trailer is very important in allowing non Linux kernels such as bare metal to make full use of the Pi.

Can you expand on this a bit? While working on the loader recently I came to the conclusion that the trailer did nothing that wasn't achievable by other means. If that isn't the case then I'm happy to reconsider.

ultibohub, 2019-09-24:
> Can you expand on this a bit? While working on the loader recently I came to the conclusion that the trailer did nothing that wasn't achievable by other means. If that isn't the case then I'm happy to reconsider.

The primary thing we rely on the kernel trailer for is to disable device tree and enable atags, we get a couple of items from the atags but the most important one is the command line arguments (from cmdline.txt) which we use extensively for low level configuration of our kernels, things like driver parameters, kernel tuning and network options.

We are aware that you can disable device tree by removing all of the overlays from the SD card or by adding settings to config.txt but both of these options are vulnerable to accidental changes, for example while updating the firmware files a user might also copy the device tree overlays or the config.txt may get deleted or overwritten with a default version. 

Using the kernel trailer to indicate explicitly to the firmware that we want atags instead of device tree makes a robust and reliable model that isn't prone to accidental changes, if the kernel has sensible compiled defaults then even a missing cmdline.txt shouldn't result in a failure.

Hope that makes sense, we don't actually use the mkknlimg script as we have our own tool that adds the trailer after linking, we just need the firmware to continue supporting this option.

pelwell, 2019-09-25:
I understand your motivation now, but loading the kernel _after_ the DTB (as is done now) makes the firmware logic simpler and more flexible. It would be fairly easy (if a bit of a backward step) to restore the trailer parsing, but I'm loathe to change the sequence. Loading the DTB first allows the firmware to make certain decisions based on the content. It's conceptually simple to check the trailer in a separate pass before loading the DTB, but the TFTP network boot support would mean downloading the kernel twice, which is unoptimal.

One option is to retain the DTB-before-kernel order and use the trailer to decide whether to pass the ATAGs or DTB to the kernel (why they couldn't pass the DTB address in an ATAG or different register is beyond me...). The only danger (that I can think of) as far as you are concerned is that loading the DTB could result in the firmware initialising the wrong UART. Would that bother you?

Another option would be to use the kernel filename (or the presence of another file - say "NODT") to indicate that ATAGs are required.

What do you think?

popcornmix, 2019-09-25:
It feels like
```
disable_commandline_tags=1
device_tree=-
```
in config.txt does what you want. The user copying unwanted device tree files won't have any effect.
Sure if you delete config.txt then it won't work, but that is the same of many files in /boot.

pelwell, 2019-09-25:
I suspect @ultibohub's "kernel" needs the ATAGs, otherwise it could just ignore the DTB pointer, so `disable_commandline_tags=1` isn't going to work. `device_tree=` should be sufficient to ensure that no DTB is loaded and ATAGs are passed to the "kernel".

jeffersonchua, 2019-09-25:
I tried again with zImage and there's no error message. But, it just won't
compile! Perhaps it's the new kernel issue.

It seems I've bigger problem now. Somewhere, either the kernel is broken or
my gcc/glibc is broken. I can't even compile a helloWorld.c

I'm testing on 5.20 to see if I can compile a simple helloWorld.c

My best.,
Jeff

pelwell, 2019-09-25:
32- or 64-bit?

pelwell, 2019-09-25:
I can compile on both arm and arm64 kernels, without a trailer. I think your userspace has been trashed in some way - why not install a fresh Raspbian or Raspbian Lite on a spare card, update the kernel and confirm that compilation still works?

jeffersonchua, 2019-09-25:
On Wed, Sep 25, 2019 at 10:13 PM Phil Elwell <notifications@github.com>
wrote:

> 32- or 64-bit?
>
32-bit.

Went back 5.2.15 and compiled HelloWorld ok. Compiling other codes work as
well.

And yes, DON'T need mkkln.img anymore .. zImage works fine... so, something
seems broken between 5.2.15 and 5.3.x

Jeff.

jeffersonchua, 2019-09-25:
On Wed, Sep 25, 2019 at 11:59 PM Phil Elwell <notifications@github.com>
wrote:

> I can compile on both arm and arm64 kernels, without a trailer. I think
> your userspace has been trashed in some way - why not install a fresh
> Raspbian or Raspbian Lite on a spare card, update the kernel and confirm
> that compilation still works?
>

Could be.... I'm verifying ...

Jeff.

pelwell, 2019-09-25:
I'm trying on 5.3...

pelwell, 2019-09-25:
I get the same results on 5.3.0-v7l+ and 5.3.0-v8_:
```
$ gcc -o hello_world hello_world.c
$ ./hello_world
Hello World!
```
gcc is version 8.3.0 (Raspbian 8.3.0-6+rpi1)

ultibohub, 2019-09-26:
> One option is to retain the DTB-before-kernel order and use the trailer to decide whether to pass the ATAGs or DTB to the kernel (why they couldn't pass the DTB address in an ATAG or different [...]

That would be perfect and wouldn't cause any problem to us at all, being a rather small cog in a big wheel we expect that things will change without our knowledge so we always code defensively and we make no assumptions about the configuration being in a certain state. 

>Another option would be to use the kernel filename (or the presence of another file - say "NODT") to indicate that ATAGs are required.
What do you think?

Like config.txt that is still based on the fragile notion of a file existing or not which isn't a great solution, one of the primary drivers for working in bare metal is to maximize reliability and remove as many unknown elements as possible.

pelwell, 2019-09-26:
After chatting with @popcornmix I've come round to his viewpoint. Either the absence of a compatible .dtb or `device_tree=` in config.txt is sufficient to disable DT support and guarantee the kernel gets ATAGs, so both should provide you with enough of a safety net. All of these files are in the FAT partition, so remedying a fault is easy enough. There are many ways to prevent a Pi from booting by adding or removing files from the SD card, so I don't think that having one more is going to put your users in jeopardy.

ultibohub, 2019-09-26:
>After chatting with @popcornmix I've come round to his viewpoint. Either the absence of a compatible .dtb or device_tree= in config.txt is sufficient to disable DT support and guarantee the kernel gets ATAGs, so both should provide you with enough of a safety net.

Fair enough, we'l explore creating a minimal DTB parser to extract just the information we need and reduce our dependence on ATAGs which are clearly on the decline. That will be useful for aarch64 support as well.

jeffersonchua, 2019-09-28:
On Thu, Sep 26, 2019 at 12:22 AM Phil Elwell <notifications@github.com>
wrote:

> I get the same results on 5.3.0-v7l+ and 5.3.0-v8_:
>
> $ gcc -o hello_world hello_world.c
> $ ./hello_world
> Hello World!
>
> gcc is version 8.3.0 (Raspbian 8.3.0-6+rpi1)
>
So, it's the SD that was trashed. I rebuild a new one and upgraded to
gcc-8.3.1 and glibc-2.29 with linux-5.3.0. All works now without mkknlimg.

Sorry for the false alarm, and glad to get my working rpi3 without mkklnimg.

Thank you!

Jeff.

pelwell, 2019-10-08:
The ripples from this change will take a while to dissipate completely, but I think this issue is addressed sufficiently here.

closed 2019-10-08 by pelwell