raspberrypi/linux #3250: DTS overlay build error in 5.3.y

glassman81, 2019-09-22:
When attempting to compile the 5.3.y linux kernel with a custom overlay in the overlays folder, such as infineon-tpm-overlay.dts, the build fails.

I was able to do this with kernel 4.19.y, but now I get a makefile error error when attempting to cross-compile with the following command

`make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs overlays/infineon-tpm.dtbo -j8`

Here is what the error looks like:

`make: *** No rule to make target 'overlays/infineon-tpm.dtbo'.  Stop.`
`make: *** Waiting for unfinished jobs....`
  `UPD     include/generated/utsrelease.h`
`user@ubuntu:~/Documents/raspberry-pi/linux$`

Is there anything I'm missing? I know that without using `overlays/infineon-tpm.dtbo`, the build succeeds, but I should be able to make a custom overlay without issue, correct?

pelwell, 2019-09-23:
> but I should be able to make a custom overlay without issue, correct?

I'm not sure why you would expect to be able to use kernel's build system to build a file that hasn't been added to the makefiles. The fact that it used to work before the DTB build rules were made common (rather than architecture-specific) in 4.20 doesn't constitute sufficient reason.

glassman81, 2019-09-23:
`I'm not sure why you would expect to be able to use kernel's build system to build a file that hasn't been added to the makefiles.`

I would expect to be able to do it this way because the manufacturer has an official app note that has the user do it that way. If this is not an issue, that's fine, but if it's not too much trouble, could you point me in the right direction?

Your last sentence was a little unclear. Are you saying that 4.20 and onward doesn't allow this, but 4.19 and below did? Would it be sufficient to then just put a reference to the dts file inside the makefile located in the overlays folder?

pelwell, 2019-09-23:
>  Are you saying that 4.20 and onward doesn't allow this, but 4.19 and below did?

Yes, that is what I was saying, but it's fixed now. As I wrote in the commit message that resolves this issue:
```
    Before 4.20, it was possible to build an arbitrary overlay by copying
    it to arm/boot/dts/overlays/mytest-overlay.dts and running:
    
        make ARCH=arm overlays/mytest.dtbo
    
    In 4.20 the .dtb build rules were centralised, requiring the dowstream
    .dtbo build rules to be changed. They were, enough to support "make ...
    dtbs", but not sufficiently to allow this ad-hoc, one-off building of
    individual files.
```

glassman81, 2019-09-25:
Thanks for the help @pelwell.

closed 2019-09-25 by glassman81