raspberrypi/linux #3257: i2c speed halves on raspbian buster (Zero W, Pi3B, Pi4B), fine on raspbian stretch (PiZeroW, Pi3B)

drondistortion, 2019-09-30:
**Describe the bug**
I have a problem with I2C clock in Raspbian Buster (using either SMbus or i2c-dev). It defaults to 50 kHz on RPi 4B and to 62.5 kHz on RPi 3B and Zero W. However, if I set the speed i2c_arm_baudrate=200000 > /boot/config.txt on RPi 4B, then speed is 100 kHz.

But the strange thing is that for the first few seconds it is actually 200 kHz, so I2C slave devices that have some kind of initialization only work if I initialize them again after waiting for a couple of seconds in the program. The clock mismatch happens no matter what i2c devices I use, in fact it happens even if there are no devices on the bus:

**To reproduce**
hook up o-scope to SDA, SCL and:
python:
```python
import smbus
from time import sleep
bus = smbus.SMBus(1)

while True:
    try:
        bus.write_byte(9, 0)
        sleep(.1)
    except OSError:
        pass
```
c:
```c
#include <stdio.h>
#include <unistd.h>
#include <fcntl.h>
#include <linux/i2c-dev.h>
#include <sys/ioctl.h>
#define delay(A) usleep(A*1000)

char *i2cbus = "/dev/i2c-1";
int addr = 9;
int data[1] = {0};

void loop(void)
{
        int i2cfile = open(i2cbus, O_RDWR);
        ioctl(i2cfile, I2C_SLAVE, addr);
        write(i2cfile, data, 1);
}

int main()
{
        for(;;) {
                loop();
                delay(100);
        }
}
```
**Expected behaviour**
Expected 100 kHz clock when `i2c_arm_baudrate` is set to 100000

**Actual behaviour**
Clock initially is 100kHz, but after few seconds falls to 50 kHz on RPi4B and 62,5 kHz on RPi3B, PiZeroW.

**System**
* Pi4B, Pi3B+, PiZeroW (possibly any on Buster)
* Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage5
* Aug 15 2019 12:03:51
Copyright (c) 2012 Broadcom
version 9f8431fb7839c7f00f52b81f5822ddab2b31d0db (clean) (release) (start)
* Linux fourpi 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l GNU/Linux

**Additional context**
Everything works as expected on Raspbian Stretch.

drondistortion, 2019-10-01:
I accessed I2C1 registers and here's what i've found (reg=5 is clock divider):

Pi4B, Buster /boot/config.txt > i2c_arm_baudrate=100000 (50 kHz actual reading):

reg=0 val=       0
reg=1 val=      50
reg=2 val=       2
reg=3 val=       9
reg=4 val=       9
reg=5 val=    1388 = 5000 (DEC)
reg=6 val= 13804E2

Pi4B, Buster /boot/config.txt > i2c_arm_baudrate=200000 (100 kHz actual reading):

reg=0 val=       0
reg=1 val=      50
reg=2 val=       0
reg=3 val=       0
reg=4 val=       0
reg=5 val=     9C4 = 2500 (DEC)
reg=6 val=  9C0271

Pi3B, Buster /boot/config.txt > i2c_arm_baudrate=100000 (62.5 kHz actual reading), sending bytes to address 9:

reg=0 val=    8300
reg=1 val=      50
reg=2 val=       1
reg=3 val=       9
reg=4 val=       0
reg=5 val=     FA0 = 4000 (DEC)
reg=6 val=  FA03E8

Pi3B, Buster /boot/config.txt > i2c_arm_baudrate=160000 (100 kHz actual reading), sending bytes to address 9:

reg=0 val=    8300
reg=1 val=      50
reg=2 val=       1
reg=3 val=       9
reg=4 val=     132
reg=5 val=     9C4 = 2500 (DEC)
reg=6 val=  9C0271

Pi3B, Stretch /boot/config.txt > i2c_arm_baudrate=100000, sending bytes to address 9:

reg=0 val=    8300
reg=1 val=      50
reg=2 val=       1
reg=3 val=       9
reg=4 val=       0
reg=5 val=     9C4 = 2500 (DEC)
reg=6 val=  9C0271

this suggest that buster has incorrectly set dividers for i2c clock. tl;dr: Buster has 5000 by default for clock divider on RPi4B, 4000 by default on RPi3B but should be 2500. Right?

pelwell, 2019-10-01:
What is in your config.txt? Without an explicit `core_freq=500` and `core_freq_min=500` the firmware has to choose the divider assuming the higher clock speed to prevent running the bus too fast.

drondistortion, 2019-10-01:
Yep, everything works now. Thanks. Keep up the good work!

closed 2019-10-01 by drondistortion