raspberrypi/linux #3258: Wifi connection crashes when connection is reestablished

sub0ne, 2019-09-30:
I'm on Linux raspberrypi 4.19.69-v7+ #1256 and after running few hours without problems I got the following entries in my syslog. Interesting stuff begins at 00:46:53 where brcmfmac starts to fail and never returns. The problem occurs when wifi signal is lost and should be reestablished. In my case triggered by a channel change of my router (AVM Fritz Box).

The wlan0 carrier gets lost, is reaquired and then brcmfmac starts to mess up.

No hostapd installed, just simple wifi connection to my router.

```
Sep 23 00:00:22 raspberrypi rsyslogd:  [origin software="rsyslogd" swVersion="8.1901.0" x-pid="321" x-info="https://www.rsyslog.com"] rsyslogd w$
Sep 23 00:00:22 raspberrypi systemd[1]: logrotate.service: Succeeded.
Sep 23 00:00:22 raspberrypi systemd[1]: Started Rotate log files.
Sep 23 00:00:23 raspberrypi systemd[1]: man-db.service: Succeeded.
Sep 23 00:00:23 raspberrypi systemd[1]: Started Daily man-db regeneration.
Sep 23 00:17:01 raspberrypi CRON[15085]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Sep 23 00:45:50 raspberrypi dhcpcd[475]: wlan0: carrier lost
Sep 23 00:45:50 raspberrypi dhcpcd[475]: wlan0: deleting address 2003:e0:7f1f:8500:35d3:4c67:a30b:2ce8/64
Sep 23 00:45:50 raspberrypi dhcpcd[475]: wlan0: deleting route to 2003:e0:7f1f:8500::/64
Sep 23 00:45:50 raspberrypi dhcpcd[475]: wlan0: deleting default route via fe80::3a10:d5ff:fe60:4a2a
Sep 23 00:45:50 raspberrypi avahi-daemon[297]: Withdrawing address record for 2003:e0:7f1f:8500:35d3:4c67:a30b:2ce8 on wlan0.
Sep 23 00:45:50 raspberrypi avahi-daemon[297]: Leaving mDNS multicast group on interface wlan0.IPv6 with address 2003:e0:7f1f:8500:35d3:4c67:a30$
Sep 23 00:45:50 raspberrypi avahi-daemon[297]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::ef45:8c3a:42af:1e8d.
Sep 23 00:45:50 raspberrypi avahi-daemon[297]: Registering new address record for fe80::ef45:8c3a:42af:1e8d on wlan0.*.
Sep 23 00:45:50 raspberrypi avahi-daemon[297]: Withdrawing address record for 192.168.178.163 on wlan0.
Sep 23 00:45:50 raspberrypi avahi-daemon[297]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 192.168.178.163.
Sep 23 00:45:50 raspberrypi avahi-daemon[297]: Interface wlan0.IPv4 no longer relevant for mDNS.
Sep 23 00:45:50 raspberrypi dhcpcd[475]: wlan0: deleting route to 192.168.178.0/24
Sep 23 00:45:50 raspberrypi dhcpcd[475]: wlan0: deleting default route via 192.168.178.1
Sep 23 00:45:55 raspberrypi dhcpcd[475]: wlan0: carrier acquired
Sep 23 00:45:55 raspberrypi dhcpcd[475]: wlan0: IAID eb:3e:1a:e0
Sep 23 00:45:55 raspberrypi dhcpcd[475]: wlan0: rebinding lease of 192.168.178.163
Sep 23 00:45:55 raspberrypi dhcpcd[475]: wlan0: probing address 192.168.178.163/24
Sep 23 00:45:56 raspberrypi dhcpcd[475]: wlan0: soliciting an IPv6 router
Sep 23 00:45:56 raspberrypi dhcpcd[475]: wlan0: Router Advertisement from fe80::3a10:d5ff:fe60:4a2a
Sep 23 00:45:56 raspberrypi dhcpcd[475]: wlan0: adding address 2003:e0:7f1f:8500:35d3:4c67:a30b:2ce8/64
Sep 23 00:45:56 raspberrypi dhcpcd[475]: wlan0: adding route to 2003:e0:7f1f:8500::/64
Sep 23 00:45:56 raspberrypi avahi-daemon[297]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::ef45:8c3a:42af:1e8d.
Sep 23 00:45:56 raspberrypi dhcpcd[475]: wlan0: adding default route via fe80::3a10:d5ff:fe60:4a2a
Sep 23 00:45:56 raspberrypi avahi-daemon[297]: Joining mDNS multicast group on interface wlan0.IPv6 with address 2003:e0:7f1f:8500:35d3:4c67:a30$
Sep 23 00:45:56 raspberrypi avahi-daemon[297]: Registering new address record for 2003:e0:7f1f:8500:35d3:4c67:a30b:2ce8 on wlan0.*.
Sep 23 00:45:56 raspberrypi dhcpcd[475]: wlan0: soliciting a DHCPv6 lease
Sep 23 00:45:56 raspberrypi avahi-daemon[297]: Withdrawing address record for fe80::ef45:8c3a:42af:1e8d on wlan0.
Sep 23 00:45:57 raspberrypi dhcpcd[475]: wlan0: DHCPv6 REPLY: in stateless mode
Sep 23 00:45:58 raspberrypi dhcpcd[475]: wlan0: DHCPv6 REPLY: in stateless mode
Sep 23 00:46:00 raspberrypi dhcpcd[475]: wlan0: DHCPv6 REPLY: in stateless mode
Sep 23 00:46:01 raspberrypi dhcpcd[475]: wlan0: leased 192.168.178.163 for 864000 seconds
Sep 23 00:46:01 raspberrypi avahi-daemon[297]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.178.163.
Sep 23 00:46:01 raspberrypi avahi-daemon[297]: New relevant interface wlan0.IPv4 for mDNS.
Sep 23 00:46:01 raspberrypi dhcpcd[475]: wlan0: adding route to 192.168.178.0/24
Sep 23 00:46:01 raspberrypi dhcpcd[475]: wlan0: adding default route via 192.168.178.1
Sep 23 00:46:01 raspberrypi avahi-daemon[297]: Registering new address record for 192.168.178.163 on wlan0.IPv4.
Sep 23 00:46:04 raspberrypi dhcpcd[475]: wlan0: DHCPv6 REPLY: in stateless mode
Sep 23 00:46:11 raspberrypi dhcpcd[475]: wlan0: DHCPv6 REPLY: in stateless mode
Sep 23 00:46:53 raspberrypi kernel: [190079.012984] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted
Sep 23 00:47:00 raspberrypi kernel: [190086.579355] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Sep 23 00:47:00 raspberrypi kernel: [190086.579566] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Sep 23 00:47:00 raspberrypi kernel: [190086.579575] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
Sep 23 00:47:17 raspberrypi kernel: [190103.139510] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Sep 23 00:47:17 raspberrypi kernel: [190103.139998] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Sep 23 00:47:17 raspberrypi kernel: [190103.140014] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
Sep 23 00:47:19 raspberrypi kernel: [190105.699521] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Sep 23 00:47:19 raspberrypi kernel: [190105.699962] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Sep 23 00:47:19 raspberrypi kernel: [190105.699979] brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
Sep 23 00:47:22 raspberrypi kernel: [190108.259470] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Sep 23 00:47:22 raspberrypi kernel: [190108.259690] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Sep 23 00:49:47 raspberrypi dhcpcd[475]: wlan0: fe80::3a10:d5ff:fe60:4a2a is unreachable, expiring it
...
Sep 23 00:50:04 raspberrypi kernel: [190270.020426] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sep 23 00:50:04 raspberrypi kernel: [190270.020459] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
Sep 23 00:50:06 raspberrypi kernel: [190272.580445] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Sep 23 00:50:06 raspberrypi kernel: [190272.580462] brcmfmac: brcmf_cfg80211_get_tx_power: error (-110)
```

lategoodbye, 2019-10-01:
The message "mailbox indicates firmware halted" said that the Wifi firmware has crashed.

Please provide more information:
- Raspberry Pi model
- Wifi firmware version (dmesg)
- Model version of your Fritzbox
- Raspberry Pi Wifi configuration (without password ;-) )
- Does this issue always happend after a few hours or does it sometimes takes longer?
- Are there any other wifi clients which could trigger the crash?
- Are there any additional network services on the Raspberry Pi which could trigger the crash?

sub0ne, 2019-10-07:
- Raspberry Pi model
3b+

- Wifi firmware version (dmesg)
will post later

- Model version of your Fritzbox
Fritzbox 7490, latest OS Version 7.12

- Raspberry Pi Wifi configuration (without password ;-) )
will post later

- Are you able to trigger this issue by moving the Raspberry Pi out of reach and back again?
didn't try this yet. Maybe I can also try to turn my routers wifi off and back on.

- Are there any additional network services on the Raspberry Pi which could trigger the crash?
it's running iobroker, ha-bridge and deconz but the error already occured before setting up my home automation.

JamesH65, 2019-10-18:
@sub0ne Any answers to the remaining questions? Or has the problem fixed itself?

sub0ne, 2020-02-11:
I now did an apt-get update/full-upgrade and some brcmf files were updated. Hoping the problem would be fixed but it is not. After 2 weeks of flawless connection wifi connection dropped again and I am not able to reach the pi. This think is indeed pretty instable...

JamesH65, 2020-02-14:
> I now did an apt-get update/full-upgrade and some brcmf files were updated. Hoping the problem would be fixed but it is not. After 2 weeks of flawless connection wifi connection dropped again and I am not able to reach the pi. This think is indeed pretty instable...

Once a day is unstable, once every two weeks is pretty good in my experience! Anyway, do you have any dmesg reports at the point of failure?

sub0ne, 2020-02-14:
once every two weeks is good? I'm using my pi as productive home automation and I have to say it sucks when you wanna go to bed and the light is not turning off. 

Regarding the log...is this different somehow from my first syslog post? it's always repeating the same line:

```
Sep 23 00:46:11 raspberrypi dhcpcd[475]: wlan0: DHCPv6 REPLY: in stateless mode
Sep 23 00:46:53 raspberrypi kernel: [190079.012984] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted
Sep 23 00:47:00 raspberrypi kernel: [190086.579355] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Sep 23 00:47:00 raspberrypi kernel: [190086.579566] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Sep 23 00:47:00 raspberrypi kernel: [190086.579575] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
```

JamesH65, 2020-02-14:
Not good overall, just in my experience with wireless adapters etc.

It is a problem with the wireless firmware. Which we don't have access to so will need to be reported to Cypress who provide the chip. They are not a fast responding company. Not the only issue IIRC that also involves the Fritzbox , so probably a combination that causes the problem. What that combo is is anyone's guess.

Chetan2312, 2023-07-12:
I am using raspberry pi 4b+ 4gb model. I am facing same issues. The dhcp service keeps on deleting the assigned ip and avahi-daemon restarting the network service again and again. I tried turning off the power_save mode using:
sudo iw wlan0 set power_saave off
But still facing the issues with connection to wifi. Unable to connect to wifi.

loomsen, 2024-04-28:
I am still facing this issue with a raspi 5 8GB model.

peterharperuk, 2024-04-29:
If you think you have wifi firmware issues and you want to help debug them, you could add the following to the end of the kernel command line brcmfmac.debug=0x100400

You do this by by adding that to the end of the first line in /boot/firmware/cmdline.txt or /boot/cmdline.txt. Then reboot, reproduce the problem and attach the output of dmesg as a file. Thanks.

loomsen, 2024-04-30:
Hi @peterharperuk 

Sure, I enabled the debug and this is the outcome:

```
Mar 15 16:09:07 raspberrypi kernel: brcmfmac: F1 signature read @0x18000000=0x15264345
Mar 15 16:09:07 raspberrypi kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Mar 15 16:09:07 raspberrypi kernel: usbcore: registered new interface driver brcmfmac
Mar 15 16:09:07 raspberrypi kernel: brcmfmac_wcc: brcmf_wcc_attach: executing
Mar 15 16:09:07 raspberrypi kernel: brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
Mar 15 16:09:07 raspberrypi kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2
Apr 26 14:08:15 raspberrypi NetworkManager[748]: <info>  [1714133295.2259] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/axi/1001100000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
Apr 26 14:08:15 raspberrypi kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Apr 26 14:08:22 raspberrypi kernel: brcmfmac: F1 signature read @0x18000000=0x15264345
Apr 26 14:08:22 raspberrypi kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Apr 26 14:08:22 raspberrypi kernel: usbcore: registered new interface driver brcmfmac
Apr 26 14:08:23 raspberrypi kernel: brcmfmac_wcc: brcmf_wcc_attach: executing
Apr 26 14:08:23 raspberrypi kernel: brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
Apr 26 14:08:23 raspberrypi kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2
Apr 26 14:08:23 raspberrypi NetworkManager[771]: <info>  [1714133303.6859] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/axi/1001100000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
Apr 26 14:08:23 raspberrypi kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Apr 26 14:13:03 raspberrypi kernel: brcmfmac: F1 signature read @0x18000000=0x15264345
Apr 26 14:13:03 raspberrypi kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Apr 26 14:13:04 raspberrypi kernel: usbcore: registered new interface driver brcmfmac
Apr 26 14:13:04 raspberrypi kernel: brcmfmac_wcc: brcmf_wcc_attach: executing
Apr 26 14:13:04 raspberrypi kernel: brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
Apr 26 14:13:04 raspberrypi kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2
Apr 26 14:13:05 raspberrypi NetworkManager[748]: <info>  [1714133585.4291] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/axi/1001100000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
Apr 26 14:13:05 raspberrypi kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Apr 26 14:12:46 raspberrypi kernel: brcmfmac: F1 signature read @0x18000000=0x15264345
Apr 26 14:12:46 raspberrypi kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Apr 26 14:12:46 raspberrypi kernel: usbcore: registered new interface driver brcmfmac
Apr 26 14:12:47 raspberrypi kernel: brcmfmac_wcc: brcmf_wcc_attach: executing
Apr 26 14:12:47 raspberrypi kernel: brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
Apr 26 14:12:47 raspberrypi kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2
Apr 26 14:14:07 raspberrypi NetworkManager[743]: <info>  [1714133647.3445] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/axi/1001100000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
Apr 26 14:14:07 raspberrypi kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Apr 26 15:13:14 raspberrypi kernel: brcmfmac: F1 signature read @0x18000000=0x15264345
Apr 26 15:13:14 raspberrypi kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Apr 26 15:13:14 raspberrypi kernel: usbcore: registered new interface driver brcmfmac
Apr 26 15:13:15 raspberrypi kernel: brcmfmac_wcc: brcmf_wcc_attach: executing
Apr 26 15:13:15 raspberrypi kernel: brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
Apr 26 15:13:15 raspberrypi kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2
Apr 26 15:13:15 raspberrypi NetworkManager[733]: <info>  [1714137195.8032] rfkill1: found Wi-Fi radio killswitch (at /sys/devices/platform/axi/1001100000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1/ieee80211/phy0/rfkill1) (driver brcmfmac)
Apr 26 15:13:15 raspberrypi kernel: brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Apr 27 18:16:41 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd022 fail, reason -52
Apr 27 18:16:42 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd026 fail, reason -52
Apr 27 18:16:42 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd02a fail, reason -52
Apr 27 18:16:42 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd02e fail, reason -52
Apr 27 18:16:43 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd090 fail, reason -52
Apr 27 18:16:43 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd095 fail, reason -52
Apr 27 18:16:43 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd099 fail, reason -52
Apr 27 18:16:43 raspberrypi kernel: brcmfmac: brcmf_set_channel: set chanspec 0xd09d fail, reason -52
```

The last line is repeated for like a thousand times. Does this help?
I'll also attach the full log.

[log.txt](https://github.com/raspberrypi/linux/files/15167897/log.txt)

peterharperuk, 2024-05-01:
@loomsen I don't see any firmware crashes in there. You might want to raise a new issue if you think you have a problem. Are you having trouble using 5G Wifi? What is the history of this device with the version of software installed?

loomsen, 2024-05-01:
@peterharperuk The issue is, that the device loses its connection after some time, and doesn't reconnect to the wifi. Only a restart helps, I restarted wpa_supplicant multiple times, I tried configuring the wifi info in the wpa_supplicant.conf, I re-installed Raspberry Pi OS lite multiple times, all with the same outcome. After a while, WiFi connection is lost and only a reboot helps to re-establish it. I went on and connected the PI with a cable, as I need it to be reliable. Don't know what causes it, but it seems to be a known issue for some raspi releases, judging by my google searches.

contracode, 2024-05-06:
I think that this is an issue with `NetworkManager` and its interaction with `wpa_supplicant`. The reason I believe this to be the case is that when I switch to `systemd-networkd` and `wpa_supplicant` for **wlan0** configuration, the WiFi connection is robust. (Of interest, I had a _very_ similar issue, using _**Arch Linux**_ and `netctl`, alongside this issue with _**Raspberry Pi OS Lite (bookworm)**_).

My system is running **Debian GNU/Linux 12 (bookworm)** on kernel **6.6.28+rpt-rpi-v8**.

**The fix:**
1. Create a `/etc/systemd/network/wlan0.network` file with the below configuration, uncommenting the appropriate block in the `[Network]` section.
```
[Match]
Name=wlan0

[Network]
# Uncomment one of the blocks, below, to set up a static or dynamic IP address.

# The configuration below is for a static IP address.
#Address=192.168.1.100/24    # Change to the IP address you have reserved in your DHCP server.
#Gateway=192.168.1.1
#DNS=192.168.1.1

# The configuration below is for a dynamic IP address.
#DHCP=yes
#DNSSEC=no

[DHCP]
ClientIdentifier=mac
RouteMetric=50
```
2. Create a `/etc/wpa_supplicant/wpa_supplicant-wlan0.conf`, populating this with:
```
ctrl_interface=/run/wpa_supplicant
update_config=1
```
3. Start `wpa_supplicant` with `wpa_supplicant -B -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant-wlan0.conf`.
4. Enter the `wpa_supplicant` interactive prompt with `wpa_cli`.
5. Use the `scan` and `scan_results` commands to see the available networks.
```
> scan
OK
<3>CTRL-EVENT-SCAN-RESULTS
> scan_results
bssid / frequency / signal level / flags / ssid
00:00:00:00:00:00 2462 -49 [WPA2-PSK-CCMP][ESS] MYSSID
11:11:11:11:11:11 2437 -64 [WPA2-PSK-CCMP][ESS] ANOTHERSSID
```
6. To associate with _**MYSSID**_, add the network, and set the credentials and enable it
 ```
> add_network
0
> set_network 0 ssid "MYSSID"
> set_network 0 psk "passphrase"
> enable_network 0
<2>CTRL-EVENT-CONNECTED - Connection to 00:00:00:00:00:00 completed (reauth) [id=0 id_str=]
```
7. Finally save this network in the configuration file and quit wpa_cli:
```
> save_config
OK
> quit
```
a complete `etc/wpa_supplicant/wpa_supplicant-wlan0.conf` file should look similar to this:
```
ctrl_interface=/var/run/wpa_supplicant
update_config=1
country=US

network={
	ssid="MYSSID"
	psk=02f939c889a892c21ccebbca6d6b29b6754c16bf9533ea096bd17fa2d105
	mesh_fwding=1
}
```
8. Install `resolved` with `sudo apt install libnss-resolve`.
9. Disable `NetworkManager` with `sudo systemctl disable NetworkManager.service`.
10. Enable `systemd-networkd` and `systemd-resolved` with `sudo systemctl enable systemd-networkd.service systemd-resolved.service`.
11. Enable the WPA-supplicant service with `sudo systemctl enable wpa_supplicant@wlan0.service`.
12. Reboot with `sudo reboot`.

**Sources:**
1. https://wiki.archlinux.org/title/systemd-networkd
2. https://wiki.archlinux.org/title/wpa_supplicant
3. https://wiki.archlinux.org/title/Network_configuration#Network_managers
4. https://docs.pikvm.org/wifi/#setting-up-wi-fi-manually