raspberrypi/linux #3260: RPi 4: Bluetooth serdev driver failed to probe with Mainline devicetree

lategoodbye, 2019-10-01:
**Describe the bug**
I'm trying to get Bluetooth running on the Raspberry Pi 4 with Linux 5.4-rc1 and Upstream DTS. Currently the Bluetooth part is an exact copy of the Raspberry Pi 3 B+ (bcm2837-rpi-3-b-plus.dts):

```
&uart0 {
	pinctrl-names = "default";
	pinctrl-0 = <&uart0_ctsrts_gpio30 &uart0_gpio32 &gpclk2_gpio43>;
	status = "okay";

	bluetooth {
		compatible = "brcm,bcm43438-bt";
		max-speed = <2000000>;
		shutdown-gpios = <&expgpio 0 GPIO_ACTIVE_HIGH>;
	};
};
```

Unfortunately Bluetooth serdev driver doesn't probe as expected. Since the Bluetooth serdev driver probe Raspberry Pi 3 B+ with the same kernel i think we can rule out a kernel issue. My suspicion is that the Device Tree part doesn't fit to the real hardware. The fact that according to dt-blob.dts the GPIO 43 is used as chip select for the EEPROM and not as GPIO clock seems to confirm my concerns. Also i cannot remember that Bluetooth ever worked with 5.3,

**Expected behaviour**
Bluetooth serdev driver probe without any issues

**Actual behaviour**
Bluetooth serdev driver fails while setting baudrate:
```
[    4.199331] Bluetooth: HCI device and connection manager initialized
[    4.199344] Bluetooth: HCI socket layer initialized
[    4.199351] Bluetooth: L2CAP socket layer initialized
[    4.199361] Bluetooth: SCO socket layer initialized
[    4.229986] Bluetooth: HCI UART driver ver 2.3
[    4.229994] Bluetooth: HCI UART protocol H4 registered
[    4.230242] Bluetooth: HCI UART protocol Broadcom registered
[    4.231686] hci_uart_bcm serial0-0: serial0-0 supply vbat not found, using dummy regulator
[    4.231734] hci_uart_bcm serial0-0: serial0-0 supply vddio not found, using dummy regulator
[    4.347908] uart-pl011 fe201000.serial: no DMA platform data
[    6.477633] Bluetooth: hci0: command 0xfc18 tx timeout
[   14.557631] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   14.557640] Bluetooth: hci0: Failed to set baudrate
[   16.637638] Bluetooth: hci0: command 0x0c03 tx timeout
[   24.797634] Bluetooth: hci0: BCM: Reset failed (-110)
[   26.877634] Bluetooth: hci0: command 0xfc18 tx timeout
[   35.037648] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   35.037657] Bluetooth: hci0: Failed to set baudrate
[   37.117626] Bluetooth: hci0: command 0x0c03 tx timeout
[   45.277644] Bluetooth: hci0: BCM: Reset failed (-110)
```

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Raspberry Pi 4
* Which OS and version (`cat /etc/rpi-issue`)? Raspbian Buster
* Which kernel version (`uname -a`)? Linux 5.4-rc1

lategoodbye, 2019-10-01:
@vianpl Did you ever see Bluetooth probing with a Mainline DTS?

vianpl, 2019-10-02:
@lategoodbye No, I actually always saw the error.

lategoodbye, 2019-10-02:
@pelwell Any ideas?

pelwell, 2019-10-02:
I've not played with the upstream BT support, but the compatible string is at least inaccurate - it's a bcm43455 (or cyw43455) - but that may be the best you can do.

It could be a sequencing issue between the BT_ON and WL_ON signals. Also make sure that the brcmfmac43455-sdio.txt contains `boardflags3=0x44200100` (_not_ `boardflags3=0x48200100` as it used to be).

lategoodbye, 2019-10-02:
I will check the boardflags in the NVRAM configuration.

Is GPIO 43 still provider of the lower power clock?
Are the GPIO 30,31 actually used as hardware handshake?
Can you point me the used baud rate by hciattach?

pelwell, 2019-10-02:
On Pi4 the LPO clock is internally generated by the 43455, as indicated by the boardflags3 change.

The [btuart script](https://github.com/RPi-Distro/pi-bluetooth/blob/master/usr/bin/btuart) runs hciattach with the required parameters.

> Note: Regarding the compatible your comment is correct, but it doesn't make sense to maintain a useless device id list on a "intelligent" bus.

Are you saying that the 43455 identifies itself over the UART using some common configuration mechanism?

lategoodbye, 2019-10-02:
Thanks

> Are you saying that the 43455 identifies itself over the UART using some common configuration mechanism?

AFAIK the Bluetooth chip provides some kind of ID via UART and the driver translate this accordingly `[    8.209258] Bluetooth: hci0: BCM4345C0`

The DT compatible is only a place holder to probe the right driver.

pelwell, 2019-10-02:
> `[8.209258] Bluetooth: hci0: BCM4345C0`
That comes from the HCI layer, which only works (properly) after the BT modem firmware is "patched", which requires knowing what the device is. I don't think you can call a UART an intelligent bus, but if you have some idea of the kind of device at the other end then you may be able to correctly identify the remote device using some device-specific handshake.

> Are the GPIO 30,31 actually used as hardware handshake?

Yes.

Trust the content of dt-blob.dts.

lategoodbye, 2019-10-02:
Okay, i checked and got `boardflags3=0x44200100`. So this is okay.

I tried the following steps without any improvement:
- reduce pinctrl to GPIO 32,32
- swap reset polarity
- reduce baudrate to 115200

lategoodbye, 2019-10-03:
I attached my cmdline and relevant config.txt settings to the initial post.

Dark-Sky, 2019-10-04:
I am assuming you you compiled the Linux 5.4-rc1 upstream kernel using it's arm/multi_v7_defconfig but I am not clear on what .dtb you are using to boot with when you say Upstream DTS. I would like to mess with this if I could.

lategoodbye, 2019-10-04:
@Dark-Sky Thanks for trying to help. There is no accepted DTS in Linux 5.4-rc1 yet. But you can find my current WIP here:
https://github.com/lategoodbye/rpi-zero/commits/bcm2711-initial-v4

Be aware this is frequently rebased.

lategoodbye, 2019-10-04:
Okay, at least i'm able to upload the Bluetooth firmware manually:

`hciattach /dev/ttyAMA0 bcm43xx 115200 flow -`

with BT_ON = 1. So at least the polarity seems to be wrong.

Dark-Sky, 2019-10-04:
Well you blew me out of the water lategoodbye; I was going to try something similar on the 5.4 kernel. lol

On arch-arm64 I have been having to use brcm-patchram-plus which I made a systemd .service file with the pi3/pi4. Something in the raspian image is different for some reason.

Here is a snip in the .service file that started the command if it helps in any small way:

`ExecStart=/usr/bin/brcm_patchram_plus --patchram /lib/firmware/updates/brcm/BCM4345C0.hcd --baudrate 3000000 --enable_hci --use_baudrate_for_download --no2bytes --tosleep=1000000 /dev/ttyAMA0`

lategoodbye, 2019-10-05:
Okay, i think i've found the issue. The hci_bcm doesn't behave like the suggested init sequences by Broadcom/Cypress in the CYW43455 datasheet (Fig. 7).

Expected sequence:
1. Enable BT_ON
2. wait a little bit
3. Enable RTS

Current hci_bcm:
1. Enable RTS
2. Enable BT_ON
3. wait a little bit

Looks like the BT firmware is triggering on the RTS edge.

lategoodbye, 2019-10-05:
Here is the proposed fix:
https://github.com/lategoodbye/rpi-zero/commit/f7ad01a69fd1e319354912f926b9accd59757b24

closed 2019-10-05 by lategoodbye