raspberrypi/linux #3267: Pi4 - Kernel 5.3.x - Setting/change of screen refresh rate fails inside Kodi

mkreisl, 2019-10-04:
The setting or change of screen refresh rate fails inside Kodi if Raspberry Pi4 is running kernel 5.3.x (as time of writing 5.3.3).
A short flickering of screen is noticeable, but rate is **not** changed. Don't see any ERROR-log in the log of Kodi or syslog/dmesg
Works on kernel 4.19.x though.

Running same kernel on Pi3B+ hardware,  changing screen refresh rate is ok there, so being unsure if issue depends on Kodi or kernel. 

FXI: Posted same issue [in Kodi forum](https://forum.kodi.tv/showthread.php?tid=347991)

6by9, 2019-10-04:
5.3 is still a work in progress with v3d - #3162

mkreisl, 2019-10-04:
That's what I was afraid of.

And even at 4.19, the v3d driver seems to have serious problems using BTRFS with compression enabled

But in this regard I will open a new issue later

mkreisl, 2019-12-02:
5.4.1 is working now, so closing issue

closed 2019-12-02 by mkreisl