raspberrypi/linux #3269: rpi4: bluetoothctl finds no controller although "hcitool scan" works

elmicha, 2019-10-05:
On a freshly installed and updated Buster (kernel 4.19.57-v7l+) on Raspberry Pi 4B `bluetoothctl` hangs, and after using ctrl-c and entering `devices`, it says "No default controller available". 

`hciconfig` shows normal output with "UP RUNNING", `hcitool scan` finds some devices, `hciconfig features` shows the same output as on my Pi3B+. `dmesg` doesn't show any Bluetooth-related errors.

JamesH65, 2019-10-18:
@elmicha What were you doing in bluetoothctl to make it hang? Just tried it on a  Pi4 here and starts up and the devices command gives no errors.

JamesH65, 2019-10-18:
Note, I am using a more up to date kernel, 4.19.79.

elmicha, 2019-10-18:
I only try to start it from Bash. It hangs before it shows its first prompt. If I press ctrl-c then, it shows the [bluetooth] prompt. `list` shows nothing. `show` says `No default controller available`.

I used rpi-update to upgrade to kernel 4.19.79-v7l+ and alas it is the same as before.

JamesH65, 2019-10-19:
Odd, since it works fine for me. I'll have to try again on Monday. ANything strange plugged in to your system?

elmicha, 2019-10-20:
No, just an USB3 enclosure with an SSD (with /boot on an SD card).

But now I tried it with a fresh system on another SD card, and bluetoothctl works without problems. I also used `apt update && apt upgrade` and it still works. So I broke my original system somehow.

I will try to figure out what the problem is, but I guess I should close this ticket.

Thank you for your help and for your work with the Raspberry Pi Foundation!

closed 2019-10-20 by elmicha