raspberrypi/linux #3273: Drive sdio wifi module on uSD slot

pengguang001, 2019-10-09:
From this link [https://github.com/raspberrypi/linux/issues/2003](url) it should be possible to run sdio wifi module on uSD slot. My steps:

1. boot system from usb stick instead of tf-card.
2. patch sdio-overlay.dts to config gpio48-53 as sdio usage and update sdio.dtbo to device.
```
@@ -66,6 +66,15 @@
                };
        };
 
+       fragment@7 {
+               target = <&sdio_ovl_pins>;
+               __dormant__ {
+                       brcm,pins = <48 49 50 51 52 53>;
+                       brcm,function = <7>; /* ALT3 = SD1 */
+                       brcm,pull = <0 2 2 2 2 2>;
+               };
+       };
+
        __overrides__ {
                poll_once = <&sdio_ovl>,"non-removable?";
                bus_width = <&sdio_ovl>,"bus-width:0";
@@ -73,5 +82,6 @@
                gpios_22_25 = <0>,"=3";
                gpios_34_37 = <0>,"=4";
                gpios_34_39 = <0>,"=5";
+               gpios_48_53 = <0>,"=7";
        };
 };
```
3. modify /boot/config.txt
```
[sdio]
dtoverlay=sdio,gpios_48_53,poll_once=off
```

Unfortunately it always fail to enumerate sdio device. Kernel log:

```
[    1.942099] mmc-bcm2835 3f300000.mmc: mmc_debug:0 mmc_debug2:0
[    1.950471] mmc-bcm2835 3f300000.mmc: DMA channel allocated
[    1.984310] pinctrl-bcm2835 3f200000.gpio: pin gpio48 already requested by 3f300000.mmc; cannot claim for 3f202000.mmc
[    2.000114] pinctrl-bcm2835 3f200000.gpio: pin-48 (3f202000.mmc) status -22
[    2.009757] pinctrl-bcm2835 3f200000.gpio: could not request pin 48 (gpio48) from group gpio48  on device pinctrl-bcm2835
[    2.025944] sdhost-bcm2835 3f202000.mmc: Error applying setting, reverse things back
[    2.039148] sdhost-bcm2835: probe of 3f202000.mmc failed with error -22
[    2.048883] Indeed it is in host mode hprt0 = 00021501
[    2.049988] of_cfs_init
[    2.062439] of_cfs_init: OK
[    2.068639] Waiting for root device PARTUUID=569449d1-02...
[    2.143520] random: fast init done
[    2.261725] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    2.271111] Indeed it is in host mode hprt0 = 00001101
[    2.511948] usb 1-1: New USB device found, idVendor=0424, idProduct=2514, bcdDevice= b.b3
[    2.525716] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.536448] hub 1-1:1.0: USB hub found
[    2.543183] hub 1-1:1.0: 4 ports detected
[    2.861695] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    2.991944] usb 1-1.1: New USB device found, idVendor=0424, idProduct=2514, bcdDevice= b.b3
[    3.005917] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.016781] hub 1-1.1:1.0: USB hub found
[    3.023646] hub 1-1.1:1.0: 3 ports detected
[    3.121693] usb 1-1.3: new high-speed USB device number 4 using dwc_otg
[    3.265080] usb 1-1.3: New USB device found, idVendor=05e3, idProduct=0751, bcdDevice=14.04
[    3.278792] usb 1-1.3: New USB device strings: Mfr=3, Product=4, SerialNumber=0
[    3.289001] usb 1-1.3: Product: USB Storage
[    3.296066] usb 1-1.3: Manufacturer: USB Storage
[    3.304540] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[    3.314380] scsi host0: usb-storage 1-1.3:1.0
[    3.475810] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    3.485767] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    3.495584] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    3.506502] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[    3.569944] mmc1: error -110 whilst initialising SDIO card
[    3.624529] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    3.634530] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    3.644421] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    3.655830] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[    3.735322] mmc1: error -110 whilst initialising SDIO card
```

Can you help me? @pelwell

pelwell, 2019-10-09:
This line is very telling:
```
[    1.984310] pinctrl-bcm2835 3f200000.gpio: pin gpio48 already requested by 3f300000.mmc; cannot claim for 3f202000.mmc
```
Both SD interfaces are claiming GPIOs 48-53, which is never going to work. `dtoverlay=mmc` enables SD1 on 48-54 _and_ disables SD0.

pengguang001, 2019-10-10:
Thanks pelwell! It's good idea to overlay mmc. After a look of mmc-overlay.dts, it disables mmcnr (which driver onboard wifi) and enable mmc and use gpio48-53.

Unfortunately still meet -110 error. Here's the log:

```
[    0.828563] mmc-bcm2835 3f300000.mmc: /aliases ID not available
[    0.831149] mmc-bcm2835 3f300000.mmc: could not get clk, deferring probe
[    0.833907] sdhci-pltfm: SDHCI platform and OF driver helper
[    0.837952] ledtrig-cpu: registered to indicate activity on CPUs
[    0.840621] hidraw: raw HID events driver (C) Jiri Kosina
[    0.843366] usbcore: registered new interface driver usbhid
[    0.845939] usbhid: USB HID core driver
[    0.849192] vchiq: vchiq_init_state: slot_zero = (ptrval), is_master = 0
[    0.853225] [vc_sm_connected_init]: start
[    0.861988] [vc_sm_connected_init]: end - returning 0
[    0.865460] Initializing XFRM netlink socket
[    0.867819] NET: Registered protocol family 17
[    0.870258] Key type dns_resolver registered
[    0.873034] Registering SWP/SWPB emulation handler
[    0.876077] registered taskstats version 1
[    0.878418] Loading compiled-in X.509 certificates
[    0.887912] uart-pl011 3f201000.serial: cts_event_workaround enabled
[    0.890372] 3f201000.serial: ttyAMA0 at MMIO 0x3f201000 (irq = 81, base_baud = 0) is a PL011 rev2
[    0.896712] console [ttyS0] disabled
[    0.899086] 3f215040.serial: ttyS0 at MMIO 0x0 (irq = 53, base_baud = 31250000) is a 16550
[    1.931460] console [ttyS0] enabled
[    1.938382] mmc-bcm2835 3f300000.mmc: /aliases ID not available
[    1.947080] mmc-bcm2835 3f300000.mmc: mmc_debug:0 mmc_debug2:0
[    1.955407] mmc-bcm2835 3f300000.mmc: DMA channel allocated
[    1.990190] of_cfs_init
[    1.995141] of_cfs_init: OK
[    2.000769] Waiting for root device PARTUUID=569449d1-02...
[    2.041790] Indeed it is in host mode hprt0 = 00021501
[    2.134198] random: fast init done
[    2.251698] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    2.260732] Indeed it is in host mode hprt0 = 00001101
[    2.501949] usb 1-1: New USB device found, idVendor=0424, idProduct=2514, bcdDevice= b.b3
[    2.515137] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.525623] hub 1-1:1.0: USB hub found
[    2.532074] hub 1-1:1.0: 4 ports detected
[    2.851697] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    2.981942] usb 1-1.1: New USB device found, idVendor=0424, idProduct=2514, bcdDevice= b.b3
[    2.995373] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.005891] hub 1-1.1:1.0: USB hub found
[    3.012508] hub 1-1.1:1.0: 3 ports detected
[    3.111697] usb 1-1.3: new high-speed USB device number 4 using dwc_otg
[    3.245076] usb 1-1.3: New USB device found, idVendor=05e3, idProduct=0751, bcdDevice=14.04
[    3.258794] usb 1-1.3: New USB device strings: Mfr=3, Product=4, SerialNumber=0
[    3.269094] usb 1-1.3: Product: USB Storage
[    3.276148] usb 1-1.3: Manufacturer: USB Storage
[    3.284609] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[    3.294423] scsi host0: usb-storage 1-1.3:1.0
[    3.392800] mmc2: queuing unknown CIS tuple 0x80 (2 bytes)
[    3.402739] mmc2: queuing unknown CIS tuple 0x80 (3 bytes)
[    3.412538] mmc2: queuing unknown CIS tuple 0x80 (3 bytes)
[    3.423440] mmc2: queuing unknown CIS tuple 0x80 (7 bytes)
[    3.434830] mmc2: queuing unknown CIS tuple 0x81 (9 bytes)
[    3.528852] mmc2: error -110 whilst initialising SDIO card
[    3.583374] mmc2: queuing unknown CIS tuple 0x80 (2 bytes)
[    3.593256] mmc2: queuing unknown CIS tuple 0x80 (3 bytes)
[    3.603026] mmc2: queuing unknown CIS tuple 0x80 (3 bytes)
[    3.614271] mmc2: queuing unknown CIS tuple 0x80 (7 bytes)
[    3.626209] mmc2: queuing unknown CIS tuple 0x81 (9 bytes)
[    3.744978] mmc2: error -110 whilst initialising SDIO card

```

I have dumped the device dts tree, it seems good. Pin config is good and mmcnr is disabled.
```

                mmc@7e300000 {
                        compatible = "brcm,bcm2835-mmc\0brcm,bcm2835-sdhci";
                        clocks = < 0x03 0x1c >;
                        status = "okay";
                        interrupts = < 0x02 0x1e >;
                        brcm,overclock-50 = < 0x00 >;
                        bus-width = < 0x04 >;
                        dma-names = "rx-tx";
                        phandle = < 0x2b >;
                        reg = < 0x7e300000 0x100 >;
                        pinctrl-0 = < 0x7e >;
                        dmas = < 0x0a 0x0b >;
                        pinctrl-names = "default";
                };

                        mmc_pins {
                                brcm,pins = < 0x30 0x31 0x32 0x33 0x34 0x35 >;
                                phandle = < 0x7e >;
                                brcm,pull = < 0x00 0x02 0x02 0x02 0x02 0x02 >;
                                brcm,function = < 0x07 >;
                        };

                mmcnr@7e300000 {
                        compatible = "brcm,bcm2835-mmc\0brcm,bcm2835-sdhci";
                        clocks = < 0x03 0x1c >;
                        status = "disabled";
                        interrupts = < 0x02 0x1e >;
                        brcm,overclock-50 = < 0x00 >;
                        bus-width = < 0x04 >;
                        dma-names = "rx-tx";
                        phandle = < 0x2c >;
                        reg = < 0x7e300000 0x100 >;
                        pinctrl-0 = < 0x1a >;
                        dmas = < 0x0a 0x0b >;
                        non-removable;
                        pinctrl-names = "default";
                };

                        sdio_pins {
                                brcm,pins = < 0x22 0x23 0x24 0x25 0x26 0x27 >;
                                phandle = < 0x1a >;
                                brcm,pull = < 0x00 0x02 0x02 0x02 0x02 0x02 >;
                                brcm,function = < 0x07 >;
                        };
```

pengguang001, 2019-10-10:
When use on-board wifi, ios shows a good clock:

```
pi@raspberrypi:~ $ sudo cat /sys/kernel/debug/mmc1/ios
clock:		50000000 Hz
actual clock:	50000000 Hz
vdd:		21 (3.3 ~ 3.4 V)
bus mode:	2 (push-pull)
chip select:	0 (don't care)
power mode:	2 (on)
bus width:	2 (4 bits)
timing spec:	2 (sd high-speed)
signal voltage:	0 (3.30 V)
driver type:	0 (driver type B)
```

If overlay mmc and plug in wifi module on uSD slot, ios is incorrect (it's on mmc2)

```
pi@raspberrypi:~ $ sudo cat /sys/kernel/debug/mmc2/ios
clock:		0 Hz
vdd:		0 (invalid)
bus mode:	2 (push-pull)
chip select:	0 (don't care)
power mode:	0 (off)
bus width:	0 (1 bits)
timing spec:	0 (legacy)
signal voltage:	0 (3.30 V)
driver type:	0 (driver type B)
```

pengguang001, 2019-10-10:
Since mmc is doing continuous card enumeration, sometimes I found ios shows 100K Hz clock. This should be a reasonable clock during enumeration.

```
pi@raspberrypi:/sys/class/mmc_host/mmc2/power $ sudo cat /sys/kernel/debug/mmc2/ios
clock:		100000 Hz
actual clock:	100000 Hz
vdd:		21 (3.3 ~ 3.4 V)
bus mode:	2 (push-pull)
chip select:	0 (don't care)
power mode:	2 (on)
bus width:	0 (1 bits)
timing spec:	0 (legacy)
signal voltage:	0 (3.30 V)
driver type:	0 (driver type B)
```

closed 2019-10-11 by pengguang001