raspberrypi/linux #3274: leds: pca963x: Broken open-drain initialization

majorz, 2019-10-09:
When the pca963x driver is initialized in open-drain mode it does not set properly the OUTDRV setting (bit 2) of the Mode register 2. Instead of setting bit 2 to 0 the driver keeps it as 1.

The issue was introduced in bb29b9cccd95 ("leds: pca963x: Add bindings to invert polarity").

I have a small patch ready that fixes this and it looks like it should go into the mainline kernel first. I have a basic understanding of the principles in submitting patches for the mainline kernel, but I am struggling with the steps necessary upstreaming it, so that it gets on a Raspbian release sooner (not when Raspbian catches up with the mainline).

Will adding `Cc: stable@vger.kernel.org` in the sign-off area of the patch be enough for this?

I will appreciate any advice on what is necessary to propagate the patch to the right place and proper way.

pelwell, 2019-10-09:
A "Fixes: " tag is a good way to ensure it gets back-ported to stable branches, but in general we're happy to back-port any (working) commit into earlier kernels from the moment it gets merged into linux-next.

majorz, 2019-10-09:
Awesome, thanks a lot! I will follow up once it progresses further.

majorz, 2020-01-07:
Hi @pelwell, the patch landed on linux-leds. Would it be possible for it to be incorporated here, so that it gets included in an upcoming Raspbian release?
https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/drivers/leds/leds-pca963x.c?id=697529091ac7a0a90ca349b914bb30641c13c753

pelwell, 2020-01-07:
No problem - it's in rpi-4.19.y and rpi-5.4.y now.

majorz, 2020-01-07:
Awesome, thank you so much!

closed 2020-01-07 by majorz