raspberrypi/linux #3276: Enable modules for Wiznet chips in the default config

geeks-r-us, 2019-10-09:
Please add the kernel modules for the Wiznet SPI Ethernet chips to ease the use of chips like W5100 and W5500, which are very cheap and support 100 MBit/s in opposite to the ENC28J60.

To make it more convenient I've attached these tested .dts files to connect the W5500 on CE0 and CE1

Many thx

[w5500-overlay.dts.txt](https://github.com/raspberrypi/linux/files/3707415/w5500-overlay.dts.txt)
[w5500-overlay-spi2.dts.txt](https://github.com/raspberrypi/linux/files/3707416/w5500-overlay-spi2.dts.txt)

pelwell, 2019-10-10:
See #3278 - please test it before I merge it.

geeks-r-us, 2019-10-12:
works fine. 
Thx

pelwell, 2019-10-16:
#3278 is now merged, and will be in future builds.

closed 2019-12-20 by JamesH65