raspberrypi/linux #3282: (ds1307/ds3231) hwclock: ioctl(RTC_RD_TIME) to /dev/rtc0 to read the time failed: Invalid argument

boatbodger, 2019-10-11:
**Describe the bug**
When running hwclock on a Pi 4 with a DS3231 RTC, an "invalid argument" error occurs

**To reproduce**
Install DS3231-based RTC on i2c bus
Add following line to config.txt:
`dtoverlay=i2c-rtc,ds3231 `
Follow steps to remove fake hardware clock given in
https://learn.adafruit.com/adding-a-real-time-clock-to-raspberry-pi/set-rtc-time
reboot
Verify presence of kernel module (rtc-ds1307), 
verify device 'adopted' (UU showing in i2cdetect -y 1)
issue` hwclock --verbose` command

**Expected behaviour**
Expect to see contents of RTC displayed

**Actual behaviour**

IOCTL failure 'invalid argument'

root@ftk-mys-svr:~# hwclock --verbose
hwclock from util-linux 2.33.1
System Time: 1570818907.938321
Trying to open: /dev/rtc0
Using the rtc interface to the clock.
Last drift adjustment done at 1570811357 seconds after 1969
Last calibration done at 1570811357 seconds after 1969
Hardware clock is on UTC time
Assuming hardware clock is kept in UTC time.
Waiting for clock tick...
**ioctl(3, RTC_UIE_ON, 0): Invalid argument**
Waiting in loop for time from /dev/rtc0 to change
hwclock: ioctl(RTC_RD_TIME) to /dev/rtc0 to read the time failed: Invalid argument
...synchronization failed
root@ftk-mys-svr:

**System**
Raspberry Pi 4, 2Gb

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage4

* Which firmware version (`vcgencmd version`)?
Sep 24 2019 17:34:30
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux ftk-mys-svr 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

**Logs**
dmesg after startup:
[    4.996772] rtc-ds1307 1-0068: registered as rtc0
No further entrise after issuing hwclock command

**Additional context**
 This seems very similar to 
https://github.com/raspberrypi/firmware/issues/1065

The same issue, but for a different RTC chip.
(I raised this issue in Linux, because I don't think it is a firmware issue - I could easily be wrong)

I also note that despite having ds3231 in the overlay command, the ds1307 driver is pulled in, and I see ds3231 as an alias.

I notice that the chip on the RTC board is actually a DS3231**M**, which I believe simply means it uses an MEMS oscillator rather than a quartz crystal.  I doubt that makes any difference to the register behaviour, though.

boatbodger, 2019-10-11:
Out of curiosity, I left the system alone for three hours, and then issued 'hwclock' again.
That time I got a response.
If I do it again immediately afterwards, the error recurs.
It feels like reading it "knocks it out", and after some time it is sufficiently recovered for another read:
`root@ftk-mys-svr:~# hwclock`
`2019-10-11 23:16:05.987908+01:00`
`root@ftk-mys-svr:~# hwclock`
`hwclock: ioctl(RTC_RD_TIME) to /dev/rtc0 to read the time failed: Invalid argument`
`root@ftk-mys-svr:~# top`
`top - 23:19:13 up  3:34,  4 users,  load average: 1.00, 1.04, 1.01`

boatbodger, 2019-10-12:
After more testing, using verbose mode, even on a *successful* hwclock, the ioctl Invalid argument occurs, but hwclock reports receiving a clock tick:
`root@ftk-mys-svr:~# hwclock --verbose --show`
`hwclock from util-linux 2.33.1`
`System Time: 1570885984.794421`
`Trying to open: /dev/rtc0`
`Using the rtc interface to the clock.`
etc etc then
`Waiting for clock tick...`
`ioctl(3, RTC_UIE_ON, 0): Invalid argument`
`Waiting in loop for time from /dev/rtc0 to change`
`...got clock tick`
`Time read from Hardware Clock: 2019/10/12 13:13:05`
etc etc then
`2019-10-12 14:13:04.971991+01:00`
`root@ftk-mys-svr:~#`
So despite the Invalid argument it succeeds.

By contrast, a failed attempt reads thus:
`root@ftk-mys-svr:~# hwclock --verbose --show`
`hwclock from util-linux 2.33.1`
`System Time: 1570885997.152220`
`Trying to open: /dev/rtc0`
`Using the rtc interface to the clock.`
etc etc
`Waiting for clock tick...`
`ioctl(3, RTC_UIE_ON, 0): Invalid argument`
`Waiting in loop for time from /dev/rtc0 to change`
`hwclock: ioctl(RTC_RD_TIME) to /dev/rtc0 to read the time failed: Invalid argument`
`...synchronization failed`

From this I conclude that a) there is a probelm causing ioctl(3, RTC_UIE_ON, 0) to always fail
and b) something 'goes wrong' during the looping waiting for the seconds to change.

I have also inspected the source code for rtc-ds1307.c and all appears to be correct.
So I'm closing this now, and will look for the source code for hwclock and see if I can find a problem there.

closed 2019-10-12 by boatbodger

pelwell, 2019-10-12:
The error is caused by the lack of an interrupt signal from the RTC - the I and the E in RTC_UIE stand for Interrupt Enable. As you've discovered, that shouldn't stop the basic RTC functions working.

qjayjayp, 2021-04-15:
I had the same error, and for me the cause was that the initial value for the month was 0x1f, which is neither BCD (as it is supposed to) nor a value somewhere within the proper range 8yes it is accidentaly decimal 31, but it was the month field). I finally took a python script that could write the system date and time via /dev/i2c, from this moment on everything worked as expected.

V10lator, 2021-09-26:
Why has this been closed? This is an issue when using Chrony, for example:
```chronyd[383]: Could not enable RTC interrupt : Invalid argument```
This pevents chrony from measuring the RTC clock drift and as a result from acting against it. The DS3231 is capapble of emitting a 1 Hz signal (which corresponds to "interrupt on every clock update, for RTCs that support this once-per-second interrupt") on the SQW pin. Uptream DT even mentions it but doesn't tell how to route it to a GPIO pin and from there back to the driver: https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/rtc/maxim%2Cds3231.txt - Raspberry DT completely ignores the `#clock-cells` parameter.

pelwell, 2021-09-27:
Answering in the Forums - please don't cross-post.

seamusdemora, 2022-02-28:
@pelwell : Which forum is that answer in, please? 

I've ***almost*** the same error (`ioctl(4, RTC_UIE_ON, 0): Invalid argument`), `dmesg` says, `[    6.579872] rtc-ds1307 0-0068: 'wakeup-source' is set, request for an IRQ is disabled!`, and I do have the `wakeup-source` param set in `/boot/config.txt`: `dtoverlay=i2c-rtc,ds3231,wakeup-source,i2c0`

by, 2022-04-17:
When compiling a customer kernel with CONFIG_RTC_INTF_DEV_UIE_EMUL, the error is gone.

kurahaupo, 2022-11-29:
> [...] the initial value for the month was 0x1f [...] decimal 31

That could also be `-1` expressed as a 5-bit 2's-complement number, perhaps trying to indicate that the month isn't available?

alexdelprete, 2023-02-16:
> Why has this been closed? This is an issue when using Chrony, for example: `chronyd[383]: Could not enable RTC interrupt : Invalid argument` This pevents chrony from measuring the RTC clock drift and as a result from acting against it.

has this ever been solved? I have the exact same issue with an uputronics gps hat with rv3028.

I couldn't find a follow-up discussion on the forums...

boatbodger, 2023-02-16:
I closed it because in my case, the RTC module I was using did not have an interrupt output connected - and as pelwell explained, the error message was effectively a warning about that being the case (albeit not a very obvious way of saying it).
For me, therefore, there was no bug to fix.   My use case did not need the RTC to be able to create interrupts.  If you think you need an interrupt, make sure your hardware actually has one connected, and then fiddle with the devicetree to enable it.

alexdelprete, 2023-02-16:
> When compiling a customer kernel with CONFIG_RTC_INTF_DEV_UIE_EMUL, the error 'hwclock: ioctl(3, RTC_UIE_ON, 0): Invalid argument' is gone.

Can `CONFIG_RTC_INTF_DEV_UIE_EMUL` be defined in `/boot/config.txt` or a custom kernel is the only way?

alexdelprete, 2023-02-18:
> When compiling a customer kernel with CONFIG_RTC_INTF_DEV_UIE_EMUL, the error 'hwclock: ioctl(3, RTC_UIE_ON, 0): Invalid argument' is gone.

Thanks a lot. I compiled 6.1.11 and it worked perfectly. I can finally use chrony RTC drift monitoring functionality.