raspberrypi/linux #3283: ERROR: ld.so:

0xh0n3y, 2019-10-13:
i'm getting this error in raspberrypi jessi:-
ERROR: ld.so: object '�'�&�&�'�&�&�'�&�&�'�&�&�'�&�&�' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
whenever i type ls command in the pi terminal
[image]

XECDesign, 2019-10-13:
Looks like your /etc/ld.so.preload files got corrupted (at least) rather than an issue with the Linux kernel.

JamesH65, 2019-10-18:
Indeed - this looks like a corrupted SD card. I suggest re-imaging the card to see if that helps, if not, then getting a new card -  this one could be end of life.

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2019-10-27 by 0xh0n3y