raspberrypi/linux #3284: Kernel panic when using Raspberry Pi as a Repeater

cecilcosta, 2019-10-13:
**Describe the bug**
Following this tutorial (https://pimylifeup.com/raspberry-pi-wifi-extender/) got a kernel panic when a device connects to the wifi host 

**To reproduce**
Follow this tutorial: https://pimylifeup.com/raspberry-pi-wifi-extender/ (even the dongle is the same), however in my case Wlan0 is Wlan1 and Wlan1 is Wlan0.
At the end we get a new Wifi host running, I try to connect with a MacBook and automatically the Rasberry gets a kernel panic.

**Expected behaviour**
The raspberry should keep alive and should be used as a Wifi extender

**Actual behaviour**
Kernel panic: slab.c:3392

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi?  Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)? 2019-09-26 (lite-buster)
* Which firmware version (`vcgencmd version`)? 6820edeee4ef3891b95fc01cf02a7abd7ca52f17
* Which kernel version (`uname -a`)? 4.19.75-v7+

**Logs**
Got kenel panic

**Additional context**
fresh install, no addiotional software were installed, just apt-get update and upgrade.

lategoodbye, 2019-10-13:
Please provide the following information:
- involved Wifi devices on the Raspberry Pi 3B+
- complete kernel output after panic (dmesg / /var/log/kern.log)

cecilcosta, 2019-10-13:
hwinfo --network

`15: None 01.0: 1070a WLAN
  [Created at net.126]
  Unique ID: djUx.QXn1l67RSa1
  Parent ID: 0DfK.dZc7syf8JZ2
  SysFS ID: /class/net/wlan1
  SysFS Device Link: /devices/platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1
  Hardware Class: network interface
  Model: "WLAN network interface"
  Driver: "brcmfmac"
  Driver Modules: "brcmfmac", "brcmfmac"
  Device File: wlan1
  HW Address: b8:27:eb:7c:a2:9b
  Permanent HW Address: b8:27:eb:7c:a2:9b
  Link detected: yes
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #5 (Network controller)

16: None 00.0: 10700 Loopback
  [Created at net.126]
  Unique ID: ZsBS.GQNx7L4uPNA
  SysFS ID: /class/net/lo
  Hardware Class: network interface
  Model: "Loopback network interface"
  Device File: lo
  Link detected: yes
  Config Status: cfg=new, avail=yes, need=no, active=unknown

17: None 00.0: 10701 Ethernet
  [Created at net.126]
  Unique ID: usDW.ndpeucax6V1
  Parent ID: Vl7j.8_I5RcIC_05
  SysFS ID: /class/net/eth0
  SysFS Device Link: /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.1/1-1.1.1:1.0
  Hardware Class: network interface
  Model: "Ethernet network interface"
  Driver: "lan78xx"
  Driver Modules: "lan78xx"
  Device File: eth0
  HW Address: b8:27:eb:29:f7:ce
  Permanent HW Address: b8:27:eb:29:f7:ce
  Link detected: no
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #9 (Ethernet controller)

18: None 00.0: 1070a WLAN
  [Created at net.126]
  Unique ID: AYEt.QXn1l67RSa1
  Parent ID: N0O2.nGSTlewb0M9
  SysFS ID: /class/net/wlan0
  SysFS Device Link: /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0
  Hardware Class: network interface
  Model: "WLAN network interface"
  Driver: "rtl8192cu"
  Driver Modules: "8192cu"
  Device File: wlan0
  HW Address: 74:da:38:ea:80:b4
  Permanent HW Address: 74:da:38:ea:80:b4
  Link detected: yes
  Config Status: cfg=new, avail=yes, need=no, active=unknown
  Attached to: #7 (Unclassified device)
`

kern.log

`Jul 10 01:21:21 raspberrypi kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jul 10 01:21:21 raspberrypi kernel: [    0.000000] Linux version 4.19.57-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1244 SMP Thu Jul 4 18:45:25 BST 2019
Jul 10 01:21:21 raspberrypi kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Jul 10 01:21:21 raspberrypi kernel: [    0.000000] CPU: div instructions available: patching division code
[... 938 lines cut ...]
Oct 13 13:31:58 raspberrypi kernel: [   16.524843] Bluetooth: HCI device and connection manager initialized
Oct 13 13:31:58 raspberrypi kernel: [   16.524903] Bluetooth: HCI socket layer initialized
Oct 13 13:31:58 raspberrypi kernel: [   16.524925] Bluetooth: L2CAP socket layer initialized
Oct 13 13:31:58 raspberrypi kernel: [   16.524987] Bluetooth: SCO socket layer initialized
Oct 13 13:31:58 raspberrypi kernel: [   16.553871] Bluetooth: HCI UART driver ver 2.3
Oct 13 13:31:58 raspberrypi kernel: [   16.553892] Bluetooth: HCI UART protocol H4 registered
Oct 13 13:31:58 raspberrypi kernel: [   16.554045] Bluetooth: HCI UART protocol Three-wire (H5) registered
Oct 13 13:31:58 raspberrypi kernel: [   16.554403] Bluetooth: HCI UART protocol Broadcom registered
Oct 13 13:31:58 raspberrypi kernel: [   16.924172] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Oct 13 13:31:58 raspberrypi kernel: [   16.924193] Bluetooth: BNEP filters: protocol multicast
Oct 13 13:31:58 raspberrypi kernel: [   16.924241] Bluetooth: BNEP socket layer initialized
Oct 13 13:33:55 raspberrypi kernel: [  123.432621] Voltage normalised (0x00000000)
Oct 13 13:34:10 raspberrypi kernel: [  138.008132] Under-voltage detected! (0x00050005)
Oct 13 13:38:01 raspberrypi kernel: [  368.865954] Voltage normalised (0x00000000)
Oct 13 13:38:03 raspberrypi kernel: [  370.941428] Under-voltage detected! (0x00050005)
Oct 13 13:38:42 raspberrypi kernel: [  410.465949] Voltage normalised (0x00000000)
Oct 13 13:38:44 raspberrypi kernel: [  412.558148] Under-voltage detected! (0x00050005)
Oct 13 13:39:03 raspberrypi kernel: [  431.265964] Voltage normalised (0x00000000)
Oct 13 13:39:18 raspberrypi kernel: [  445.832646] Under-voltage detected! (0x00050005)
Oct 13 13:39:30 raspberrypi kernel: [  458.315951] Voltage normalised (0x00000000)
Oct 13 13:39:53 raspberrypi kernel: [  481.182627] Voltage normalised (0x00000000)
Oct 13 13:45:05 raspberrypi kernel: [  793.190634] rpi_firmware_get_throttled: 5 callbacks suppressed
Oct 13 13:45:05 raspberrypi kernel: [  793.190642] Voltage normalised (0x00000000)
Oct 13 13:45:07 raspberrypi kernel: [  795.282661] rpi_firmware_get_throttled: 7 callbacks suppressed
Oct 13 13:45:07 raspberrypi kernel: [  795.282670] Under-voltage detected! (0x00050005)

cecilcosta, 2019-10-13:
BTW: these files are from the previous compilations (10/07/2019) as I tried to see it if was a recent issue or a new one.

lategoodbye, 2019-10-13:
Unfortunately there is no kernel panic in the log, except of a unreliable power supply.

Any chance to get the full stacktrace (even a photo could be helpful)?

cecilcosta, 2019-10-13:
https://youtu.be/T7YIHVtg-wE

cecilcosta, 2019-10-13:
[KernelPanic2.mov.zip](https://github.com/raspberrypi/linux/files/3721945/KernelPanic2.mov.zip)

JamesH65, 2019-10-18:
Before looking further into this, I would suggest you get a decent power supply and retest. You are using a separate Realtek USB Wifi adapter by the looks of it, and with a low power supply, that will also be under powered, so anything could be happening. We need to determine that its not a power related problem before anything else.

cecilcosta, 2019-10-19:
@JamesH65 It seems that you are right. I changed the power supply and now it doesn't generate a kernel panic any more. Even more, now the wlan0 is the built-in adapter.

closed 2019-10-19 by cecilcosta