raspberrypi/linux #3285: tc358743: broke with unicam change

mdevaev, 2019-10-14:
**Describe the bug**
After updating the kernel from version 4.19.76 to 4.19.79, the video capture module Auvidea B101 based on HDMI bridge TC358743 stopped working. Instead of an image, it shows a green screen, while maintaining the proportions and the correct resolution of the HDMI source (like 720p or 1080p).

After a little research, I realized that this was most likely due to the latest changes in the unicam driver: #3265. Somehow it changed the search order of the fourcc format by `struct unicam_fmt formats[]`. Previously, when searching, the correct result was `MEDIA_BUS_FMT_UYVY8_1X16` (0x200F). Now it's a `MEDIA_BUS_FMT_UYVY8_2X8` (0x2006). This causes a device initialization error: `tc358743_set_csi_color_space: Unsupported format code 0x2006`.

**To reproduce**
Install Auvidea B101 with kernel 4.19.79 (https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=120702&p=1339178#p1339178) and run any capture software that supports DV-timings (https://github.com/6by9/yavta ).

**Expected behaviour**
Getting a normal image from HDMI.

**Actual behaviour**
Just a green screen instead of it.

**System**
* Which model of Raspberry Pi?
4B 1Gb
* Which OS and version?
Arch Linux ARM (uses the kernel from this repo).
* Which firmware version?
```
Oct 11 2019 18:39:26
Copyright (c) 2012 Broadcom
version d4a63855399ed5c2e6cd94e6dbcd850d881c856d (tainted) (release) (start_x)
```
* Which kernel version (`uname -a`)?
```
Linux pikvm 4.19.79-1-ARCH #1 SMP PREEMPT Sat Oct 12 16:59:03 UTC 2019 armv7l GNU/Linux
```

**Logs**
Normal work on 4.19.76:
```
tc358743 0-000f: tc358743_set_csi_color_space: YCbCr 422 16-bit
unicam fe801000.csi: __subdev_set_format 640x480 code:200f
unicam fe801000.csi: __subdev_get_format 640x480 code:200f
unicam fe801000.csi: unicam_calc_format_size_bpl: fourcc: UYVY size: 640x480 bpl:1280 img_size:614400
unicam fe801000.csi: unicam_s_fmt_vid_cap 640x480, mbus_fmt 0000200F, V4L2 pix UYVY.
```

Incorrect work on 4.19.79:
```
tc358743 0-000f: tc358743_set_csi_color_space: Unsupported format code 0x2006
unicam fe801000.csi: __subdev_set_format 640x480 code:2006
unicam fe801000.csi: __subdev_get_format 640x480 code:2006
unicam fe801000.csi: unicam_calc_format_size_bpl: fourcc: UYVY size: 640x480 bpl:1280 img_size:614400
unicam fe801000.csi: unicam_s_fmt_vid_cap 640x480, mbus_fmt 00002006, V4L2 pix UYVY.
```

mdevaev, 2019-10-14:
@6by9 I apologize if I act incorrectly, I just inform you about this problem, because you're the author of the changes which, I assume, are the cause of the problem. You also seem to be the author of tc358743 driver. I'm grateful to you for your work.

PS: Sorry for my English.

6by9, 2019-10-14:
You're totally right - thanks for the report.

I'd been focussing on Bayer sensors and had forgotten about the silly one V4L2 fourcc to many media bus formats conversion for YUYV. It makes no sense on a serial bus such as CSI, only on a parallel bus, but we have to support it some way or other.
Doubly annoying is that the tc358743 uses ```UYVY8_1X16```, but the adv7180 (which we also nominally support) uses ```UYVY8_2X8``` :-(
The tc358743 driver also appears to not meet the V4L2 spec in that if the format isn't supported then the driver should normally choose an alternate format that is supported. Even if that weren't the case, then it would probably default back to RGB888.

As to the correct solution that's harder to say. I suspect we need to iterate calling ```enum_mbus_code``` during set_fmt/try_fmt each time to ensure that it picks up the format that the sensor advertises. Ugly, but probably necessary.

mdevaev, 2019-10-14:
Thanks for the quick response! Could you fix this? Unfortunately, I’m not a kernel hacker and I’m unlikely to cope with this, given the specifics of this driver.

6by9, 2019-10-14:
Yes, I will be looking to fix it, but I'm afraid it may take a few days due to other priorities.

6by9, 2019-10-16:
I think I have a solution to this, but out of time to push it now. Hopefully tomorrow.

6by9, 2019-10-17:
PR #3290 raised that fixes this.

mdevaev, 2019-10-17:
Thanks! Now I'll wait for a new kernel release.

JamesH65, 2019-12-20:
@mdevaev Did the new kernel release fix the issue?

mdevaev, 2019-12-20:
@JamesH65 yes

closed 2019-12-20 by pelwell