raspberrypi/linux #3286: sdtweak poll_once doesn't work on Pi 4

XECDesign, 2019-10-14:
When network booting a pi for with `dtoverlay=sdtweak,poll_once`, the card is still polled, causing log spam every 10 seconds:

```
[  144.474158] mmc0: Timeout waiting for hardware cmd interrupt.
[  144.474182] mmc0: sdhci: ============ SDHCI REGISTER DUMP ===========
[  144.474199] mmc0: sdhci: Sys addr:  0x00000000 | Version:  0x00001002
[  144.474205] mmc0: sdhci: Blk size:  0x00000000 | Blk cnt:  0x00000000
[  144.474213] mmc0: sdhci: Argument:  0x00000000 | Trn mode: 0x00000000
[  144.474221] mmc0: sdhci: Present:   0x1fff0001 | Host ctl: 0x00000001
[  144.474228] mmc0: sdhci: Power:     0x0000000f | Blk gap:  0x00000080
[  144.474234] mmc0: sdhci: Wake-up:   0x00000000 | Clock:    0x0000f447
[  144.474242] mmc0: sdhci: Timeout:   0x00000000 | Int stat: 0x00000000
[  144.474250] mmc0: sdhci: Int enab:  0x00ff1003 | Sig enab: 0x00ff1003
[  144.474257] mmc0: sdhci: ACmd stat: 0x00000000 | Slot int: 0x00000000
[  144.474262] mmc0: sdhci: Caps:      0x45ee6432 | Caps_1:   0x0000a525
[  144.474267] mmc0: sdhci: Cmd:       0x0000371a | Max curr: 0x00080008
[  144.474275] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
[  144.474280] mmc0: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[  144.474284] mmc0: sdhci: Host ctl2: 0x00000000
[  144.474288] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x00000000
[  144.474293] mmc0: sdhci: ============================================
```

pelwell, 2019-10-14:
The immediate cause of sdtweak's failure to set the non-removable flag on pi4 is that it isn't meant to:
```
pi@raspberrypi:~ $ dtoverlay -h sdtweak
Name:   sdtweak

Info:   Tunes the bcm2835-sdhost SD/MMC driver
        N.B. This functionality is now available via the sd_* dtparams in the
        base DTB.
...
```
What is needed is a generic `sd_pollonce` dtparam, which is easy enough to add. However, it might not be what you want in that may not react well when an SD card is removed (or inserted late).

Perhaps what is really needed, in addition to `sd_pollonce`, is a driver change to reduce the log spam, perhaps making it conditional on `sd_debug`.

loyclay, 2020-01-30:
can this be resolved, please?

pelwell, 2020-01-30:
I've pushed updated .dts files to rpi-4.19.y that add sd_poll_once dtparams to all downstream .dtbs. You can grab a pre-built version of bcm2711-rpi-4-b.dtb [here](https://drive.google.com/file/d/1ZLULdHQ0r6qgL3AwEuZCmND2USNwh-7b/view?usp=sharing). Use with `dtparam=sd_poll_once`.

ricktendo, 2020-06-17:
After using `sd_poll_once` now the green LED does not work

ctborg, 2020-11-03:
Hi @pelwell you pushed a build to Google Drive back in January that fixed this issue.  Any chance you can post the patch here?

pelwell, 2020-11-03:
As the sdtweak documentation says, the functionality it provides is now available via the board-specific dtparams:
```
dtparam=sd_poll_once
```
sdtweak is effectively deprecated, and will be formally deprecated and removed.

ssackett, 2021-02-22:
@pelwell 

Hi,

I am hoping you can help me with a problem similar to the one in this thread.

I have a PI 4B and am experiencing a kworker that hogs the cpu only when using WIFI.  I have updated the kernel to 5.10.16-v7l+ and added "dtparam=sd_poll_once=on" to config.txt.  I have verified that "sd_poll_once" is active using "dtc -I fs /proc/device-tree" which shows "sd_poll_once = "\0\0\0?non-removable?";".  None of these changes fixed the problem.

Using top the the first two lines are:

1387 root      20   0       0      0      0 I  21.6   0.0   0:16.66 kworker/u8:2-brcmf_wq/mmc1:0001:1                                                                             
  839 root      20   0    6964   2380      4 S  14.0   0.1   0:18.22 vhusbdarmpi4                           

The offending kworker always shows as "kworker/u8:2-brcmf_wq/mmc1:0001:1" with the number after "/u8:" changing between 2, 3, and 4.  Also the number is followed alternately by "+"  and "-" (ie: "u8:2+brcmf" and "u8:2-brcmf").  

The process "vhusbdarmpi4" is running as a service started by systemd,  it is communicating via WIFI and is running a USBIP protocol.   If I use a wired connection instead of WIFI the kworker disappears and the service runs as expected.

Thanks,

Steve

pelwell, 2021-02-22:
Please start a new thread for a new issue.

andrewufrank, 2021-04-18:
I have observed that the patch to insert the `sd_poll_once` dtparam is not in the raspi-firmware `bmc2711-rpi-4-b.dtb` for bullseye and threrore it is not surprising that the dtparam does not have an effect. I tried to copy the new version from the linux project to the firmware project, but do not understand about the necessary conversion etc. What are the correct procedures to have this change arrive in the firmware used for bullseye? I pose this question in both, the firmware and the linux project, not knowing who should act. Thank you for help and clarification!

closed 2022-10-31 by pelwell