raspberrypi/linux #3287: Incorrectly configured baudrate on SC16IS752 dual uart using CM3 pi

neiljlbennett, 2019-10-16:
Hi,
I have been searching all day but can't find any mention of the problem I am having. So, I really hope that I am not alone with this and that somebody could suggest something helpful or maybe confirm this bug.

I have a CM3 connected to a dual UART device SC16IS752 using i2c.

I have written a small test in C that makes use of the following driver (kernel module)

$ modinfo sc16is7xx.ko
filename: /lib/modules/4.19.50-v7+/kernel/drivers/tty/serial/sc16is7xx.ko
description: SC16IS7XX serial driver

The immediate problem is that the UART appears to end up being configured to be 1/8 th of the baud-rate I have defined when opening the serial port !!
e.g I specify 9600bps but actually get 1200bps on the wire, this x/8 relationship is true for other baud-rates too.

I included the driver with the following entry in config.txt

dtoverlay=sc16is752-i2c,int-pin=5,addr=0x4D,xtal=1843200

N.B. Playing around with the xtal value makes no difference and I have confirmed the crystal frequency using a 'scope.

The test code is shown below (it's based on an example I found) .

It's worth pointing out that when I ditched the SC16IS7xx driver and tested the UART with a python script using the smbus (i2c) library the baudrate was correct.

I haven't been working with RPi for very long and I have exhausted my limited knowledge, any help is much appreciated or have I actually found a bug?

Kind Regards
NeilB

```

#include <fcntl.h>			//Used for UART
#include <termios.h>	//Used for UART

#define UART_DEV1    "/dev/ttySC0"
#define UART_DEV2    "/dev/ttySC1"

int main(void)
{
    if(wiringPiSetupGpio() < 0) { //use BCM2835 Pin number table
        printf("set wiringPi lib failed	!!! \r\n");
        return 1;
    } else {
        printf("set wiringPi lib success  !!! \r\n");
    }

    int fd = -1;
    /*
    if((fd = serialOpen (UART_DEV2, 115200)) < 0) {
        printf("serial err\n");
        return -1;
    }
    */
    //OPEN THE UART
	//The flags (defined in fcntl.h):
	//	Access modes (use 1 of these):
	//		O_RDONLY - Open for reading only.
	//		O_RDWR - Open for reading and writing.
	//		O_WRONLY - Open for writing only.
	//
	//	O_NDELAY / O_NONBLOCK (same function) - Enables nonblocking mode. When set read requests on the file can return immediately with a failure status
	//											if there is no input immediately available (instead of blocking). Likewise, write requests can also return
	//											immediately with a failure status if the output can't be written immediately.
	//
	//	O_NOCTTY - When set and path identifies a terminal device, open() shall not cause the terminal device to become the controlling terminal for the process.
	fd = open(UART_DEV2, O_RDWR | O_NOCTTY | O_NDELAY);		//Open in non blocking read/write mode
	if (fd == -1)
	{
		//ERROR - CAN'T OPEN SERIAL PORT
		printf("Error - Unable to open UART.  Ensure it is not in use by another application\n");
	}
    //CONFIGURE THE UART
	//The flags (defined in /usr/include/termios.h - see http://pubs.opengroup.org/onlinepubs/007908799/xsh/termios.h.html):
	//	Baud rate:- B1200, B2400, B4800, B9600, B19200, B38400, B57600, B115200, B230400, B460800, B500000, B576000, B921600, B1000000, B1152000, B1500000, B2000000, B2500000, B3000000, B3500000, B4000000
	//	CSIZE:- CS5, CS6, CS7, CS8
	//	CLOCAL - Ignore modem status lines
	//	CREAD - Enable receiver
	//	IGNPAR = Ignore characters with parity errors
	//	ICRNL - Map CR to NL on input (Use for ASCII comms where you want to auto correct end of line characters - don't use for bianry comms!)
	//	PARENB - Parity enable
	//	PARODD - Odd parity (else even)
	struct termios options;
	tcgetattr(fd, &options);
	options.c_cflag = B9600 | CS8 | CLOCAL | CREAD;		//<Set baud rate
	options.c_iflag = IGNPAR;
	options.c_oflag = 0;
	options.c_lflag = 0;
	tcflush(fd, TCIFLUSH);
	tcsetattr(fd, TCSANOW, &options);
    
    int setBROut = 0;
    int setBRIn = 0;
    setBROut = cfgetospeed(&options);
    setBRIn = cfgetispeed(&options);
    printf("BR Out: %d\r\n", setBROut);  
    printf("BR In: %d\r\n", setBRIn); 
    
    serialFlush(fd);
    serialPrintf(fd,"\r\n");

    char *buf = "abcdefgh";    
    serialPuts(fd, buf);

    printf("send data: %s\r\n", buf);    

    serialClose(fd);
    return 0;
}
```

**Additional context**
Add any other relevant context for the problem.

pelwell, 2019-10-16:
Your `int-pin` parameter is wrong - it should be `int_pin` - but I'll assume that's a typo in this issue and not in your config.txt. Also, your xtal value of 1.8432MHz seems low - should that have been 18.432MHz?

pelwell, 2019-10-16:
And since you have a scope, could you attach a photo/screen shot of a few bytes worth of waveform, with the timebase visible (ideally with markers positioned before a start bit and after the corresponding stop bit)?

neiljlbennett, 2019-10-16:
Thanks for your interest Phil , I corrected the int_pin parameter as pointed out, but it didn't change anything.
I can confirm the crystal is 1.8432MHz - I fiddled around with this parameter but it had no effect on the transmitted baudrate.
I should stress that I send correctly formed bytes to my terminal app (Teraterm) but I have to change the baudrate on the teraterm port to 1200 baud to see them.

Here's the waveform for the character '1' (0x31)
Thanks NeilB

[image]

neiljlbennett, 2019-10-16:
Hers a few bytes together '1''2''3'
[image]

pelwell, 2019-10-16:
Ah ha - I was just looking at a list of relatively recent commits and saw one called "overlays: sc16ic752-i2c: Fix xtal parameter". The overlay was patching the wrong property, resulting in the default value of 14.7456MHz being used - 1.8432/14.7456 = 0.125, so the baud rate would be an eighth of what was expected. The fix went in mid-August, so try:
```
$ sudo apt update
$ sudo apt install raspberrypi-kernel
```

neiljlbennett, 2019-10-16:
Thankyou very much Phil! That did the trick.
I had seen the commit you found on my searches but I didn't connect it with my issue.

closed 2019-10-16 by neiljlbennett