raspberrypi/linux #3294: Please enable CGROUP_BPF so one can use ip vrf exec

mateuszviste, 2019-10-20:
A couple of weeks ago VRF support has been enabled in the raspbian kernel via ae5fd2e, and that's truly awesome. Many applications may use VRFs now, but sometimes there is no other way for running an vrf-unaware application that using the 'ip vrf exec' command. 'ip vrf exec' lauches an application and makes sure that said application is bound to a VRF, and that it uses the vrf-associated routing table instead of the default (0) rtable.

Unfortunately, 'ip vrf exec' cannot be used on Raspbian for the time being, as it appears to be missing CGROUP_BPF. For example trying to bind dhclient to a VRF leads to an error message:

`root@raspberrypi:~# ip vrf exec myvrf dhclient -nw`
`Failed to load BPF prog: 'Function not implemented'`
`Kernel compiled with CGROUP_BPF enabled?`
`root@raspberrypi:~#`

Would it be possible to enabled CGROUP_BGP please?

mateuszviste, 2019-10-24:
I am wondering now, that perhaps my description may sound somewhat obscure. If that's the case, please tell, so I will do my best to clarify the issue.

pelwell, 2019-10-24:
I suspect nobody's looked at this yet, so a quick explanation of what CGROUP_BPF does would be helpful when considering whether to include it as standard. If the option can be enabled as a module and doesn't take any extra space or resources when not loaded then it's usually an easy decision to make.

Ferroin, 2019-10-24:
@pelwell In short, it's a way to attach an eBPF program to a control group.  This has a bunch of potential uses depending on what controllers are attached to the control group, but the most common is cases like this where you need to enforce certain network interactions for all packets coming from or leaving the cgroup.

`CGROUP_BPF` itself is all or nothing, though most of what it actually does is add hooks at various locations in the control groups.  Total runtime overhead is pretty negligible when not in use, though I don't have exact numbers.

Actually enabling and using it also requires `BPF_SYSCALL`, which does not appear to be enabled currently either.  That should have zero runtime overhead other than however big the compiled code for it is, as all the other stuff that would pull in already appears to be getting pulled in by other things that use BPF.

mateuszviste, 2019-10-26:
This update just to say that I managed to work around the missing CGROUP_BPF support. My need was mainly about running dhclient for a VRF interface - I achieved that by hacking dhclient lease scripts, so they updated the correct (vrf-bound) routing table instead of the default rtable. In this specific case, no need for 'ip vrf exec' after all. Feel free to close this issue if you think that the feature is non-essential.

pelwell, 2019-10-27:
We were short-staffed last week, but I'll get it tested and (probably) merged tomorrow.

pelwell, 2019-10-28:
Enabling CGROUP_BPF and BPF_SYSCALL didn't reduce free memory noticeably, and neither iperf nor sysbench see any real degradation.

Are you happy for me to proceed, @popcornmix?

popcornmix, 2019-10-28:
If you are happy, then proceed.

pelwell, 2019-10-30:
These should be enabled in the next release.

JamesH65, 2019-12-20:
Closing this issue as questions answered/issue resolved.
Please update to the latest kernel which may contain a fix for this issue.

closed 2019-12-20 by JamesH65