raspberrypi/linux #3296: Digione driver mclk_div parameter need to correct as in  4.14.y

allocom, 2019-10-21:
@pelwell 

in 4.19.y  latest drivers 19k sample rate files are some time not playing properly.
we analyzed the code and noticed that mclk_div condition not satisfying in new driver.

refer previously working driver 
https://github.com/raspberrypi/linux/blob/rpi-4.14.y/sound/soc/bcm/allo-digione.c
lines 111- 133  : AlloDigiOne working  always mclk_div = WM8804_MCLKDIV_256FS;

[image]

On latest driver can you please add the condition for AlloDigiOne to configure only  WM8804_MCLKDIV_256FS  

https://github.com/raspberrypi/linux/blob/rpi-4.19.y/sound/soc/bcm/rpi-wm8804-soundcard.c
line 97 to 103

pelwell, 2019-10-22:
Does #3299 work for you?

allocom, 2019-10-23:
yes working. 
please proceed with merging.
thanks.

pelwell, 2019-10-25:
Closing issue, as the matter has been resolved.

closed 2019-10-25 by pelwell