raspberrypi/linux #3302: AARCH64: Unable to initialise DMA  channel. Falling back to PIO

sbromberger, 2019-10-27:
**Describe the bug**
New installation of 19.10 on Raspberry Pi 4 (4GB) results in `dmesg` output indicating that `mmc-bcm2835` is using PIO, which is not optimal.

**To reproduce**
Installed new OS, used `apt update` / `apt upgrade` to get it up to current, then examined dmesg output (I should note that it is my perception that the system is slower than raspbian, but perhaps that's expected and/or subjective.)

**Expected behaviour**
I'd expect mmc to use DMA instead of PIO.

**Actual behaviour**
```
[    3.516136] mmc-bcm2835 fe300000.mmcnr: mmc-bcm2835: Unable to initialise DMA
 channel. Falling back to PIO

...

[    4.157526] mmc0: No suitable DMA available - falling back to PIO
[    4.202533] mmc0: SDHCI controller on fe340000.emmc2 [fe340000.emmc2] using P
IO
[    4.290355] usb 1-1: device descriptor read/64, error 18
[    4.315241] mmc0: new ultra high speed DDR50 SDXC card at address 59b4
[    4.326484] mmcblk0: mmc0:59b4 7MKHS 58.4 GiB
```

**System**

 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

[Raspinfo output on pastebin](https://pastebin.com/URUi2QvR)

**Logs**
See above

**Additional context**
I tried adding the ppa for the vgencmd but it gave me errors about not being secure (something about `Releases` not being present?) so I removed it for the time being.

sbromberger, 2019-10-28:
Update: [following instructions here](https://jamesachambers.com/raspberry-pi-4-ubuntu-server-desktop-18-04-3-image-unofficial/comment-page-4/#comment-2223), I installed the new kernel and no longer see the PIO fallback messages.

pelwell, 2019-10-28:
Closing, as this issue has already been fixed (and was an Ubuntu problem).

closed 2019-10-28 by pelwell